vincentwolsink/home_assistant_enphase_envoy_installer #2: Support Q-Relay 3P

bes-r, 2023-05-03:
The current version does not support a multi phase system as the Q-Relay 3P. It would be nice if this will be integratie as seen in: 
https://github.com/posixx/home_assistant_custom_envoy

vincentwolsink, 2023-05-03:
I will take a look if I can copy over the 3 phase support. I cannot test it though, since I don’t have a multi phase system myself. Would be nice if you can help with that.

vincentwolsink, 2023-05-03:
@bes-r can you please try [0.1.6-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.1.6-beta1)

bes-r, 2023-05-03:
@vincentwolsink 0.1.6-beta2 is working with gateway software 07.06.114.
It shows the following additions for a 3-phase system:

- Power Consumption L1
- Power Consumption L2
- Power Consumption L3
- Power Production L1
- Power Production L2
- Power Production L3
- Lifetime Energy Consumption L1
- Lifetime Energy Consumption L2
- Lifetime Energy Consumption L3
- Lifetime Energy Production L1
- Lifetime Energy Production L2
- Lifetime Energy Production L3
- Today's Energy Consumption L1
- Today's Energy Consumption L2
- Today's Energy Consumption L3
- Today's Energy Production L1
- Today's Energy Production L2
- Today's Energy Production L3

Thank you for this!

closed 2023-05-03 by bes-r