vincentwolsink/home_assistant_enphase_envoy_installer #5: Unexpected error fetching envoy data: list index out of range

FreddyX078, 2023-05-09:
Hi, many thanks for your contibutions!

After update 0.1.8, addon will not load.
The error:

`Logger: custom_components.enphase_envoy Source: custom_components/enphase_envoy/envoy_reader.py:655  Integration: Enphase Envoy (Installer)  First occurred: 13:27:38 (1 occurrences)  Last logged: 13:27:38
Unexpected error fetching envoy Envoy 1234567891011 data: list index out of range
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _async_refresh
    self.data = await self._async_update_data()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 217, in _async_update_data
    return await self.update_method()
  File "/config/custom_components/enphase_envoy/__init__.py", line 108, in async_update_data
    data[description.key] = await envoy_reader.consumption_phase(
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 655, in consumption_phase
    return int(raw_json["consumption"][0]["lines"][phase_map[phase]]["wNow"])
IndexError: list index out of range`

Gateway is: Envoy-S-Metered-EM without CT's
Firmware is: 7.0.88

Did a fresh install, but got same error.
Rolling back to version 0.1.7 and everything is working again.

vincentwolsink, 2023-05-09:
Hi @FreddyX078, so many systems, so many different configurations 😄 

Can you maybe try [0.1.9-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.1.9-beta1)?

FreddyX078, 2023-05-09:
Yep, working again.
Thank you! 

_Je bent een held!_ 😉

closed 2023-05-09 by vincentwolsink