vincentwolsink/home_assistant_enphase_envoy_installer #11: Foutmelding bij initialiseren

jwluiten, 2023-05-10:
Hallo Vincent,

Ik krijg bij het instellen van de integratie de volgende foutmelding:
```
Logger: homeassistant.config_entries
Source: custom_components/enphase_envoy/__init__.py:132 
Integration: Enphase Envoy (DEV) 
First occurred: 23:47:23 (1 occurrences) 
Last logged: 23:47:23

Error setting up entry Envoy 122250101431 for enphase_envoy
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/enphase_envoy/__init__.py", line 132, in async_setup_entry
    hass.config_entries.async_setup_platforms(entry, PLATFORMS)
AttributeError: 'ConfigEntries' object has no attribute 'async_setup_platforms'
```

Ik heb geen idee wat het veroorzaakt. Ik gebruik de volgende versie van HomeAssistant:
```
Home Assistant 2023.5.2
Supervisor 2023.04.1
Operating System 10.1
Frontend-versie: 20230503.3 - latest
```

vincentwolsink, 2023-05-11:
Hoi Jan Willem,

Er lijkt iets niet goed gegaan tijdens de installatie. Deze integratie heet `Enphase Envoy (Installer)` maar de foutmelding gaat over `Enphase Envoy (DEV)`. Zorg ervoor dat alle andere varianten van de integratie eerst volledig verwijderd zijn en probeer dan de installatie opnieuw (via HACS).

jwluiten, 2023-05-11:
Hoi Vincent,

Dank voor je antwoord. Die stond inderdaad nog in custom_components.Ik heb die met de hand verwijderd en de boel opnieuw via HACS geïnstalleerd. Nu werkt het uitstekend.

Hartelijke groet,

Jan Willem

closed 2023-05-11 by vincentwolsink