vincentwolsink/home_assistant_enphase_envoy_installer #13: Lot's of status changes (not available)

posixx, 2023-05-16:
I see a lot of disconnects using this integration for the last week now. Every device (gateway. relay and inverters) often disconnects resulting in not available for some time. This happens about every hour.

[image]

posixx, 2023-05-16:
No nothing that i'm aware of. On PC and Phone  i use owner account when using enlighten.

vincentwolsink, 2023-05-16:
Does this happen only for the binary sensors (producing/communicating) or also for the other sensors? And how many solar panels do you have installed?

vincentwolsink, 2023-05-16:
Interesting, because it claims the token validity is a year there. Owner tokens are valid for a year, installer tokens only for 12 hours. Your installer account, is it a true installer account or a DHZ one?

Can you also maybe give a bit more context around this? The HTTP request just before the token refresh for example.

posixx, 2023-05-16:
I have a installer account no DHZ. You can download the log here: https://saasgui.yodacloud.net/sharing/bM1ezvaOe

Let me know if you got it; the link can be used only once..

vincentwolsink, 2023-05-16:
@posixx The url returns a page not found.

posixx, 2023-05-16:
Well then someone else has downloaded it. Sorry i don't have any mailadres or someting to reach you

posixx, 2023-05-17:
> Hi @posixx, do you have any other script/integration/job running that uses the installer account to connect to your Envoy? The Envoy only allows one installer session at a time.

I have created an installer account and connected it to my system. The original installer (the company who installed panels and configured everything) also still has connection to my system. Could that be the issue? Cannot imagine this would cause the problem as this installer doesn’t have access to internal ip etc..

vincentwolsink, 2023-05-17:
The limitation is on the local connection. The installer company uses the Enphase cloud so that is not an issue.

Can you please send the debug log to this (temporary) email: ziener06joduur@icloud.com

posixx, 2023-05-17:
Done!

vincentwolsink, 2023-05-22:
Thanks. In the logfile I see it is reusing the token like it should, but suddenly at 16:01 it either starts another update parallel to the regular one, or it crashed halfway during the run and lost the token (although I don't see any exceptions or errors).

Which version of Home Assistant are you running?

posixx, 2023-05-22:
> Thanks. In the logfile I see it is reusing the token like it should, but suddenly at 16:01 it either starts another update parallel to the regular one, or it crashed halfway during the run and lost [...]

Latest, 2023.5.3

Also, every day at exactly 23:00 and/or 0:00 i get a timeout on the envoy:

2023-05-16 23:00:42.448 ERROR (MainThread) [custom_components.enphase_envoy] Timeout fetching envoy Envoy xxxxxxxxxxxxxx data
2023-05-17 23:01:36.229 ERROR (MainThread) [custom_components.enphase_envoy] Timeout fetching envoy Envoy xxxxxxxxxxxxxx data
2023-05-18 23:01:46.359 ERROR (MainThread) [custom_components.enphase_envoy] Timeout fetching envoy Envoy 122220065544 data
2023-05-19 00:01:19.357 ERROR (MainThread) [custom_components.enphase_envoy] Timeout fetching envoy Envoy xxxxxxxxxxxxxx data
2023-05-20 23:01:47.160 ERROR (MainThread) [custom_components.enphase_envoy] Timeout fetching envoy Envoy xxxxxxxxxxxxxx data
2023-05-21 00:01:25.252 ERROR (MainThread) [custom_components.enphase_envoy] Timeout fetching envoy Envoy xxxxxxxxxxxxxx data
2023-05-22 00:00:49.062 ERROR (MainThread) [custom_components.enphase_envoy] Timeout fetching envoy Envoy xxxxxxxxxxxxxx data

vincentwolsink, 2023-05-22:
The timeout on the Envoy must be something outside the integration, so either on the Envoy itself, the network etc. I cannot reproduce the issue with the token currently. It is very suspicious that the token issue happens every hour. Actually the code around tokens did not change between this and your integration at all. What if you run your own versions of the integration with your installer account, does it also have the issue? And what about if you run it with your owner account?

posixx, 2023-05-22:
I never used my own fork with the D7 firmware; i switched over to yours. Also i didn't touch the token code nor did i look at it to understand it; i only expanded the repo from brian to include the 3-phase sensors. I will do some test if i find the time for it.

posixx, 2023-05-22:
I switched to my fork with installer login and will update tomorrow.

posixx, 2023-05-23:
Vincent,

Not a single disconnect using my fork for mort than a day:

[image]

The timeout messages are still there in the HA log at 23u or 00u.

vincentwolsink, 2023-05-24:
Then the issue is not in the token code, since that didn’t change between forks. Maybe something with the high number of sensors. But currently I have no way of debugging this since I cannot reproduce the issue.

posixx, 2023-05-24:
OK, then i will continue to use my fork for now..

closed 2023-05-24 by posixx

vincentwolsink, 2023-05-24:
@posixx You could also help me and maybe others by doing some debugging for me, since you are also somewhat familiar with the code.