vincentwolsink/home_assistant_enphase_envoy_installer #15: Net alle metrics/sensoren beschikbaar

reinvk, 2023-05-17:
Hallo Vincent,

Dank voor de mooi integratie in Home Assistant.
Sinds 1,5 maand ook een Enphase PV systeem.
Zag dat er met deze integratie alle informatie per Inverter zichtbaar zou moeten zijn.
Zou graag deze details ook in Home Assistent beschikbaar hebben.
Dat is bij mij niet het geval maar komt waarschijnlijk doordat ik een DHZ account heb in Enphase Enlighten.
Of zou het met dit account ook moeten werken ?

bes-r, 2023-05-18:
Mijn DHZ-account toont alle informatie. Welke versie heb je draaien?

reinvk, 2023-05-18:
> Mijn DHZ-account toont alle informatie. Welke versie heb je draaien? 

Ah, dat biedt mogelijkheden hoop ik. De Envoy heeft versie D7.6.114. Of bedoel je andere versies van Home assistant bijv. is 2023.4.6

vincentwolsink, 2023-05-18:
Hoi @reinvk, het zou met een DHZ account ook moeten werken. Ook de software versie van je Envoy ziet er goed uit. Welke gegevens zie je wel?

reinvk, 2023-05-18:
> Hoi @reinvk, het zou met een DHZ account ook moeten werken. Ook de software versie van je Envoy ziet er goed uit. Welke gegevens zie je wel?

Hoi Vincent,

De meeste(denk alle) entiteiten van Evoy zie ik wel zoals production, consumption, 3 Phase L1,2,3.

Maar van de Inverter is alleen de Inverter<nr>Production zicht de geleverde hoeveelheid Watt (current).

reinvk, 2023-05-18:
> Hoi Vincent,
>
> De meeste(denk alle) entiteiten van Evoy zie ik wel zoals production, consumption, 3 Phase L1,2,3.
>
> Maar van de Inverter is alleen de Inverter<nr>Production zicht de geleverde [...]
 Dus niet zoals in het voorbeeld zichtbaar was in screenshot hoeveelheid: Ampere, temperatuur van de micro's. Dat kan ik wel via DHZ account in Enligthen zien.

Groeten Rein

reinvk, 2023-05-18:
Hoi Vincent,

De meeste(denk alle) entiteiten van Evoy zie ik wel zoals production, consumption, 3 Phase L1,2,3.

Maar van de Inverter is alleen de Inverter<nr>Production zichtbaar de geleverde hoeveelheid Watt (current).
Dus niet zoals in het voorbeeld zichtbaar was in screenshot hoeveelheid: Ampere, temperatuur, etc. van de micro's. Dat kan ik wel via DHZ account in Enligthen inzien.

Groeten Rein

reinvk, 2023-05-18:
[image]

reinvk, 2023-05-18:
Dan kom ik in mijn Envoy terecht. Als ik daar met mijn credentials aanmeld zie ik diverse gegevens van de Envoy (metingswaarden, settings,etc).
Moet naar iets specifieks kijken of screenshots sturen?

reinvk, 2023-05-18:
Nadat aangelogd ben dan zie ik een output met metrics die nu veelal op 0 staan omdat de  zon onder is, zal dit morgenvroeg herhalen.

[image]

reinvk, 2023-05-19:
Inderdaad nu de PV actief is komen de metrics er wel uit met de URL:

[image]

reinvk, 2023-05-22:
Maar deze output is niet beschikbaar in Home Assistant.

posixx, 2023-05-22:
@reinvk: weet je zeker dat je de juiste integratie draait? Heb je, voordat je deze integratie ging gebruiken, een eerdere integratie geinstalleerd via HACS? In dat geval dien je de integratie te verwijderen, en na verwijderen eerst de map /config/custom_components/enphase_envoy volledig te verwijderen. Daarna installeer je deze integratie opnieuw.

reinvk, 2023-05-22:
> @reinvk: weet je zeker dat je de juiste integratie draait? Heb je, voordat je deze integratie ging gebruiken, een eerdere integratie geinstalleerd via HACS? In dat geval dien je de integratie te [...]

Ja, ik had (Dev)/Posix. En las ergens dat iemand problemen had. Daarom heb ik na het gelezen te hebben alles verwijder incl. folder en daarna opnieuw deze integratie geïnstalleerd. Resultaat bleef gelijk.

reinvk, 2023-05-22:
> Hoi @reinvk, de screenshots zien er goed uit. Kun je misschien een paar minuten debugging aan zetten voor de integratie en mij de output sturen naar dit (tijdelijke) e-mail [ziener06joduur@icloud.com](mailto:ziener06joduur@icloud.com) ?

Ga ik doen en zal resultaat mailen.

reinvk, 2023-05-22:
> @reinvk I changed something in [0.2.0](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.2.0) that might also fix the sensors not showing up for you. Can you try that version?

Na de debug output mail zal ik deze nieuwe versie proberen en resultaat delen.

reinvk, 2023-05-22:
> ziener06joduur@icloud.com

Heb de DEBUG log gemaild met een stukje tekst waarin ik vermeld heb dat er meer zijn die gelijkwaardige problemen ondervinden. Het gaat fout met error 401 waarschijnlijk als gevolg doordat Token wordt ingewisseld voor een SessionId: (cookie). Het werkt wel in een browser maar niet met bijv. curl en in deze integratie.
Heb je daar een idee bij wat het zou kunnen zijn ? (zie mail)

reinvk, 2023-05-22:
> > @reinvk I changed something in [0.2.0](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.2.0) that might also fix the sensors not showing up for you. Can you try that version?
> 
> Na de debug output mail zal ik deze nieuwe versie proberen en resultaat delen.

Heb de nieuwste verse 0.2.0 opnieuw geupdate en opnieuw geinstalleerd door ook de folder in custom te verwijderen en opnieuw aan te maken etc... Maar de meldingen blijven hetzelfde en de extra metrics komen niet beschikbaar.

vincentwolsink, 2023-05-23:
Bedankt voor de logs. De Envoy antwoord inderdaad een 401. Maar dat zou niet moeten. De integratie doet precies wat het document aangeeft. Authenticatie via een `Bearer token` header. Welke firmware versie Envoy heb je?

reinvk, 2023-05-23:
> Bedankt voor de logs. De Envoy antwoord inderdaad een 401. Maar dat zou niet moeten. De integratie doet precies wat het document aangeeft. Authenticatie via een `Authorization: Bearer token` header. Welke firmware versie Envoy heb je?

Envoy firmware is recent versie D7.6.114 om heel precies te zijn: 

SW VERSION D7.6.114.230120 (43769e)

vincentwolsink, 2023-05-24:
Ik zal een support case openen bij Enphase. Eens zien wat zij hierover zeggen.

reinvk, 2023-05-24:
> Ik zal een support case openen bij Enphase. Eens zien wat zij hierover zeggen.

Wat ik gevonden heb is dat er meer gelijkwaardige problemen hebben dat sommige api's wel en niet werken met een error 401.

vincentwolsink, 2023-05-25:
Hoi @reinvk, misschien kun je zelf een support case openen? Ze vragen namelijk om gegevens van de Envoy. En die van mij heeft dit probleem niet.

reinvk, 2023-05-25:
> Hoi @reinvk, misschien kun je zelf een support case openen? Ze vragen namelijk om gegevens van de Envoy. En die van mij heeft dit probleem niet.

Ga het proberen en houd je op de hoogte.

reinvk, 2023-05-26:
Support case aangemeld en contact gehad via chat en mail. Is nu doorgezet naar development.Er is beloofd binnen 48 workinghours een reactie te hebben.

reinvk, 2023-05-26:
> > > @reinvk I changed something in [0.2.0](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.2.0) that might also fix the sensors not showing up for you. Can you try that version?
> > 
> > 
> > Na de debug output mail zal ik deze nieuwe versie proberen en resultaat delen.
> 
> Heb de nieuwste verse 0.2.0 opnieuw geupdate en opnieuw geinstalleerd door ook de folder in custom te verwijderen en opnieuw aan te maken etc... Maar de meldingen blijven hetzelfde en de extra metrics komen niet beschikbaar. Heb dit opnieuw gestest door DEBUG aan te zetten en dan komt de 401 steeds terug. Ook bij het opvragen van: https://{envoy}/ivp/mod/603980032/mode/power krijg ik error 401.
> 
> De Technical brief van januari 2023 zul je vast hebben gelezen (bijgesloten): [https://enphase.com/download/iq-gateway-access-using-local-apis-or-local-ui-token-based-authentication-tech-brief](url)

(26-5-2023 Update Reinvk)
Heb er nu wel een nieuw issue bij gekregen dat de verbinding van de integratie elke keer weg valt en na een tijdje terug komt en dat herhaalt zich diverse keren per dag. De errors die dan in de home-assistant.log verschijnen zijn:

2023-05-26 17:44:24.824 ERROR (MainThread) [custom_components.enphase_envoy] Error fetching envoy Envoy 111111111111 data: Error communicating with API: peer closed connection without sending complete message body (incomplete chunked read)
2023-05-26 18:16:09.801 ERROR (MainThread) [custom_components.enphase_envoy] Error fetching envoy Envoy 111111111111 data: Error communicating with API: peer closed connection without sending complete message body (incomplete chunked read)
2023-05-26 19:05:52.412 ERROR (MainThread) [custom_components.enphase_envoy] Error fetching envoy Envoy 111111111111 data: Error communicating with API: peer closed connection without sending complete message body (incomplete chunked read)
2023-05-26 19:35:18.270 ERROR (MainThread) [custom_components.enphase_envoy] Error fetching envoy Envoy 111111111111 data: Error communicating with API: peer closed connection without sending complete message body (incomplete chunked read)
2023-05-26 21:11:56.176 ERROR (MainThread) [custom_components.enphase_envoy] Error fetching envoy Envoy 111111111111 data: Error communicating with API: peer closed connection without sending complete message body (incomplete chunked read)

Het lijkt opgetreden te zijn sinds het moment dat ik de upgrade van 0.1.9 naar 0.2.0 heb gedaan.
Om dat uit te sluiten ben ik nu net gedowngrade naar 0.1.9 en kijken af het dan wel stabiel blijft.

reinvk, 2023-05-26:
> (26-5-2023 Update Reinvk)
>
> Heb er nu wel een nieuw issue bij gekregen dat de verbinding van de integratie elke keer weg valt en na een tijdje terug komt en dat herhaalt zich diverse keren per dag. [...]

Krijg de errors ook met versie 0.1.9 dus daar ligt het niet aan

mikevankan, 2023-05-28:
Ik heb hetzelfde probleem, informatie welke in jouw screenshots staan mis ik en ook de schakelaar om productie te stoppen mis ik. Kan ik iets doen om jou te helpen dit te troubleshooten?

ortixx, 2023-05-28:
Ik heb hier hetzelfde probleem... De envoy_reader werkt ook gewoon;
192.168.1.10
Reading...
production:              195
consumption:             396
daily_production:        3374
daily_consumption:       2229
seven_days_production:   7127
seven_days_consumption:  2270
lifetime_production:     22617
lifetime_consumption:    6555

die van Brian werkt wel.

 probleem hierboven is verholpen door python3-async*, in de logs van Hass zie ik de data ook voorbij komen, maar met deze extra output nu:

2023-05-28 19:40:04.878 DEBUG (MainThread) [custom_components.enphase_envoy] Finished fetching envoy Envoy [122024035197](tel:122024035197) data in 30.003 seconds (success: False)

en dan:

This error originated from a custom integration.

Logger: homeassistant.config_entries
Source: custom_components/enphase_envoy/envoy_reader.py:175
Integration: Enphase Envoy (Installer) 
First occurred: 7:42:17 PM (1 occurrences) 
Last logged: 7:42:17 PM

Error setting up entry Envoy 122024035197 for enphase_envoy
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/anyio/streams/tls.py", line 130, in _call_sslobject_method
    result = func(*args)
  File "/usr/local/lib/python3.10/ssl.py", line 917, in read
    v = self._sslobj.read(len)
ssl.SSLWantReadError: The operation did not complete (read) (_ssl.c:2578)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/enphase_envoy/__init__.py", line 142, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 226, in async_config_entry_first_refresh
    await self._async_refresh(
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 258, in _async_refresh
    self.data = await self._async_update_data()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 217, in _async_update_data
    return await self.update_method()
  File "/config/custom_components/enphase_envoy/__init__.py", line 54, in async_update_data
    await envoy_reader.getData()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 360, in getData
    await self.detect_model()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 383, in detect_model
    await self._update_from_pc_endpoint()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 134, in _update_from_pc_endpoint
    await self._update_endpoint(
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 158, in _update_endpoint
    response = await self._async_fetch_with_retry(
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 175, in _async_fetch_with_retry
    resp = await client.get(
  File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1757, in get
    return await self.request(
  File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1530, in request
    return await self.send(request, auth=auth, follow_redirects=follow_redirects)
  File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1617, in send
    response = await self._send_handling_auth(
  File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1645, in _send_handling_auth
    response = await self._send_handling_redirects(
  File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1682, in _send_handling_redirects
    response = await self._send_single_request(request)
  File "/usr/local/lib/python3.10/site-packages/httpx/_client.py", line 1719, in _send_single_request
    response = await transport.handle_async_request(request)
  File "/usr/local/lib/python3.10/site-packages/httpx/_transports/default.py", line 353, in handle_async_request
    resp = await self._pool.handle_async_request(req)
  File "/usr/local/lib/python3.10/site-packages/httpcore/_async/connection_pool.py", line 253, in handle_async_request
    raise exc
  File "/usr/local/lib/python3.10/site-packages/httpcore/_async/connection_pool.py", line 237, in handle_async_request
    response = await connection.handle_async_request(request)
  File "/usr/local/lib/python3.10/site-packages/httpcore/_async/connection.py", line 90, in handle_async_request
    return await self._connection.handle_async_request(request)
  File "/usr/local/lib/python3.10/site-packages/httpcore/_async/http11.py", line 116, in handle_async_request
    raise exc
  File "/usr/local/lib/python3.10/site-packages/httpcore/_async/http11.py", line 95, in handle_async_request
    ) = await self._receive_response_headers(**kwargs)
  File "/usr/local/lib/python3.10/site-packages/httpcore/_async/http11.py", line 159, in _receive_response_headers
    event = await self._receive_event(timeout=timeout)
  File "/usr/local/lib/python3.10/site-packages/httpcore/_async/http11.py", line 195, in _receive_event
    data = await self._network_stream.read(
  File "/usr/local/lib/python3.10/site-packages/httpcore/backends/asyncio.py", line 34, in read
    return await self._stream.receive(max_bytes=max_bytes)
  File "/usr/local/lib/python3.10/site-packages/anyio/streams/tls.py", line 195, in receive
    data = await self._call_sslobject_method(self._ssl_object.read, max_bytes)
  File "/usr/local/lib/python3.10/site-packages/anyio/streams/tls.py", line 137, in _call_sslobject_method
    data = await self.transport_stream.receive()
  File "/usr/local/lib/python3.10/site-packages/anyio/_backends/_asyncio.py", line 1265, in receive
    await self._protocol.read_event.wait()
  File "/usr/local/lib/python3.10/asyncio/locks.py", line 214, in wait
    await fut
asyncio.exceptions.CancelledError

Heb de versie van postixx getest, zelfde probleem.

reinvk, 2023-05-29:
> 2023-05-26 18:16:09.801 ERROR (MainThread) [custom_components.enphase_envoy] Error fetching envoy Envoy 111111111111 data: Error communicating with API: peer closed connection without sending complete message body (incomplete chunked read)

Deze melding/error die ik sinds een aantal dagen had nadat ik intensief aan het troubleshooten geweest ben heb ik kunnen oplossen door de Envoy een power-off/power-on te doen waarmee deze opnieuw is opgestart. Sindsdien (27-5) ben ik die meldingen kwijt en verlies de connectie niet meer met de (local) Envoy. Daarmee blijft de oorspronkelijke melding over: dat de een aantal API commando's met een error 401 vanuit de integratie niet werken. Maar ze werken wel als dezelfde API vanuit een browser wordt aangeroepen maar dan moet wel eerst aangemeld worden met de Enlighten-account of het invoeren van een Token in de lokale Envoy voordat er output wordt weergegeven. (Hiervoor staat nog een case open bij Enpase sinds 26-5-2023)

reinvk, 2023-05-29:
> Ik heb hetzelfde probleem, informatie welke in jouw screenshots staan mis ik en ook de schakelaar om productie te stoppen mis ik. Kan ik iets doen om jou te helpen dit te troubleshooten?

Welke versie Firmware in de Envoy heb je ? (D7.x.x )?
Wat die heb ik ook en met de (Installer)-integratie zie ik wel grootste deel van de metrics en de Productie schakelaar was nog aanwezig in versie 0.1.9 en is verdwenen in 0.2.0 inderdaad. Die kan ik zien bij: Instellingen -> Integraties > apparaten -> envoy <serienummer> en krijg dan totaal 31 apparaten en 58  entiteiten. Dus de meeste API's in de integratie werken goed.

De API's die met een Failed eindigen zijn: 
- https://<local ip>/ivp/mod/603980032/mode/power 
- https://<local ip>/ivp/peb/devstatus   

=> Foutmeldingen:  
<Response [401 Unauthorized]>: <html>
<head><title>401 Authorization Required</title></head>
<center><h1>401 Authorization Required</h1></center>

mikevankan, 2023-05-29:
> > Ik heb hetzelfde probleem, informatie welke in jouw screenshots staan mis ik en ook de schakelaar om productie te stoppen mis ik. Kan ik iets doen om jou te helpen dit te troubleshooten?
> 
> Welke versie Firmware in de Envoy heb je ? (D7.x.x )? Wat die heb ik ook en met de (Installer)-integratie zie ik wel grootste deel van de metrics en de Productie schakelaar was nog aanwezig in versie 0.1.9 en is verdwenen in 0.2.0 inderdaad. Die kan ik zien bij: Instellingen -> Integraties > apparaten -> envoy en krijg dan totaal 31 apparaten en 58 entiteiten. Dus de meeste API's in de integratie werken goed.
> 
> De API's die met een Failed eindigen zijn:
> 
> * https:///ivp/mod/603980032/mode/power
> * https:///ivp/peb/devstatus
> 
> => Foutmeldingen: <Response [401 Unauthorized]>:
> 
> <title>401 Authorization Required</title>
> # 401 Authorization Required

Ik zit op versie 7.6.114 met mijn envoy. Met 12 micro omvormers en een envoy zit ik aan 13 devices en 16 entities. Ik mis dus redelijk wat haha.Het account is een DHZ account en indien ik op de PC ingelogd ben geweest of met de token werk dan wordt alle info wel weergegeven en ontvang ik geen 401.

reinvk, 2023-05-29:
> Ik zit op versie 7.6.114 met mijn envoy. Met 12 micro omvormers en een envoy zit ik aan 13 devices en 16 entities. Ik mis dus redelijk wat haha.Het account is een DHZ account en indien ik op de PC [...]

Heb inderdaad andere samenstelling hardware 30 micro's, 1 Envoy, 3 fases zelfde firmware en DHZ account.

reinvk, 2023-06-01:
> Ik heb nog geen reactie van Enphse ontvangen en een status update van de openstaande cases opgevraagd.
> 
> 
> 
> Wel heb ik een mogelijke oplossing gevonden van iemand die dit 3 uur geleden ge-post heeft in deze URL: https://support.enphase.com/s/question/0D53m00008vPvHLCA0/access-iq-gateway-through-curl-commands-with-tokenauthentication-only-allowed-for-some-urls-not-all-urls-giving-401-authorization-required-error-message 
> 
> 
> 
> 
> 
> 
> 
> I finally got this working.
> 
> 
> 
> To summarize:
> 
> 
> 
> the issue I encounter is that some Envoy local paths work using curl (e.g. https://envoy.local/ivp/meters/readings) but some return 401 (e.g. https://envoy.local/production.json)
> 
> the same endpoints that return 401 using `curl` works fine if accessed via browser or session-based connections (python)
> 
>  
> 
> 
> 
> To resolve:
> 
> 
> 
> Assuming JWT token is stored in $jwt_token, validate the token against `auth/check_jwt` and store the generated session cookie: `curl -kiH 'Accept: application/json' -H "Authorization: Bearer $jwt_token" "https://envoy.local/auth/check_jwt" -b cookie -c cookie`
> 
> Use the cookie on thos endpoints that returned a 401. E.g. ` curl -kiH 'Accept: application/json' -H "Authorization: Bearer $jwt_token" "https://envoy.local/production.json" -b cookie -c cookie`
> 
>  
> 
> 
> 
> Alternatively, you can do a verbose curl (`curl -v ...`) and manually capture/copy the cookie from the response header `Set-Cookie` (shows up as `Set-Cookie: sessionId=xxxx`); then manually set it for other urls using `--cookie sessionId=xxxx`. It's actually easier to just use `-b cookie -c cookie` though.
> 
> 
> 
> user='<EMAIL>'
> 
> password='<PASSWORD>'
> 
> envoy_serial='<ENVOY_GATEWAY_SN>'
> 
>  
> 
> jwt_token=$(curl -sX POST "https://entrez.enphaseenergy.com/tokens" -H "Content-Type: application/json" -d "{\"session_id\": \"$session_id\", \"serial_num\": \"$envoy_serial\", \"username\": \"$user\"}")
> 
>  
> 
> curl -k -i -H 'Accept: application/json' -H "Authorization: Bearer $jwt_token" "https://envoy.local/auth/check_jwt" -b cookie -c cookie
> 
>  
> 
> curl -k -H 'Accept: application/json' -H "Authorization: Bearer $jwt_token" "https://envoy.local/production.json?details=1" -b cookie -c cookie

@vincentwolsink Kun je eens naar deze oplossingen kijken of die in de integratie te implementeren is zodat het mogelijk ook gaat werken voor nieuwe firmware versies?

vincentwolsink, 2023-06-01:
Gezien er waarschijnlijk geen korte termijn oplossing komt van Enphase, zal ik kijken naar het implementeren van de cookies. Stay tuned.

vincentwolsink, 2023-06-01:
@reinvk Gezien mijn firmware ook zonder cookie werkt, kun je testen of [0.2.1-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.2.1-beta1) werkt voor jou?

mikevankan, 2023-06-01:
@vincentwolsink Ik heb de beta geinstalleerd, 1 keer als update en 1 keer met het verwijderen van alles voor de installatie maar het werkt helaas nog niet zoals gewenst. 

3-06-01 12:03:51.061 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.5/ivp/peb/devstatus: <Response [401 Unauthorized]>:

vincentwolsink, 2023-06-01:
@mikevankan kun je ook de debug output met de headers etc hier plakken? (De token even inkorten voor de veiligheid)

mikevankan, 2023-06-01:
@vincentwolsink 
2023-06-01 12:03:50.254 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP GET Attempt #1: https://192.168.1.5/ivp/peb/devstatus: Header:{'Authorization': 'Bearer '} Cookies:<Cookies[<Cookie sessionId= for 192.168.1.5 />]>
2023-06-01 12:03:50.429 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Received 401 from Envoy; refreshing token, attempt 1 of 2
2023-06-01 12:03:50.430 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP GET Attempt #1: https://192.168.1.5/auth/check_jwt: Header:{'Authorization': 'Bearer '} Cookies:<Cookies[<Cookie sessionId= for 192.168.1.5 />]>
2023-06-01 12:03:50.586 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.5/auth/check_jwt: <Response [200 OK]>: <!DOCTYPE html><h2>Valid token.</h2>

2023-06-01 12:03:50.588 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Token is valid
2023-06-01 12:03:50.591 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP GET Attempt #2: https://192.168.1.5/ivp/peb/devstatus: Header:{'Authorization': 'Bearer '} Cookies:<Cookies[]>
2023-06-01 12:03:50.733 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Received 401 from Envoy; refreshing token, attempt 2 of 2
2023-06-01 12:03:50.733 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP GET Attempt #1: https://192.168.1.5/auth/check_jwt: Header:{'Authorization': 'Bearer '} Cookies:<Cookies[]>
2023-06-01 12:03:50.892 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.5/auth/check_jwt: <Response [200 OK]>: <!DOCTYPE html><h2>Valid token.</h2>

2023-06-01 12:03:50.895 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Token is valid
2023-06-01 12:03:50.897 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP GET Attempt #3: https://192.168.1.5/ivp/peb/devstatus: Header:{'Authorization': 'Bearer'} Cookies:<Cookies[<Cookie sessionId= for 192.168.1.5 />]>
2023-06-01 12:03:51.061 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.5/ivp/peb/devstatus: <Response [401 Unauthorized]>: <html>
<head><title>401 Authorization Required</title></head>
<body>
<center><h1>401 Authorization Required</h1></center>
<hr><center>Redirecting to <a id='link' href=''></a></center>
<script>
const redirect_url = window.location.protocol + '//' + window.location.host + '/home';
function redirect() { window.location.href = redirect_url; }
window.onload = function() {
    const link = document.getElementById('link');
    link.href = redirect_url;
    link.text = redirect_url;
    setTimeout(redirect, 500);
}
</script>
</body>
</html>
Zonder cookie sessionID en bearer. Hopelijk heb je er wat aan. Als je meer wil mag je mij ook op tweakers benaderen onder dezelfde naam.

vincentwolsink, 2023-06-01:
In [0.2.1-beta2](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.2.1-beta2) heb ik de token code volledig herschreven, precies volgens de technical brief, met als toevoeging de sessionId.

Als het dan nog niet werkt moeten we een andere manier verzinnen om te debuggen.

reinvk, 2023-06-01:
> In [0.2.1-beta2](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.2.1-beta2) heb ik de token code volledig herschreven, precies volgens de technical brief, met [...]

Ik heb de 0.2.1-beta2 ook gestest en geeft dezelfde resultaten, zij het dat de debug logging er wel anders uit ziet natuurlijk maar blijft de 401 error geven op dezelfde API's 

(DEBUG LOGGING)
[20230601 home-assistant_enphase_envoy_2023-06-01T20-09-55.865Z.log](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11629877/20230601.home-assistant_enphase_envoy_2023-06-01T20-09-55.865Z.log)

marco73, 2023-06-08:
Ik heb sinds een week een nieuwe installatie, en krijg via de api ook alleen het actueel opgewekte en hoogst gehaald vermogen uit de inverters. Niet met de laatste stable of de twee laatste beta's.

Als ik naar https://envoy.local/ivp/peb/devstatus ga dan krijg ik ook een 401 te zien en wordt ik teruggestuurd naar /home

Ik heb wel iets aan [documentatie](https://enphase.com/download/iq-gateway-access-using-local-apis-or-local-ui-token-based-authentication-tech-brief ) gevonden, maar daat staat de endpoint /ivp/peb/devstatus niet in.

Misschien dat dit een (nieuwe) eigenschap van een nieuwe firmware?

Envoy:
Firmware: D7.6.114
Hardware: 800-00656-r06 

Microinverters:
Firmware: 521-00005-r05-v06.05.02
Hardware: 800-01359-r02

marco73, 2023-06-08:
@vincentwolsink , ik heb alleen een homeowner account. Geen DHZ/installer toegang. Ik heb wel geprobeerd het installer ww te achterhalen met wat tools maar geen een werkte. Ook een nmap/nessus/vulnerability scanner gaf geen vulnerabilities weer, dus snel toegang via een exploit om de code door te spitten zit er nog niet in.

Ik las in het Enphase topic op tweakers dat je wel een dhz account kan aanvragen/aanmaken op mijn installatie, maar ik heb nog een klein geschil met de installateur dus ik blijf daar nog vanaf.

Als ik op de installer link (onderaan de pagina op de locale webpagina) dan krijg ik ook gelijk de 401 Authorization required melding met een redirect naar Home.

mikevankan, 2023-06-08:
@marco73 je kan via enphase support je homeowner acc naar een dhz laten veranderen. Even aangeven via de chat en dan wordt het omgezet, de installateur blijft aanwezig en je mailadres etc blijft gelijk ;-)

vincentwolsink, 2023-06-08:
@marco73 Deze integratie is bedoeld voor gebruik met een installer/DHZ account. Zonder een dergelijk account zijn inderdaad niet alle gegevens beschikbaar. Dit account kun je niet lokaal exploiten, maar verkrijg je via Enphase Enlighten. Overigens kun je je eigen installer account toevoegen naast dat van je installateur, maar bij een geschil inderdaad misschien maar even wachten.

marco73, 2023-06-08:
@vincentwolsink , @mikevankan , Mooi om te weten dat je een account kan omzetten, al kan heb ik voorlopig genoeg data voor wat grafiekjes. Eerst het geschil maar eens zien op te lossen.

ortixx, 2023-06-08:
> Ik heb zelf een Envoy standard met firmware D7.6.111 (wat de laatste firmware lijkt te zijn voor een standard) en bij mij werkt het foutloos. @marco73 heb jij ook een DHZ account of een installer/dhz? @reinvk heb jij al iets van Enphase terug gehoord?
> 
> @marco73 kun je proberen om eerst als installer in te loggen en dan pas naar https://envoy.local/ivp/peb/devstatus te gaan?

Dit werkt iig bij mij wel. Heb een installer acc/FW D7.6.168 (f1975b)

reinvk, 2023-06-08:
> Ik heb zelf een Envoy standard met firmware D7.6.111 (wat de laatste firmware lijkt te zijn voor een standard) en bij mij werkt het foutloos. @marco73 heb jij ook een DHZ account of een [...]

Vincent, ik ben nog bezig met Enphase, verloopt traag en nog geen oplossing. Ben wel aan het troubleshooten/testen hoe het wel zou kunnen werken. Er zijn 2 API commando's die Error 401 geven terwijl de commandos wel in de firmware zitten omdat ze via een browser URL nadat je inlogt of token geeft een juist resultaat leveren. Dan moet dat toch ook automatisch kunnen werken. Zodra ik meer nieuws heb deel ik het.

ortixx, 2023-06-09:
Nou,

ik heb het werkend gekregen.

pip install awaits en try again. Requirement dus

vincentwolsink, 2023-06-09:
@ortixx Bedoel je dat er een requirement mist voor de integratie? Welke versie draai je? Of heb je het nu over een eigen script?

ortixx, 2023-06-09:
> @ortixx Bedoel je dat er een requirement mist voor de integratie? Welke versie draai je? Want de integratie depend niet op beide packages die je daar noemt. Of heb je het nu over een eigen script?

Requirement voor de plug-in, die heeft “awaits” nodig maar ook async asyncio. Heb het getest met haos default img install en op debian11.

geen eigen script iig. De 2.0 werkt niet.

reinvk, 2023-06-14:
Heb nog geen nieuwe reactie van Enphase op mijn test resultaten die ik gestuurd heb. 
Zag de nieuwe versie 0.2.1 die ik nu geinstalleerd heb maar voor Firmware D7.6.114 helaas nog geen oplossing is. Er blijven errors 401 komen. Het komt er op neer dat bijv. de API https://envoy.local/ivp/peb/devstatus niet via een geautomatiseer command werkt omdat het commando direct als eerst een error 401 terug krijgt.
Als ik de URL in een browser plak dan krijg ik direct een error 401 maar na even wachten (5-15 seconden) komt er een scherm met de mogelijkheid om met een Enlighten account (bijvoorbeeld DHZ account die ik gebruik) kan inloggen of een Token kan submitten. Daarna kom ik in mijn local Envoy terecht. Pas daarna als ik de URL https://envoy.local/ivp/peb/devstatus weer een keer opstart in een andere TAB in de browser krijg ik direct de juiste output zonder foutmeldingen. Dat kan ik bijvoorbeeld herhalen en blijft goed reageren zonder error 401. Na verloop van tijd lijkt de openstaande sessie niet meer actief en begint het opnieuw.
Voor deze handmatige procedure heb ik de procedure en screenshots vastgelegd hoe het precies verloopt.
Mogelijk geeft dat inzicht wat er met hogere Firmware versies anders gaat.

[Envoy error 401 manual steps.docx](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11750859/Envoy.error.401.manual.steps.docx)

Vervolgens heb ik ook een Linux shell-script gemaakt om het vanuit een geautomatiseerde procedure te doen en daarvan is het resultaat vergelijkbaar als de huidige integratie. Dat komt waarschijnlijk doordat ik de handmatige inlog procedure nog niet goed kan nabootsen met mijn shellscript. Als de handmatige procedure geprogrammeerd kan worden zou het moeten werken,
Hierbij deel ik ook het shell-script inclusief de output in onderstaande file.

[enphase script and output.log](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11750880/enphase.script.and.output.log)

reinvk, 2023-06-16:
@vincentwolsink Heb nu getest wat je vroeg:
1) Een token opgehaald en bewaard via https://entrez.enphaseenergy.com/ 
2) de URL: https://envoy.local/ivp/peb/devstatus in de browser gestart
3) krijg dan eerst de error 401 pagina en na aantal seconden verschijn de pagina waar ik kan inloggen of de token invoeren, in dit geval heb ik nu de opgehaalde token ingevuld en Submit gegeven
4) dan krijg ik  de pagina van mijn lokale envoy.local
5) als ik dan in nieuwe tab de URL https://envoy.local/ivp/peb/devstatus open dan gaat het goed. 

Als ik er eenmaal met een token in ben geweest kan ik daarna alle URl's draaien met positief resultaat. Hoe lang weet ik niet maar naar verloop van tijd moet weer een token worden ingevoerd.

Betekend dit dat hiervoor wel kansen zijn, want na de error 401 die altijd eerst lijkt te verschijnen moeten we wachten op de uiteindelijke toegang van de lokale Envoy waar we met een  valide token naar binnen kunnen.

[Envoy error 401 met Token login test.docx](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11776830/Envoy.error.401.met.Token.login.test.docx)

ortixx, 2023-06-16:
Even kleine update mbt 401 op de dev

(MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.10/api/v1/production: <Response [200 OK]>: {
"wattHoursToday": 0,
"wattHoursSevenDays": 32850,
"wattHoursLifetime": 247391,
"wattsNow": 347
}
2023-06-17 01:18:28.424 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP GET Attempt #1: https://192.168.1.10/ivp/peb/devstatus: Header:{'Authorization': 'Bearer
...

2023-06-17 01:18:28.663 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.10/ivp/peb/devstatus: <Response [200 OK]>: { "counters": {"pcu": {"expected": 10,"discovered": 10,"ctrlsTotal": 10,"ctrlsGone": 10

Ik heb een fork gemaakt, eea fouten eruit gehaald.. en geen issues meer. Time-out soms maar is dan direct terug.

vincentwolsink, 2023-06-17:
@ortixx Heb je het gefixt? Dat zou geweldig zijn! Zou je een PR willen openen met je changes?

ortixx, 2023-06-17:
> @ortixx Heb je het gefixt? Dat zou geweldig zijn! Zou je een PR willen openen met je changes?

Ik heb je invited, kan je zelf het inzien :)

ortixx, 2023-06-17:
Ohja, nog iets anders.. ik heb twee accounts. een installer en DHZ (vandaag om laten zetten van user).

De DHZ krijg ik dit:
2023-06-17 22:01:46.214 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.10/ivp/peb/devstatus: <Response [401 Unauthorized]>: <html>
<head><title>401 Authorization Required</title></head>
<body>
<center><h1>401 Authorization Required</h1></center>

Met de installer acc niet! Dus wel degelijk verschil en gelijk de reden waarom het bij mij dus wel werkte geloof ik.

reinvk, 2023-06-17:
@vincentwolsink Op advies van Enphase support heb ik mijn issue doorgestuurd naar Enphase API support dat is een ander team blijkbaar. Benieuwd of ik daar meer gerichte response ga krijgen wat gaat helpen bij een oplossing.
Houdt je op de hoogte.

ortixx, 2023-06-21:
> @vincentwolsink Op advies van Enphase support heb ik mijn issue doorgestuurd naar Enphase API support dat is een ander team blijkbaar. Benieuwd of ik daar meer gerichte response ga krijgen wat gaat helpen bij een oplossing.
> Houdt je op de hoogte.

Zou je het is met de mijne willen proberen? Ik heb de timers anders staan en andere tweaks. Heb m gesynchroniseerd iig en kreeg nu geen foutmeldingen

reinvk, 2023-06-22:
> Zou je het is met de mijne willen proberen? Ik heb de timers anders staan en andere tweaks. Heb m gesynchroniseerd iig en kreeg nu geen foutmeldingen

@ortixx Heb op diverse manier geprobeerd jou integratie te testen maar blijft initialiseren waar het niet doorheen komt. Uiteindelijk eindigt het in foutmeldingen.
Heb een DEBUG-logfile bijgevoegd zodat je kan zien wat er mis gaat. Wel wat gegevens in logfile geanonimiseerd.
Ik heb Firmware D7.6.114 en zie dat er nu D7.6.172 te downloaden is (weet niet of dat uitmaakt) en zag dat je het op een installeraccount met D7.6.168 werkend heb. Hoe kun je een account om (laten) zetten naar installer ? Want een installer aanmaken vraag om KvKnr. etc. en wordt gecontroleerd?

ortixx, 2023-06-22:
> > > @vincentwolsink Op advies van Enphase support heb ik mijn issue doorgestuurd naar Enphase API support dat is een ander team blijkbaar. Benieuwd of ik daar meer gerichte response ga krijgen wat gaat helpen bij een oplossing.
> 
> > > Houdt je op de hoogte.
> 
> > 
> 
> > Zou je het is met de mijne willen proberen? Ik heb de timers anders staan en andere tweaks. Heb m gesynchroniseerd iig en kreeg nu geen foutmeldingen
> 
> 
> 
> @ortixx Heb op diverse manier geprobeerd jou integratie te testen maar blijft initialiseren waar het niet doorheen komt. Uiteindelijk eindigt het in foutmeldingen.
> 
> Heb een DEBUG-logfile bijgevoegd zodat je kan zien wat er mis gaat. Wel wat gegevens in logfile geanonimiseerd.
> 
> Ik heb Firmware D7.6.114 en zie dat er nu D7.6.172 te downloaden is (weet niet of dat uitmaakt) en zag dat je het op een installeraccount met D7.6.168 werkend heb. Hoe kun je een account om (laten) zetten naar installer ? Want een installer aanmaken vraag om KvKnr. etc. en wordt gecontroleerd?
> 
> 
> 
> 
> 
> [home-assistant_enphase_envoy_2023-06-22T20-50-14.416Z.log](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11840542/home-assistant_enphase_envoy_2023-06-22T20-50-14.416Z.log)
> 
> 

Ziet er goed uit, je komt overal doorheen.. moet alleen even naar de Python kijken want hij gaat fout op serial nummer, maar voor de rest komt die erdoorheen!

[21 more replies]

closed 2023-07-07 by vincentwolsink