vincentwolsink/home_assistant_enphase_envoy_installer #16: Missing Lines (phases) for "Current Power Consumption"

malonip, 2023-05-18:
First at all, thanks for this nice Envoy firmware D7.x integration, which is today the only one which works with a 3-phase system.

However, as I remarked some discrepancies between the current consumption value of the integration and the one from my Enlighten application (live status), I had a look at the "Current Power Consumption" sensors and found that I have only the "Current Power Consumption L1" sensor. L2 and L3 sensors are missing.

[image]

Do you have any explanation ? Thanks a lot !

vincentwolsink, 2023-05-18:
Hi @malonip, thanks! There will probably always be some discrepancies since the sensors in this integration are updated every minute, not real-time.

The integration shows only the sensors it can find in the output from the Envoy. It could be the output from your Envoy is slightly different than expected. Can you maybe enable debug logging for a couple of minutes and send the log to this (temporary) e-mail ziener06joduur@icloud.com ? I will then try to find the issue.

malonip, 2023-05-19:
Hi Vincent,
Thanks a lot for your reply. I am sending some log files right now.
As I was charging my EV car yesterday night (no solar production), it was quite easy to see that there were big differences between the current consumption power provided by the Enlighten application and the values in the HA integration (1/3 of the expected power). I am sending the screenshots in my e-mail.
Thanks a lot for your troubleshooting.

vincentwolsink, 2023-05-22:
Hi @malonip,
Looking at the log file you provided, the Envoy actually outputs data for only one phase on consumption. Do you have a multiple Envoy setup?  Do the daily/total energy consumption values match with your Enphase app?

malonip, 2023-05-22:
Hi @vincentwolsink,

Thanks for having a look.

I did some checks by connecting to the web app of the Envoy (envoy.local) and the values provided by the gateway (= the ones that are provided to HA integration) are completely wrong and different from what is displayed in my Enlighten app (where the values are fully correct). See PDF attached. 

[Consumption differences between Enlighten App and local web app or Home Assistant Envoy integration.pdf](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11528328/Consumption.differences.between.Enlighten.App.and.local.web.app.or.Home.Assistant.Envoy.integration.pdf)

I've just written to the Enphase support to see if it is a wrong setting in the Envoy or a software bug. I will let you know as soon as I get an answer.

malonip, 2023-05-25:
Hi @vincentwolsink,
I got an answer from the Enphase support team. They recognized that the consumption measurement had only been activated on one phase. They have now activated the measurement on all phases and now the 3 phases appear in the Envoy integration:

[image]

Problem solved !

Thanks anyway for your help and thanks again for your integration !

vincentwolsink, 2023-05-25:
Great! Glad that they could fix it this easily 😃

closed 2023-05-25 by vincentwolsink