vincentwolsink/home_assistant_enphase_envoy_installer #20: [BUG] Production switch not working

Neowam, 2023-07-02:
**Describe the bug**
Today it was a negative price in the Netherlands, so I wanted to switch off the production. Unfortunately the switch was going on again.

**Account type**
- [X] Installer

**Envoy**
- [ ] Metered
- [ ] Standard

FW version:D7.0.88
Amount of micro inverters connected: 15

**Home Assistant**
Version: 2023.6.3

**Additional context**
```
Relevant snippet of Home Assistant error log. 
```

vincentwolsink, 2023-07-02:
Hi @Neowam, try to restart the Envoy and switch again. If it doesn’t work please check the Home Assistant error log for related error messages.

Neowam, 2023-07-02:
Hi @vincentwolsink 

This happens after restarting my Home Assistant and Enphase installation

[image]

It's not more available for some kind of reason. Due firmware upgrade maybe on the Enphase side?

evanlaar01, 2023-07-03:
Same issue. Only the Production is available.

[image]

vincentwolsink, 2023-07-03:
@evanlaar01 Since the sensors are there, were they available before? What changed? Which firmware are you on?

evanlaar01, 2023-07-03:
> @evanlaar01 Since the sensors are there, were they available before? What changed? Which firmware are you on?

Sensors worked until July 1st about 23:22 I can see in the log from a sensor.
Home Assistant 2023.6.3
Supervisor 2023.06.4
Operating System 10.3
Frontend 20230608.0 - latest

Envoy-S-Metered-EU
Softwareversie 7.6.175

In my opinion, nothing changed in Home Assistant, no restart or something else.

[image]

vincentwolsink, 2023-07-03:
Try restarting your Envoy. If it doesn't work please enable debug logging for the integration and send me the output @evanlaar01 @Neowam

evanlaar01, 2023-07-03:
> Try restarting your Envoy. If it doesn't work please enable debug logging for the integration and send me the output @evanlaar01 @Neowam
I've restarted the integration with debug logging. No difference.
I've removed the serial and the token in the text file.
Thanks for helping

[DebuglogEnphase.txt](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11934447/DebuglogEnphase.txt)

vincentwolsink, 2023-07-03:
@evanlaar01 It seems an supposedly authentication bug present in earlier firmwares now made it to the newest Envoy firmware. You did restart the Envoy device itself too right?

Do you have a true installer account or a DIY/DHZ one?

evanlaar01, 2023-07-03:
> @evanlaar01 It seems an supposedly authentication bug present in earlier firmwares now made it to the newest Envoy firmware. You did restart the Envoy device itself too right?
> 
> Do you have a true installer account or a DIY/DHZ one?

Yesterday I've restarted the whole device, power off the gateway, all leds down, wait for a while and power on again. 
The debug log is only from reload the integration, I'm not at home so I've remotely reload the integration with debug log enabled.
I'm not sure but I think I've a DIY/DHZ account, not a true installer.

Neowam, 2023-07-04:
This is my log after reloading the HA Add-On:

2023-07-02 17:10:15.000 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration enphase_envoy which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-07-02 17:10:15.000 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration entsoe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-07-02 17:10:15.000 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration localtuya which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-07-02 17:10:15.001 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-07-02 17:10:15.001 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration sonoff which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-07-02 17:10:23.965 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dynamische_prijzen_current_electricity_market_price (<class 'custom_components.entsoe.sensor.EntsoeSensor'>) is using state class 'measurement' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-07-02 17:10:23.968 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dynamische_prijzen_next_hour_electricity_market_price (<class 'custom_components.entsoe.sensor.EntsoeSensor'>) is using state class 'measurement' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-07-02 17:10:23.970 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dynamische_prijzen_lowest_energy_price_today (<class 'custom_components.entsoe.sensor.EntsoeSensor'>) is using state class 'measurement' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-07-02 17:10:23.971 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dynamische_prijzen_highest_energy_price_today (<class 'custom_components.entsoe.sensor.EntsoeSensor'>) is using state class 'measurement' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-07-02 17:10:23.977 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dynamische_prijzen_average_electricity_price_today (<class 'custom_components.entsoe.sensor.EntsoeSensor'>) is using state class 'measurement' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-07-02 17:10:23.980 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dynamische_prijzen_current_percentage_of_highest_electricity_price_today (<class 'custom_components.entsoe.sensor.EntsoeSensor'>) is using state class 'measurement' which is impossible considering device class ('monetary') it is using; expected None or one of 'total'; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-07-02 17:10:23.982 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dynamische_prijzen_time_of_highest_price_today (<class 'custom_components.entsoe.sensor.EntsoeSensor'>) is using state class 'measurement' which is impossible considering device class ('timestamp') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-07-02 17:10:23.984 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.dynamische_prijzen_time_of_lowest_price_today (<class 'custom_components.entsoe.sensor.EntsoeSensor'>) is using state class 'measurement' which is impossible considering device class ('timestamp') it is using; expected None; Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
2023-07-02 19:04:31.721 ERROR (MainThread) [custom_components.enphase_envoy] Error fetching envoy Envoy 122227045337 data: Error communicating with API: All connection attempts failed
2023-07-03 02:32:19.473 WARNING (Thread-8) [pychromecast.socket_client] [Bedroom speaker 2(192.168.178.50):8009] Heartbeat timeout, resetting connection
[... cut ...] 688413979', 'admin_state': 1, 'dev_type': 1, 'created_date': '1660907754', 'img_load_date': '1613405093', 'img_pnum_running': '520-00082-r01-v04.27.04', 'ptpn': '540-00135-r01-v04.27.10', 'chaneid': 1627393809, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': True, 'operating': False}}}
2023-07-04 04:14:43.948 DEBUG (MainThread) [custom_components.enphase_envoy] Finished fetching envoy Envoy 122227045337 data in 13.433 seconds (success: True)

luuksommers, 2023-07-04:
For me a new FW version rolled out on sunday evening:

Software Version
D7.6.175 (f79c8d)
Software Build Date
22 Jun, 2023 12:43 PM 

Previously I was able to go to the web interface and get into the installer page, but not I get an unauthorized message. Even the web interface of the envoy is broken with this update:

[image]

vincentwolsink, 2023-07-04:
Thanks to @mnederlof there is a possible fix. Please try [0.2.2-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.2.2-beta1)

Neowam, 2023-07-05:
> <vincentwolsink/home_assistant_enphase_envoy_installer/issues/20/1620592677
> @github.com>

I have installed the latest bèta but it still does not work. Production
Switch is still greyed out (entiteit niet beschikbaar)

Groet! Werner
06-40337180

vincentwolsink, 2023-07-05:
Hi @Neowam, is it just the switch that is not working or also most of the other sensors for the inverters?

Neowam, 2023-07-05:
A lot does not available from now on:

[image]

[image]

[image]

vincentwolsink, 2023-07-05:
Can you please completely uninstall the integration, reinstall and see if it works. If not, please enable debug mode and send the logs (as an attachment). Thanks.

Neowam, 2023-07-05:
I re-instaled, only have the basis functionality back, all other sensor information is gone:

[image]

Inverter (had many more before):
[image]

[image]

Neowam, 2023-07-05:
[home-assistant_enphase_envoy_2023-07-05T15-43-24.598Z.log](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11959727/home-assistant_enphase_envoy_2023-07-05T15-43-24.598Z.log)

vincentwolsink, 2023-07-05:
@Neowam You have to change your password. It was in plaintext in the log unfortunatly. The log with the username/password redacted: 
[home-assistant_enphase_envoy_2023-07-05T15-43-24.598Z.log](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/11959935/home-assistant_enphase_envoy_2023-07-05T15-43-24.598Z.log)

@mnederlof It seems to follow your new flow, but still returns a 401?

Neowam, 2023-07-05:
Thank you Vincent! Weird that my complete mail and password are in the logging file.

mnederlof, 2023-07-05:
> @mnederlof It seems to follow your new flow, but still returns a 401?

The big question i think it, are you able to toggle the "Production Power Enabled" switch in the envoy-ui?
You can find it in the homepage of the Installer Login

https://envoy.local/installer/setup/home
[image]

If you don't have access using the envoy ui, you won't have access using the HA plugin. (and need to contact Enphase support about your account (and which permissions it has)

mnederlof, 2023-07-05:
it also explains why you don't have any other metrics; you need to use the installer or DHZ/DIY account

Neowam, 2023-07-05:
How do I get this installer / DHZ/DIY account?

vincentwolsink, 2023-07-05:
@Neowam Don't forget to click the Installer login button at the bottom of the page after logging in. Because the login will redirect you to the default homepage. 

[image]

Neowam, 2023-07-05:
Then I get this error:

401 Authorization Required

Redirecting to https://envoy.local/home

vincentwolsink, 2023-07-05:
Ok, then it seems you somehow lost your DHZ/installer access 🤷‍♂️

Neowam, 2023-07-05:
> Ok, then it seems you somehow lost your DHZ/installer access 🤷‍♂️

Indeed! Some weird problem.

Neowam, 2023-07-05:
Got this info from support

For installer log in access, you have to complete an upgradation program. I can share the procedure for that. 

Then I get access to the installer login page.

Neowam, 2023-07-05:
To upgrade my Enlighten account to Installer access costs me 9,95 per month! or Life time for 249 EUR.

Did you both this too? I have now MyEnphase. It will be then upgraded to Enlighten Manager.

vincentwolsink, 2023-07-05:
What?!. I have it for free, but I found a support page:
https://support.enphase.com/s/article/Upgrade-for-per-module-power-playback-and-alerts-through-Enlighten-Manager

mnederlof, 2023-07-05:
No i did not do this. i just signed up as do it yourself user option on https://enlighten.enphaseenergy.com/manager/registration

vincentwolsink, 2023-07-05:
Probably they are misunderstanding your request @Neowam. I know a lot of people got their homeowner account upgraded to DIY/DHZ via support for free.

vincentwolsink, 2023-07-05:
@luuksommers Did you also lose your DIY access on the Envoy itself?

Neowam, 2023-07-05:
I just got have a long chat with support, I hope finally understanding my problem. Will give you an update soon with their answer. They are investigation now the problem why my access is gone since last sunday.

luuksommers, 2023-07-05:
> @luuksommers Did you also lose your DIY access on the Envoy itself?

Yes, I'm only a home-owner, never had an Installer / DHZ account, but I has access to the Installer pages. I think there was a bug in the envoy firmware which was patched sunday evening with v7.6.175. I also had a chat with envoy, with the same reaction as @Neowam, allthough I don't think they will "fix" it.

luuksommers, 2023-07-05:
Called with support, they will get back to me within 1 or 2 days. I told them I had access to the installer pages, but since the update I haven't, so I requested DIY account, but he needed approval from someone.

bes-r, 2023-07-05:
Hi,

I was a kind of busy and didn't check HA. Checking the status of my system. They gave me also an update to 7.6.175.

My system has the same issues. I'm short in time this period, but will also try to contact Enphase.

catsmanac, 2023-07-06:
Same here, got D7.6.175 overnight. Before all devstatus was showing, now not anymore. Just have homeowner account. 

It was using stored token, Got 401 on first production.json access, did the session cookie capture and continued. When getting devstatus it reported 401 again, tried cookies refresh twice, then fetched enphase token with local secret. Then however it reported exact same expiration time for the token as the one used before. Next it got 401 again on devstatus. Yep I have owner and not DIY account.

fnxpt, 2023-07-11:
Not sure if it happens to someone else, whenever I disable the power and then try to reconnect the inverts don't start producing energy
[image]

evanlaar01, 2023-07-12:
> Not sure if it happens to someone else, whenever I disable the power and then try to reconnect the inverts don't start producing energy [image]

It can take 30 minutes before it producing power again after the switch.

closed 2023-07-13 by vincentwolsink