vincentwolsink/home_assistant_enphase_envoy_installer #32: Wrong sensor information

cmderks93, 2023-07-08:
Hi,

I have installed your software, great!
I’m having a problem with the lifetime sensor.
I have the metered version, and have used the ct for a couple days, but they won’t fit in our electrical panel. 
so don’t use them anymore.
The lifetime sensor uses the wrong information from the envoy now.

cmderks93, 2023-07-08:
I’ve found this in production.json
{"production":

[{"type": "inverters", "activeCount":12,"reading ime":1688826364, "wNow": 1463, "whLifetime":111994 3),

{"type":"eim", "activeCount": 0, "measurementType :"production", "readingTime":1688826378, "wNow":1 0.572, "whLifetime":584366.867, "varhLeadLifetime ":0.971, "varhLagLifetime: 232755.302, "vahLifeti me":790256.454, "rmsCurrent":0.586,"rms Voltage": 724.678, "reactPwr":-0.0,"apprntPwr":141.481,"p rFactor" :0.04, "whToday":584366.867, "whLastSever Days" :584366.867, "vahToday": 790256.454, "varhLea; dToday":0.971

The data under “eim” is used in my case, but I need the data under “inverters”
I think haha
I’ve produced 1.12 Mwh in total

vincentwolsink, 2023-07-09:
Hi @cmderks93. If you don't use the CTs anymore they should be disabled in the Envoy configuration. Once they are the integration should show the correct values.

cmderks93, 2023-07-09:
Hi,
They are disabled. That’s strange…

vincentwolsink, 2023-07-09:
Is the value at `https://envoy.local/api/v1/production` correct?

cmderks93, 2023-07-09:
No it’s not
It’s also the value of the ct instead of the inverters.
My envoy is updated from 5 to 7 yesterday

vincentwolsink, 2023-07-09:
Does https://envoy.local/ show the following?

[image]

cmderks93, 2023-07-09:
Yes it is 
Just as before the update

cmderks93, 2023-07-09:
I’ve turned on and off the ct metering again, but still the wrong information.

catsmanac, 2023-07-09:
I wonder if this is a 'feature' of the new D7.6.x firmware. I see some recent reports around this in the @briancmpbll integration as well that started with updates to d7.6.175. I only have ENVOY-S EU standard so no change here. 

If you can share output of v1/production and /production.json we can compare to older examples around to confirm.

cmderks93, 2023-07-09:
Actually I don't have the 7.6 firmware, im running the 7.4 firmware. 

{
  "wattHoursToday": 0,
  "wattHoursSevenDays": 0,
  "wattHoursLifetime": 584365,
  "wattsNow": 0
}

and

{
  "production": [
    {
      "type": "inverters",
      "activeCount": 12,
      "readingTime": 1688915437,
      "wNow": 162,
      "whLifetime": 1132099
    },
    {
      "type": "eim",
      "activeCount": 0,
      "measurementType": "production",
      "readingTime": 1688915457,
      "wNow": 10.692,
      "whLifetime": 584367.664,
      "varhLeadLifetime": 0.981,
      "varhLagLifetime": 232755.336,
      "vahLifetime": 790267.262,
      "rmsCurrent": 0.589,
      "rmsVoltage": 672.71,
      "reactPwr": 0.457,
      "apprntPwr": 131.899,
      "pwrFactor": 0.08,
      "whToday": 584367.664,
      "whLastSevenDays": 584367.664,
      "vahToday": 790267.262,
      "varhLeadToday": 0.981,
      "varhLagToday": 232755.336
    }
  ],
  "consumption": [
    {
      "type": "eim",
      "activeCount": 0,
      "measurementType": "total-consumption",
      "readingTime": 1688915457,
      "wNow": 10.692,
      "whLifetime": -584366.85,
      "varhLeadLifetime": 0.981,
      "varhLagLifetime": 232755.336,
      "vahLifetime": 0.041,
      "rmsCurrent": 0.589,
      "rmsVoltage": 446.382,
      "reactPwr": -0.457,
      "apprntPwr": 262.96,
      "pwrFactor": 0.04,
      "whToday": 0.0,
      "whLastSevenDays": 0.0,
      "vahToday": 0.041,
      "varhLeadToday": 0.981,
      "varhLagToday": 232755.336
    },
    {
      "type": "eim",
      "activeCount": 0,
      "measurementType": "net-consumption",
      "readingTime": 1688915457,
      "wNow": -0.0,
      "whLifetime": 0.0,
      "varhLeadLifetime": 0.0,
      "varhLagLifetime": 0.0,
      "vahLifetime": 0.041,
      "rmsCurrent": -0.0,
      "rmsVoltage": 220.054,
      "reactPwr": -0.0,
      "apprntPwr": 0.0,
      "pwrFactor": -1.0,
      "whToday": 0,
      "whLastSevenDays": 0,
      "vahToday": 0,
      "varhLeadToday": 0,
      "varhLagToday": 0
    }
  ],
  "storage": [
    {
      "type": "acb",
      "activeCount": 0,
      "readingTime": 0,
      "wNow": 0,
      "whNow": 0,
      "state": "idle"
    }
  ]
}

vincentwolsink, 2023-07-09:
For my Envoy Standard the `whLifetime` under `production > inverters` is actually a completely wrong value. Can you maybe check yours @catsmanac?

We could switch reading to this value for Metered Envoys without installed CTs. And keep the current one for Envoy Standard. But then we need to make sure this value is always correct for the metered ones.

cmderks93, 2023-07-09:
This value is also not correct for me....

cmderks93, 2023-07-09:
Looks like there are more problems with Enphase at the moment. Also the smart energy meter from homewizard is getting wrong values. So maybe a bug with the firmware?

catsmanac, 2023-07-09:
Many jumped from 7.0.88 to 7.6.x so this issue may very well have been introduced earlier then 7.6

The ENVOY-S EU Standard I have should use the v1/production values. Only v1/production wattsNow matches production.json inverters wNow. (I'm integrating all individual panel lastreportWatts from v1/production/inverters in HA and that number is is only 50Kwh off from the envoy lifetime value 5090Kwh since mid February) 

v1/production
```
}
  "wattHoursToday": 30937,
  "wattHoursSevenDays": 342385,
  "wattHoursLifetime": 5089571,
  "wattsNow": 1068
}
```
production.json
```
{
	"production": [
		{
			"type": "inverters",
			"activeCount": 24,
			"readingTime": 1688923491,
			"wNow": 1068,
			"whLifetime": 317385
		}
	],
	"storage": [
		{
			"type": "acb",
			"activeCount": 0,
			"readingTime": 0,
			"wNow": 0,
			"whNow": 0,
			"state": "idle"
		}
	]
}
```
Well if enphase website is not getting the right numbers then they are as confused as we are.

catsmanac, 2023-07-09:
> Looks like there are more problems with Enphase at the moment. Also the smart energy meter from homewizard is getting wrong values. So maybe a bug with the firmware? Also the values mentioned in this thread, are used in the enphase manager. So according to the enphase website, there is no production today, or the last 7 days for me.

Has yours been working well before and this started with new firmware you got in the envoy? Or is it a new installation?

cmderks93, 2023-07-09:
Yes
With version 5 it worked fine
With the update to 7 it did go wrong.

catsmanac, 2023-07-09:
Ok, so you went from 5.x to 7.4?  You mentioned you had the ct's connected for a while, where the numbers correct then?

cmderks93, 2023-07-09:
Yes that’s correct. 
I’ve used the ct 2 years ago, I cannot remember what the sensors did then.
I’ve removed the cts because of the size.

vincentwolsink, 2023-07-09:
Looking at all the bug reports at similar Enphase integrations, it seems with the new firmware we need to get the lifetime production from production.json in the case of and Envoy Metered without CTs.

For Envoy standard we keep using the old API endpoint.

catsmanac, 2023-07-09:
Yes but it will break envoy that haven't updated yet. You may want to mention that specifically.

vincentwolsink, 2023-07-09:
I'm not sure about that. I havent seen any example yet where an Envoy metered without CTs had wrong numbers in `production.json` for the inverters. Please let me know if you did.

catsmanac, 2023-07-09:
I've asked the question in Brian Campbells one. [This one](https://github.com/briancmpbll/home_assistant_custom_envoy/issues/88#issuecomment-1626990268) made me think. 

But looking at the code I wonder if our plain vanilla ENVOY-S now has an issue?

catsmanac, 2023-07-10:
You are right, all reports are indeed for ENVOY-S metered without CT's. And the issue was in 7.0.88 present already. The logic to deal with the watts was there long time, but lifetime has been missed for using same method.

PR #33 (2023-07-09): If Envoy metered without CTs get lifetime_production from production.json
Should fix #32
closed 2023-07-10 by vincentwolsink