vincentwolsink/home_assistant_enphase_envoy_installer #36: [BUG]

Battiegoal, 2023-07-11:
**Describe the bug**
Error code in Home Assistant when trying to connect: Unknown error occurred.
Tried it both for Installer and Owner account

**Account type**
- [X] Installer
- [ ] DIY / DHZ
- [X] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [X] Metered without CTs
- [ ] Standard

- FW version: D7.6.175
- Amount of micro inverters connected: 16

**Home Assistant**
- Version: 2023.7.1

**Additional context**

Relevant snippet of Home Assistant error log. 
- output of /api/v1/production:
{
  "wattHoursToday": 0,
  "wattHoursSevenDays": 0,
  "wattHoursLifetime": 0,
  "wattsNow": 0
}

- output of production.json:
- {"production":[{"type":"inverters","activeCount":16,"readingTime":1689100170,"wNow":706,"whLifetime":895407},{"type":"eim","activeCount":0,"measurementType":"production","readingTime":1689100205,"wNow":-1.991,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":0.484,"rmsVoltage":251.297,"reactPwr":0.0,"apprntPwr":4.919,"pwrFactor":-0.51,"whToday":0.0,"whLastSevenDays":0.0,"vahToday":0.0,"varhLeadToday":0.0,"varhLagToday":0.0}],"consumption":[{"type":"eim","activeCount":0,"measurementType":"total-consumption","readingTime":1689100205,"wNow":-1.891,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":0.793,"rmsVoltage":242.041,"reactPwr":0.0,"apprntPwr":4.282,"pwrFactor":-0.58,"whToday":0.0,"whLastSevenDays":0.0,"vahToday":0.0,"varhLeadToday":0.0,"varhLagToday":0.0},{"type":"eim","activeCount":0,"measurementType":"net-consumption","readingTime":1689100205,"wNow":0.1,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":0.309,"rmsVoltage":242.041,"reactPwr":0.0,"apprntPwr":1.634,"pwrFactor":0.0,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

vincentwolsink, 2023-07-11:
Apart from the "Unknown Error" we have another serious issue here. `wattHoursToday` is 0, and the inverters block `production.json` does not contain `whToday`. Since you don't have CTs we cannot use the value under `eim`, so basically, there is no way to get this measurement...

catsmanac, 2023-07-11:
That's typical with ENVOY metered without CT. Only wNow and whLifetime in production[0] are available. numbers in production[1] are all over the place. Part of many discussions in other threats and where the test for ismetered is in the code.

Unknown error occurs often during first install getting token and enphase website isn't reacting. Does this happen during initial configuration? If so keep on trying is most found solution.

vincentwolsink, 2023-07-13:
@Battiegoal Can you check if the latest beta fixes your "Unknown Error", and if not, enable debug logging in Home Assistant and check if there is an error in the logs while setting up the integration.

```
# configuration.yaml
logger:
  default: debug
```

Battiegoal, 2023-07-14:
Latest beta did not fix the issue. I was able to login with my owner account (one lucky shot, did not manage to replicate this afterwards again). Thx again for putting effort in this.

Log from when I try to login with installer account:

Logger: homeassistant.config_entries
Source: custom_components/enphase_envoy/envoy_reader.py:281
Integration: Enphase Envoy (Installer) ([documentation](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/), [issues](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/issues))
First occurred: 10:12:30 (1 occurrences)
Last logged: 10:12:30

Error setting up entry Envoy XXXXXXXXXXX for enphase_envoy
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 131, in _call_sslobject_method
    result = func(*args)
             ^^^^^^^^^^^
  File "/usr/local/lib/python3.11/ssl.py", line 921, in read
    v = self._sslobj.read(len)
        ^^^^^^^^^^^^^^^^^^^^^^
ssl.SSLWantReadError: The operation did not complete (read) (_ssl.c:2576)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 390, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 191, in async_setup_entry
    await coordinator.async_config_entry_first_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 251, in async_config_entry_first_refresh
    await self._async_refresh(
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 242, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 83, in async_update_data
    await envoy_reader.getData()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 608, in getData
    await self.detect_model()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 631, in detect_model
    await self._update_from_pc_endpoint()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 226, in _update_from_pc_endpoint
    await self._update_endpoint(
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 262, in _update_endpoint
    response = await self._async_fetch_with_retry(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 281, in _async_fetch_with_retry
    resp = await client.get(
           ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpx/_client.py", line 1757, in get
    return await self.request(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpx/_client.py", line 1530, in request
    return await self.send(request, auth=auth, follow_redirects=follow_redirects)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpx/_client.py", line 1617, in send
    response = await self._send_handling_auth(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpx/_client.py", line 1645, in _send_handling_auth
    response = await self._send_handling_redirects(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpx/_client.py", line 1682, in _send_handling_redirects
    response = await self._send_single_request(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpx/_client.py", line 1719, in _send_single_request
    response = await transport.handle_async_request(request)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 353, in handle_async_request
    resp = await self._pool.handle_async_request(req)
[... 9 lines cut ...]
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 120, in handle_async_request
    raise exc
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 99, in handle_async_request
    ) = await self._receive_response_headers(**kwargs)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 163, in _receive_response_headers
    event = await self._receive_event(timeout=timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 199, in _receive_event
    data = await self._network_stream.read(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpcore/backends/asyncio.py", line 34, in read
    return await self._stream.receive(max_bytes=max_bytes)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 196, in receive
[... cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 138, in _call_sslobject_method
    data = await self.transport_stream.receive()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 1203, in receive
    await self._protocol.read_event.wait()
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

vincentwolsink, 2023-07-16:
This looks like a communication issue between Home Assistant and the local Envoy. Can you try to restart your Envoy?

Battiegoal, 2023-07-17:
I did this. Still no result. I can connect via my home owner account, so envoy is up and running. Also via internal ip I can connect.
Log that I have when trying again:

This error originated from a custom integration.

Logger: custom_components.enphase_envoy
Source: custom_components/enphase_envoy/envoy_reader.py:673
Integration: Enphase Envoy (Installer) (documentation, issues) 
First occurred: 14:23:16 (1 occurrences) 
Last logged: 14:23:16

Unexpected error fetching envoy Envoy 122207090169 data: Could not connect or determine Envoy model. Check that the device is up at 'https://192.168.2.28'.
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 242, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 83, in async_update_data
    await envoy_reader.getData()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 608, in getData
    await self.detect_model()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 673, in detect_model
    raise RuntimeError(
RuntimeError: Could not connect or determine Envoy model. Check that the device is up at 'https://192.168.2.28'.

mnederlof, 2023-07-17:
Can you please post the contents of :
- https://192.168.2.28/info.xml 
and (after login in envoy ui)
- https://192.168.2.28/api/v1/production
- https://192.168.2.28/production.json?details=1

Battiegoal, 2023-07-25:
Of course, sorry for the delay (holidays...)
I cannot login on my installer account anymore, the token via entrez.enphaseenergy.com gives the error 'Bad token'

info.xml:

This XML file does not appear to have any style information associated with it. The document tree is shown below.
<envoy_info>
<time>1690309296</time>
<device>
<sn>122207090169</sn>
<pn>800-00654-r07</pn>
<software>D7.6.175</software>
<euaid>4c8675</euaid>
<seqnum>0</seqnum>
<apiver>1</apiver>
<imeter>true</imeter>
</device>
<web-tokens>true</web-tokens>
<package name="rootfs">
<pn>500-00001-r01</pn>
<version>02.00.00</version>
<build>1210</build>
</package>
<package name="kernel">
<pn>500-00011-r02</pn>
<version>04.04.225</version>
<build>3eb4d3</build>
</package>
<package name="boot">
<pn>590-00019-r01</pn>
<version>02.00.01</version>
<build>1f421b</build>
</package>
<package name="app">
<pn>500-00002-r01</pn>
<version>07.06.175</version>
<build>f79c8d</build>
</package>
<package name="devimg">
<pn>500-00005-r01</pn>
<version>01.02.346</version>
<build>b1b724</build>
</package>
<package name="geo">
<pn>500-00008-r01</pn>
<version>02.01.24</version>
<build>a74d96</build>
</package>
<package name="backbone">
<pn>500-00010-r01</pn>
<version>07.00.20</version>
<build>176d57</build>
</package>
<package name="meter">
<pn>500-00013-r01</pn>
<version>03.02.08</version>
<build>eaa252</build>
</package>
<package name="agf">
<pn>500-00012-r01</pn>
<version>02.02.00</version>
<build>40061a</build>
</package>
<package name="security">
<pn>500-00016-r01</pn>
<version>02.00.00</version>
<build>54a6dc</build>
</package>
<package name="full">
<pn>500-00001-r01</pn>
<version>02.00.00</version>
<build>1210</build>
</package>
<build_info>
<build_id>ec2-user-envoy_uber-pkg_master:pkg-Jun-22-23-18:55:22</build_id>
<build_time_gmt>1687460237</build_time_gmt>
<release_ver>02.00.4238</release_ver>
<release_stage>700-GA</release_stage>
</build_info>
</envoy_info>

Battiegoal, 2023-07-28:
UPDATE above :)

mnederlof, 2023-07-31:
Well that information is all correct, maybe you just entered the wrong enlighten credentials during setup of the integration?

As detect_model is basically fetching those three URL's and they seem to be fine, it should select EnvoyMetered as model.

Could you retry using the latest version? (and also remove the integration from HA, download latest version from HACS, restart HA and re-add it, if you've not already removed it)

Battiegoal, 2023-07-31:
Removed the HACS integration from HA, restarted, re-installed the integration and added it. Used the right credentials (email and password only) and it created a configuration. But in the integration it keeps saying 'installation failed' and starts again to 'initializing'...

vincentwolsink, 2023-07-31:
Since everything seems ok when you download the files from your browser, could it be there is a firewall in between the Home Assistant host and the Envoy?

Battiegoal, 2023-08-01:
Not that I am aware of. My installation is a modem from KPN with standard settings and a Home Assistant running on a RPi which also runs AdGuard Home. Disabling this last one did not help either.

Battiegoal, 2023-08-04:
Just installed latest version (3.3.0) and it works now 👍🏼

closed 2023-08-07 by vincentwolsink