vincentwolsink/home_assistant_enphase_envoy_installer #40: Entity Todays power production stays at 0 since last Envoy firmware update

joverdijk, 2023-07-13:
Entity Todays power production stays at 0.

Envoy S metered???? but without CTs Softwareversie D7.6.175 (f79c8d)

Problem since last firmware update. with

https://192.168.xx.xx/production.json

{"production":[{"type":"inverters","activeCount":10,"readingTime":1689099613,"wNow":101,"whLifetime":646210},{"type":"eim","activeCount":0,"measurementType":"production","readingTime":1689099624,"wNow":-0.709,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.04,"rmsCurrent":0.319,"rmsVoltage":236.195,"reactPwr":0.853,"apprntPwr":75.334,"pwrFactor":0.0,"whToday":0.0,"whLastSevenDays":0.0,"vahToday":0.04,"varhLeadToday":0.0,"varhLagToday":0.0}],"consumption":[{"type":"eim","activeCount":0,"measurementType":"total-consumption","readingTime":1689099624,"wNow":-1.336,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":0.404,"rmsVoltage":251.885,"reactPwr":1.399,"apprntPwr":101.217,"pwrFactor":-0.01,"whToday":0.0,"whLastSevenDays":0.0,"vahToday":0.0,"varhLeadToday":0.0,"varhLagToday":0.0},{"type":"eim","activeCount":0,"measurementType":"net-consumption","readingTime":1689099624,"wNow":-0.454,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":-0.045,"rmsVoltage":251.885,"reactPwr":0.546,"apprntPwr":24.581,"pwrFactor":-0.06,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

lots of info, but the v1 api always gives 0

https://192.168.xx.xx/api/v1/production

{
  "wattHoursToday": 0,
  "wattHoursSevenDays": 0,
  "wattHoursLifetime": 0,
  "wattsNow": 0
}

thats probably why sensor Today's Energy Production stays at 0 Wh all day.

sensors for inverters and sensor 'current power production' looks ok and do give output.

vincentwolsink, 2023-07-13:
This is unfortunately an issue with the latest Envoy firmware. The Envoy S metered only exposes the Today's Energy Production/Consumption values from the CTs (not from the inverters). Since it is apparently common practise in the Netherlands to install the Envoy S metered without CTs quite some people are affected.

The Envoy S Standard does not have this issue and does expose the Today's Energy Production from the inverters.

vincentwolsink, 2023-07-13:
If you want to fix your Energy Dashboard in Home Assistant you can use the Lifetime Production sensor. Also you could create your own template sensor to get the Today Production value

mnederlof, 2023-07-14:
> If you want to fix your Energy Dashboard in Home Assistant you can use the Lifetime Production sensor. Also you could create your own `utility_meter` sensor to get the Today Production value:
> 
> ```
> utility_meter:
>   solar_today_energy_production:
>     source: sensor.envoy_XXXXXX_lifetime_energy_production
>     cycle: daily
> ```

When i did not have CT installed, i also did it like this in HA.

mnederlof, 2023-07-17:
What we might be able to do is to make sure the Today counter is not provided by plugin when using metered envoy (without CT's). Especially if all values in https://192.168.xx.xx/api/v1/production are zero..

There is no point in providing a entity which is not usable anyway...

vincentwolsink, 2023-07-23:
It seems there is another endpoint available: https://envoy.local/ivp/pdm/energy
Can you check that one @joverdijk ?

joverdijk, 2023-07-23:
Tried that endpoint but getting 401 authorization required, although I am logged in.
Connected to envoy.local, logged in with enlightenment credentials.
I don't have CTs connected. 

I now get ' too many sessions ' while trying to login. Will try again later.

mnederlof, 2023-07-23:
oeh, another endpoint 😄 
```json
{
    "production": {
        "pcu": {
            "wattHoursToday": 14291,
            "wattHoursSevenDays": 158957,
            "wattHoursLifetime": 3442571,
            "wattsNow": 13
        },
        "rgm": {
            "wattHoursToday": 0,
            "wattHoursSevenDays": 0,
            "wattHoursLifetime": 0,
            "wattsNow": 0
        },
        "eim": {
            "wattHoursToday": 14471,
            "wattHoursSevenDays": 160852,
            "wattHoursLifetime": 2043071,
            "wattsNow": 18
        }
    },
    "consumption": {
        "eim": {
            "wattHoursToday": 0,
            "wattHoursSevenDays": 0,
            "wattHoursLifetime": 0,
            "wattsNow": 0
        }
    }
}
```

mnederlof, 2023-07-26:
it is a installer only endpoint @vincentwolsink (with a owner token i'm getting a 401)
is it also available on the standard envoy (non metered)?

We could add this endpoint, assuming that the `production.pcu` values are updated even without CT's installed. (since my output is different between `eim` (CT) and `pcu`, i'd assume it should be there.

mnederlof, 2023-07-26:
(it is not that hard though: https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/compare/main...mnederlof:home_assistant_enphase_envoy_installer:pdm_endpoint) 

Other benefit would be that users with owner token no longer get the daily_production value now, as it would be None; disabling the entity.

vincentwolsink, 2023-07-26:
It is also available on the Envoy standard.

```
{
    "production": {
        "pcu": {
            "wattHoursToday": 23792,
            "wattHoursSevenDays": 113429,
            "wattHoursLifetime": 3091071,
            "wattsNow": 429
        },
        "rgm": {
            "wattHoursToday": 0,
            "wattHoursSevenDays": 0,
            "wattHoursLifetime": 0,
            "wattsNow": 0
        },
        "eim": {
            "wattHoursToday": 0,
            "wattHoursSevenDays": 0,
            "wattHoursLifetime": 0,
            "wattsNow": 0
        }
    },
    "consumption": {
        "eim": {
            "wattHoursToday": 0,
            "wattHoursSevenDays": 0,
            "wattHoursLifetime": 0,
            "wattsNow": 0
        }
    }
}
```

vincentwolsink, 2023-07-27:
I just discovered that on my Envoy standard the lifetime value in production.json is completely wrong.

`"whLifetime":705769`
vs
`"wattHoursLifetime": 3091071,`

Enphase really makes a mess out of these things.

mnederlof, 2023-07-27:
Yeah, i had the impression that whLifetime also reset after a certain amount..

catsmanac, 2023-07-27:
Envoy-s metered without connected clamps seems indeed to reset. Was reported [here ](https://github.com/home-assistant/core/issues/88983#issuecomment-1650330940) yesterday as well

vincentwolsink, 2023-07-27:
@mnederlof Yes. So unfortunately we cannot use it for all readings. I reverted to production_v1 for Envoy Standard

vincentwolsink, 2023-07-27:
@catsmanac Metered without clamps is the worst of all. The old production endpoint returns only zeros and now the production json resets the lifetime value. 😢 

I am thinking of using the new endpoint above when an Installer/DIY account is used, and using the old (buggy) endpoints for an homeowner account. I will try to support home owner accounts as much as possible, but in the end this integration was meant to be used for the extra data installer accounts give us.

catsmanac, 2023-07-27:
I'm ok with dropping homeowner support, no heartburn here. I realize it may be too limiting for your purpose. 

We're working on updating HA Core based on @briancmpbll version. It's including some of your features like the persistent store and improved sessions and would become new HA Core for homeowners and maintain legacy support as well. Installer and DIY account users wanting to unlock those features would then switch to your custom integration. As long as naming schema basis is the same or close switching is still easy.

mnederlof, 2023-07-27:
> @catsmanac Metered without clamps is the worst of all. The old production endpoint returns only zeros and now the production json resets the lifetime value. 😢
>
> I am thinking of using the new [...]

We could do something like this: https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/compare/main...mnederlof:home_assistant_enphase_envoy_installer:path_selection_by_token_type

joverdijk, 2023-07-27:
I am only a home-owner. How can I get a working installation then? I have seen some installer token but this is only very short lived token right??

mnederlof, 2023-07-27:
yes that token is only valid for 12 hours or so.

vincentwolsink, 2023-07-27:
> I am only a home-owner. How can I get a working installation then? I have seen some installer token but this is only very short lived token right??

Tokens a renewed by the integration automatically. The lifetime of a token should not matter for you as user.

joverdijk, 2023-07-27:
So with only home owner enlightenment login, the plugin should still be able to get installer tokens in the background for the good fucntioning of the plugin??? Or do you need a installer token to begin with? Is  that still that webtool that generates something based on your serialnumber??

vincentwolsink, 2023-07-27:
If you login with a home owner account, the integration will get a home owner token in the background. This will give you most of the functionality but for example extended statistics per inverter will be unavailable.

If you login with an installer or DIY account, the integration will get a installer token in the background and will give you all functionality.

So functionality will depend on the account type you have at Enphase.

mnederlof, 2023-07-31:
We've added the daily production counters using the new pdm endpoint (https://envoy.local/ivp/pdm/energy) with #60 

If you have a installed / DIY token, then you automatically get this information. Otherwise there is no way to fetch it currently..

Closing issue for now. If someone finds another way to get it for owner tokens, feel free to open a new issue for it.

closed 2023-07-31 by mnederlof