vincentwolsink/home_assistant_enphase_envoy_installer #47: [BUG]

BVGorp, 2023-07-19:
**Describe the bug**
Just installed this version of the integration comming from DEV.
Installed v 0.2.4 and tried 0.3 beta 2 both give about the same issue.
The integration shows as "Failed setup, will retry" on integrations page.

**Account type**
- [ ] Installer
- [ X] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [ X] Standard

- FW version:  D7.6.175 (f79c8d) 
- Amount of micro inverters connected: 18

**Home Assistant**
- Version: 2023.7.2

**Additional context**
```
2023-07-19 13:20:50.650 ERROR (MainThread) [custom_components.enphase_envoy] Unexpected error fetching envoy Envoy 122111010631 data: maximum recursion depth exceeded while calling a Python object
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
self.data = await self._async_update_data()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 242, in _async_update_data
return await self.update_method()
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/__init__.py", line 91, in async_update_data
data[description.key] = await envoy_reader.relay_status()
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/envoy_reader.py", line 617, in get_data
return self.data.get(name)
^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/envoy_reader.py", line 336, in __getattr__
result = getattr(self, name)
^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/envoy_reader.py", line 336, in __getattr__
result = getattr(self, name)
^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/envoy_reader.py", line 336, in __getattr__
result = getattr(self, name)
^^^^^^^^^^^^^^^^^^^
[Previous line repeated 962 more times]
File "/config/custom_components/enphase_envoy/envoy_reader.py", line 434, in relay_status
status = self._path_to_dict(
^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/envoy_reader.py", line 317, in _path_to_dict
for d in self._resolve_path(path, default=[]):
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/envoy_reader.py", line 301, in _resolve_path
result = jsonpath(self.data, path)
^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 316, in jsonpath
trace(cleaned_expr, obj, '$')
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 167, in trace
trace(x, obj[loc], s(path, loc))
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 167, in trace
trace(x, obj[loc], s(path, loc))
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 194, in trace
walk(loc, x, obj, path, f05)
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 245, in walk
funct(i, loc, expr, obj, path)
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 189, in f05
eval_result = evalx(loc, obj[int(key)])
^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 284, in evalx
loc = re.sub(r'(?<!\\)(@\.[a-zA-Z@_.0-9]+)', varmatch, loc)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/re/__init__.py", line 185, in sub
return _compile(pattern, flags).sub(repl, string, count)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 282, in varmatch
return brackets(elts[1:])
^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 273, in brackets
if isint(e):
^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 112, in isint
return x.isdigit()
^^^^^^^^^^^
RecursionError: maximum recursion depth exceeded while calling a Python object
```

vincentwolsink, 2023-07-19:
Hi! Can you please supply the output of your https://envoy.local/ivp/peb/devstatus

BVGorp, 2023-07-19:
counters	
pcu	
expected	18
discovered	18
ctrlsTotal	18
ctrlsGone	0
ctrlsCommunicating	18
chansTotal	18
chansRecent	18
chansProducing	18
acb	
expected	0
discovered	0
ctrlsTotal	0
ctrlsGone	0
ctrlsCommunicating	0
chansTotal	0
chansRecent	0
chansProducing	0
nsrb	
expected	1
discovered	1
ctrlsTotal	1
ctrlsGone	0
ctrlsCommunicating	1
chansTotal	1
chansRecent	1
chansProducing	0
pld	
expected	19
discovered	19
ctrlsTotal	19
ctrlsGone	0
ctrlsCommunicating	19
chansTotal	19
chansRecent	19
chansProducing	18
esub	
expected	0
discovered	0
ctrlsTotal	0
ctrlsGone	0
ctrlsCommunicating	0
chansTotal	0
chansRecent	0
chansProducing	0
pcu	
fields	
0	"serialNumber"
1	"devType"
2	"communicating"
3	"recent"
4	"producing"
5	"reportDate"
6	"temperature"
7	"dcVoltageINmV"
8	"dcCurrentINmA"
9	"acVoltageINmV"
10	"acPowerINmW"
values	
0	
0	"SN"
1	1
2	true
3	true
4	true
5	1689793862
6	22
7	28686
8	144
9	225288
10	3974
1	
0	"SN"
1	1
2	true
3	true
4	true
5	1689793865
6	21
7	28952
8	145
9	225768
10	4078
2	
0	"SN"
1	1
2	true
3	true
4	true
5	1689793893
6	21
7	28679
8	136
9	225520
10	3762
3	
0	"SN"
1	1
2	true
3	true
4	true
5	1689793924
6	22
7	28514
8	144
9	225296
10	3643
4	
0	"SN"
1	1
2	true
3	true
4	true
5	1689793924
6	22
7	28720
8	139
9	225632
10	3792
5	
0	"SN"
1	1
2	true
3	true
4	true
5	1689793953
6	22
7	28122
8	119
9	225640
10	3420
6	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794224
6	22
7	28306
8	106
9	225408
10	1492
7	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794225
6	22
7	27970
8	96
9	225216
10	1283
8	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794225
6	22
7	28003
8	88
9	224816
10	1302
9	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794226
6	23
7	28022
8	94
9	225040
10	1278
10	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794227
6	23
7	28044
8	102
9	223720
10	1232
11	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794228
6	23
7	28200
8	91
9	223928
10	1355
12	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794228
6	23
7	27717
8	86
9	223552
10	1141
13	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794229
6	23
7	27724
8	84
9	223776
10	1194
14	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794230
6	23
7	28051
8	85
9	223760
10	1250
15	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794230
6	22
7	28287
8	110
9	224488
10	1074
16	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794231
6	23
7	28147
8	87
9	223920
10	1110
17	
0	"SN"
1	1
2	true
3	true
4	true
5	1689794232
6	23
7	28128
8	83
9	223864
10	1174
18	
0	"SN"
1	12
2	true
3	true
4	false
5	1689794203
6	0
7	0
8	0
9	0
10	0
acb	
fields	
0	"serialNumber"
1	"SOC"
2	"minCellTemp"
3	"maxCellTemp"
4	"capacity"
5	"totVoltage"
6	"sleepEnabled"
7	"sleepMinSoc"
8	"sleepMaxSoc"
values	[]
nsrb	
fields	
0	"serialNumber"
1	"relay"
2	"forced"
3	"reason_code"
4	"reason"
5	"line-count"
6	"line1-connected"
7	"line2-connected"
8	"line3-connected"
values	
0	
0	"122052067927"
1	"closed"
2	false
3	-1
4	"ok"
5	1
6	true
7	false
8	false

BVGorp, 2023-07-19:
removed the serial numbers

vincentwolsink, 2023-07-23:
Can you please share the error output from version 0.2.4? (The one in additional context seems to be from beta 0.3.0)

mnederlof, 2023-07-23:
with the beta version, there seems to be a bug, when you only have 1 relay installed 😅 
The PR for this bug will follow soon..

(PS: as for the stable release, i did not test what went wrong there..)

BVGorp, 2023-07-24:
beta 3 works thanks