vincentwolsink/home_assistant_enphase_envoy_installer #48: [BUG] dead disconnection

jerome6994, 2023-07-19:
Good morning
I just installed the integration on Home Assistant and great job for the community bravo.
On the other hand, it seems that at times there are disconnections because at times my entities are inaccessible and my templates in error then it comes back without doing anything, then it starts again like this all day!
Is there a setting to be made in particular to make it communicate more fluidly? is there a way to have if short disconnection the previous info in retain at least? If disconnection is abnormally long ok to have an unknown value for example

I have this :
[image]

after that :
[image]

vincentwolsink, 2023-07-19:
Hi! Do you maybe have both an IPv4 and IPv6 address assigned to your Envoy?

Yellimur, 2023-07-19:
Hi,

Here exactly the same problem. After update of Enphase in HACS and restart no connect with Envoy anymore.

jerome6994, 2023-07-19:
> Hi! Do you maybe have both an IPv4 and IPv6 address assigned to your Envoy?

Good evening
I don't know how to say certain IPV4 but IPV6 normally no I haven't done anything for it.
I have a UDM-Pro router in DMZ but then the IPV6 I don't think I have activated it

vincentwolsink, 2023-07-20:
Indeed, you cannot have multiple active sessions to the Envoy. Can you maybe share your debug logs of the integration?

jerome6994, 2023-07-20:
Hello

I cut my test HA to do the observation today but it still works weird at the moment
I started the debug logs we will wait a moment to have information

[image]

[image]

vincentwolsink, 2023-07-20:
The reason for not doing a PR to the official HA is twofold:
1. This integration started out as an installer account only. It did not support non-installer accounts. Which will impact a lot of users of the current official integration. Also it doesn't support the D5 firmware, which the official integration does.
2. Since this integration is still in heavy development, I do not want to be bound to the HA release cycle and have to wait to release fixes and/or improvements.

jerome6994, 2023-07-20:
OK I understand
But I just saw one thing too is that the integration only supports installer accounts and not owner accounts only.

But my account is not an installer so can the disconnections come from there?

vincentwolsink, 2023-07-20:
The integration nowadays supports Installer, DIY and owner accounts. When using with an owner account there is less information available, and please make sure to check the owner account option during setup or via the configuration button afterwards, to prevent useless data lookups.

jerome6994, 2023-07-20:
I had not checked the box circled in red

[image]

vincentwolsink, 2023-07-23:
Looking at your screenshots you seem to have a DIY account and you should not tick that box. If the issue is not fixed yet, can you please share your debug logs?

jerome6994, 2023-07-23:
Good evening
Thanks for your feedback.

So I waited a bit but on Friday I moved on to another less rich but stable integration while waiting for yours to evolve, I hope in the right direction for me. 

My installation was done by a company. I didn't do anything myself!

Then I have two size concerns!
I don't know where to find the integration logs! I've never needed to watch this kind of thing so I want to have the process.
Then, are there still traces due to the change of integration?
Then I can also come back to your integration, let it run for a few days and send you fresh traces.

jerome6994, 2023-07-23:
it still picks up even though I did not check the box as requested
[image]
[journal.txt](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/12139964/journal.txt)

jerome6994, 2023-07-26:
Hello
Today's update is worse for me!
I no longer have a connection to enphase at all

bes-r, 2023-07-27:
The beta version 0.3.1 will fix your connection issue.

vincentwolsink, 2023-08-15:
This issue seems stale. Closing it for now. Please reopen if you still encounter problems.

closed 2023-08-15 by vincentwolsink

jerome6994, 2023-08-19:
Hello Vincent
I'm back from vacation and I just put the integration back today.
For the moment no disconnection
The fix seems ok, I don't know what it came from but great job