vincentwolsink/home_assistant_enphase_envoy_installer #54: [BUG] maximum recursion depth exceeded while calling a Python object

FreddyX078, 2023-07-26:
**Describe the bug**
After updating to 0.3.0, integration is not loading and keeps initializing.
Downgrading back to 0.2.4 and it is working again.

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [X] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [X] Metered without CTs
- [ ] Standard

- FW version: D7.6.175
- Amount of micro inverters connected: 12

**Home Assistant**
- Version: 2023.7.3

**Additional context**
```
Logger: custom_components.enphase_envoy Source: custom_components/enphase_envoy/envoy_reader.py:301  Integration: Enphase Envoy (Installer) ([documentation](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/), [issues](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/issues))  First occurred: 15:19:52 (4 occurrences)  Last logged: 15:22:15
Unexpected error fetching envoy Envoy xxxxxxxxxxxxx data: maximum recursion depth exceeded while calling a Python object
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 283, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 242, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 108, in async_update_data
    battery_data = await envoy_reader.battery_storage()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 621, in get_data
    return self.data.get(name)
           ^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 340, in __getattr__
    result = getattr(self, name)
             ^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 340, in __getattr__
    result = getattr(self, name)
             ^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 340, in __getattr__
    result = getattr(self, name)
             ^^^^^^^^^^^^^^^^^^^
  [Previous line repeated 971 more times]
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 453, in battery_storage
    storage = self._resolve_path("endpoint_production_json_results.storage[0]", {})
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 301, in _resolve_path
    result = jsonpath(self.data, path)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 311, in jsonpath
    cleaned_expr = normalize(expr)
                   ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jsonpath.py", line 84, in normalize
    x = re.sub(r"[\['](\??\(.*?\))[\]']", f1, x)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/re/__init__.py", line 185, in sub
    return _compile(pattern, flags).sub(repl, string, count)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/re/__init__.py", line 274, in _compile
    if isinstance(flags, RegexFlag):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RecursionError: maximum recursion depth exceeded while calling a Python object
```

mnederlof, 2023-07-26:
could you please share the output of https://envoy.local/production.json?details=1 and https://envoy.local/ivp/ensemble/inventory

pbvdven, 2023-07-26:
same issue here for log files see #55

pbvdven, 2023-07-26:
https://envoy.local/production.json?details=1

{"production":[{"type":"inverters","activeCount":15,"readingTime":1690394683,"wNow":1718,"whLifetime":1085210},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1690394716,"wNow":1386.618,"whLifetime":8322232.025,"varhLeadLifetime":4218.678,"varhLagLifetime":3526430.732,"vahLifetime":10790318.515,"rmsCurrent":5.972,"rmsVoltage":236.52,"reactPwr":248.643,"apprntPwr":1412.314,"pwrFactor":0.98,"whToday":22460.025,"whLastSevenDays":125622.025,"vahToday":24870.515,"varhLeadToday":0.678,"varhLagToday":5094.732,"lines":[{"wNow":1386.618,"whLifetime":8322232.025,"varhLeadLifetime":4218.678,"varhLagLifetime":3526430.732,"vahLifetime":10790318.515,"rmsCurrent":5.972,"rmsVoltage":236.52,"reactPwr":248.643,"apprntPwr":1412.314,"pwrFactor":0.98,"whToday":22460.025,"whLastSevenDays":125622.025,"vahToday":24870.515,"varhLeadToday":0.678,"varhLagToday":5094.732}]}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1690394716,"wNow":457.66,"whLifetime":12002818.218,"varhLeadLifetime":10201739.104,"varhLagLifetime":-3525334.716,"vahLifetime":19123311.47,"rmsCurrent":0.851,"rmsVoltage":236.414,"reactPwr":-482.763,"apprntPwr":201.306,"pwrFactor":1.0,"whToday":8736.218,"whLastSevenDays":99567.218,"vahToday":28503.47,"varhLeadToday":14614.104,"varhLagToday":0.0,"lines":[{"wNow":457.66,"whLifetime":12002818.218,"varhLeadLifetime":10201739.104,"varhLagLifetime":-3525334.716,"vahLifetime":19123311.47,"rmsCurrent":0.851,"rmsVoltage":236.414,"reactPwr":-482.763,"apprntPwr":201.306,"pwrFactor":1.0,"whToday":8736.218,"whLastSevenDays":99567.218,"vahToday":28503.47,"varhLeadToday":14614.104,"varhLagToday":0.0}]},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1690394716,"wNow":-928.958,"whLifetime":3904391.083,"varhLeadLifetime":10205957.782,"varhLagLifetime":1096.016,"vahLifetime":19123311.47,"rmsCurrent":-5.12,"rmsVoltage":236.414,"reactPwr":-731.406,"apprntPwr":-1210.547,"pwrFactor":-1.0,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0,"lines":[{"wNow":-928.958,"whLifetime":3904391.083,"varhLeadLifetime":10205957.782,"varhLagLifetime":1096.016,"vahLifetime":19123311.47,"rmsCurrent":-5.12,"rmsVoltage":236.414,"reactPwr":-731.406,"apprntPwr":-1210.547,"pwrFactor":-1.0,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}]}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

https://envoy.local/ivp/ensemble/inventory empty
[]

FreddyX078, 2023-07-26:
For me the output is like @pbvdven 

https://envoy.local/production.json?details=1
`{"production":[{"type":"inverters","activeCount":12,"readingTime":1690396533,"wNow":174,"whLifetime":573065},{"type":"eim","activeCount":0,"measurementType":"production","readingTime":1690396563,"wNow":-0.0,"whLifetime":0.002,"varhLeadLifetime":0.041,"varhLagLifetime":1.457,"vahLifetime":147.656,"rmsCurrent":0.124,"rmsVoltage":235.24,"reactPwr":0.0,"apprntPwr":28.854,"pwrFactor":0.0,"whToday":0.002,"whLastSevenDays":0.002,"vahToday":147.656,"varhLeadToday":0.041,"varhLagToday":1.457,"lines":[{"wNow":-0.0,"whLifetime":0.002,"varhLeadLifetime":0.035,"varhLagLifetime":1.106,"vahLifetime":147.611,"rmsCurrent":0.124,"rmsVoltage":233.822,"reactPwr":0.0,"apprntPwr":28.854,"pwrFactor":0.0,"whToday":0.002,"whLastSevenDays":0.002,"vahToday":147.611,"varhLeadToday":0.035,"varhLagToday":1.106},{"wNow":-0.0,"whLifetime":0.0,"varhLeadLifetime":0.006,"varhLagLifetime":0.35,"vahLifetime":0.045,"rmsCurrent":0.0,"rmsVoltage":1.418,"reactPwr":0.0,"apprntPwr":0.0,"pwrFactor":0.0,"whToday":0.0,"whLastSevenDays":0.0,"vahToday":0.045,"varhLeadToday":0.006,"varhLagToday":0.35}]}],"consumption":[{"type":"eim","activeCount":0,"measurementType":"total-consumption","readingTime":1690396563,"wNow":-1.069,"whLifetime":0.002,"varhLeadLifetime":-0.041,"varhLagLifetime":-1.457,"vahLifetime":0.0,"rmsCurrent":-0.026,"rmsVoltage":239.384,"reactPwr":0.0,"apprntPwr":-6.01,"pwrFactor":1.0,"whToday":0.002,"whLastSevenDays":0.002,"vahToday":0.0,"varhLeadToday":0.0,"varhLagToday":0.0,"lines":[{"wNow":-1.069,"whLifetime":0.002,"varhLeadLifetime":-0.035,"varhLagLifetime":-1.106,"vahLifetime":0.0,"rmsCurrent":-0.026,"rmsVoltage":233.968,"reactPwr":0.0,"apprntPwr":-6.01,"pwrFactor":-1.0,"whToday":0.002,"whLastSevenDays":0.002,"vahToday":0.0,"varhLeadToday":0.0,"varhLagToday":0.0},{"wNow":0.0,"whLifetime":0.0,"varhLeadLifetime":-0.006,"varhLagLifetime":-0.35,"vahLifetime":0.0,"rmsCurrent":0.0,"rmsVoltage":5.416,"reactPwr":-0.0,"apprntPwr":0.0,"pwrFactor":0.01,"whToday":0.0,"whLastSevenDays":0.0,"vahToday":0.0,"varhLeadToday":0.0,"varhLagToday":0.0}]},{"type":"eim","activeCount":0,"measurementType":"net-consumption","readingTime":1690396563,"wNow":-1.069,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":-0.15,"rmsVoltage":239.384,"reactPwr":-0.0,"apprntPwr":-35.061,"pwrFactor":0.0,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0,"lines":[{"wNow":-1.069,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":-0.15,"rmsVoltage":233.968,"reactPwr":-0.0,"apprntPwr":-35.061,"pwrFactor":-1.0,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0},{"wNow":0.0,"whLifetime":0.0,"varhLeadLifetime":0.0,"varhLagLifetime":0.0,"vahLifetime":0.0,"rmsCurrent":0.0,"rmsVoltage":5.416,"reactPwr":-0.0,"apprntPwr":0.0,"pwrFactor":0.03,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}]}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}`

And https://envoy.local/ivp/ensemble/inventory downloads an empty file.

vincentwolsink, 2023-07-26:
Same error for me after applying #56, did not experience the error before. Trying to find what is wrong.

vincentwolsink, 2023-07-26:
Actually #52 broke this. Seeing how to fix this without breaking the relays again.

smytangel, 2023-07-26:
Beta 3.02 is working fine for me, but wen i updat to 3.0 its failed setup, will retry. Also 3.03 and 3.04 beta
Firmware: D7.6.175
Home Assistant 2023.7.3
Supervisor 2023.07.1
Operating System 10.3
Frontend 20230705.1 - latest

vincentwolsink, 2023-07-26:
Should be fixed in [0.3.1-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.3.1-beta1)

PR #57 (2023-07-26): Fix recursion lookup error
Should fix #54
closed 2023-07-27 by vincentwolsink