vincentwolsink/home_assistant_enphase_envoy_installer #75: Help wanted! Product ID mapping

vincentwolsink, 2023-08-03:
In the newest version of the integration we started displaying friendly model names for Enphase devices. Since there is no way to get these names from the Envoy API itself, and a public mapping does not seem to exist, we need to create this mapping ourselves.

We would be very happy if you could add the product number (this can be found in Home Assistant at the device page and then `Hardware`) and corresponding product name for the Enphase devices you own to this issue, so we can add them to the mapping.

[image]

Thanks 🚀

catsmanac, 2023-08-03:
Just confirming Envoy is showing 800-00656-r06 / Envoy-S-Standard-EU / ENV-S-WB-230  as well

Firmware D7.6.175  

My inverters are IQ7A Microinverter  SKU: IQ7PLUS-72-M-ACM-INT with PN 800-01127-r02 reported in inventory.json

Inventory.json reports for these as well
```
"img_pnum_running": "520-00082-r01-v04.27.04",
"ptpn": "540-00169-r01-v04.27.09",
```
And these numbers show as firmware on Enphase web

[image]

catsmanac, 2023-08-03:
Oh, my IQ 3 phase relay has label/sku on it: Q-Relay-3P-INT  and shows in Inventory as

```
    "type": "NSRB",
    "devices": [
      {
        "part_num": "800-00597-r02",
        "installed": "1675855823",
        "serial_num": "122230040706",
        "device_status": [
          "envoy.global.ok"
        ],
        "last_rpt_date": "1691076686",
        "admin_state": 1,
        "dev_type": 12,
        "created_date": "1675855823",
        "img_load_date": "1529083534",
        "img_pnum_running": "520-00086-r01-v02.12.11",
        "ptpn": "540-00139-r01-v02.12.00",
        "chaneid": 1811939601,
        "device_control": [
          {
            "gficlearset": false
          }
        ],
        "producing": false,
        "communicating": true,
        "provisioned": true,
        "operating": true,
        "relay": "closed",
        "reason_code": -1,
        "reason": "ok",
        "line-count": 3,
        "line1-connected": true,
        "line2-connected": true,
        "line3-connected": true
      }
    ]
```

1nsomnia, 2023-08-03:
Thanks for all the work Vincent. Just installed beta 5. Hope i can add the solar system to the energy dashboard. 

Envoy-S Metered
Firmware: D7.6.175
Hardware: ENV-S-WM-230 (800-00654-r06) 
Inverter
Firmware: 520-00082-r01-v04.28.07
Hardware: 800-01127-r02

vincentwolsink, 2023-08-03:
Hi @1nsomnia, thanks! Which inverters do you have?

mnederlof, 2023-08-04:
> Inverter IQ7A SKU: IQ7A-72-M-INT 
> Hardware: 800-01127-r02

Hmm, same hardware id, different SKU. I'd opt to use this SKU though, as it matches with the other ones 😄

vincentwolsink, 2023-08-04:
Yeah. No clue what the ACM part in the other SKU is.

catsmanac, 2023-08-04:
> Yeah. No clue what the ACM part in the other SKU is.

It showed on the enphase website in my quipment list. I checked again, but it's showing for all 24 inverters. So if you hit that specific SKU you know it's an IQ7A I'd say.

maart84, 2023-08-29:
I’ve got these inverters:
Hardware: 800-00631-r02
SKU: IQ7PLUS-72-2-INT
IQ7+ Microinverter
Thanks for this integration!

Matthew1471, 2023-08-29:
Greetings from the author of https://github.com/Matthew1471/Enphase-API 😄 

| Model|Device Part Number|Device Type
|-------------|-------------|-------------|
|ME215-60|800-00090-r01|pcu
|PE215-60|800-00091-r01|pcu
|ME215-60|800-00090-r02|pcu
|ME215-60|800-00090-r03|pcu
|ME215-60|800-00090-r04|pcu
|ME215-60|800-00090-r05|pcu
|M250-60 |800-00146-r01|pcu
|M250-60 |800-00146-r02|pcu
|M250-60 |800-00146-r03|pcu
|MP250-60 |800-00181-r04|pcu
|MP250-60 |800-00181-r06|pcu
|ME215-60-230-S22-E |800-00192-r06|pcu
|ME215-60-230-S22-E |800-00241-r03|pcu
|ME215-60-230-S22-E |800-00241-r04|pcu
|M215-60-230-S22-UK |800-00341-r06|pcu
|M215-60-230-S22-E |800-00357-r03|pcu
|M215-60-230-S22-E |800-00357-r04|pcu
|M215-60-230-S22-E-Eaglet-B |800-00364-r01|pcu
|M215-60-230-S22-E-Eaglet-B |800-00364-r02|pcu
|ME250-60Hz-72 |800-00227-r01|pcu
|ME250-60Hz-72 |800-00227-r03|pcu
|C250-60Hz-72 |800-00354-r03|pcu
|C250-60Hz-72 |800-00354-r04|pcu
|C250-60Hz-72-Eaglet-B |800-00365-r01|pcu
|C250-60Hz-72-Eaglet-B |800-00365-r02|pcu
|C250-60Hz-72 |800-00354-r01|pcu
|C250-60Hz-72 |800-00354-r02|pcu
|M250-50Hz-72 |800-00235-r03|pcu
|M250-50Hz-72 |800-00235-r04|pcu
|M250-50Hz-72 |800-00356-r03|pcu
|M250-50Hz-72 |800-00356-r04|pcu
|M250-50Hz-72-Eaglet-B |800-00362-r01|pcu
|M250-50Hz-72-Eaglet-B |800-00362-r02|pcu
|S275-60-P0|800-00530-r01|pcu
|S275-60-P0|800-00530-r02|pcu
|S230-60-P0|800-00532-r01|pcu
|S230-60-P0|800-00532-r02|pcu
|S230-60-P0|800-00532-r03|pcu
|S230-60-P0|800-00532-r04|pcu
|S230-60-P0|800-00532-r05|pcu
|S230-60-LN-2-AU|800-00580-r05|pcu
|S230-60-LN-2-AU|800-00582-r02|pcu
|S230-60-LN-2-AU|800-00582-r03|pcu
|S270-72-LN-2|800-00535-r01|pcu
|S270-72-LN-2|800-00535-r02|pcu
|S270-72-LN-2|800-00535-r03|pcu
|S270-72-LN-2|800-00535-r04|pcu
|S270-72-LN-2|800-00535-r05|pcu
|S270-72-LN-2-AU - Vogel|800-00539-r05|pcu
|S270-72-LN-2|800-00584-r03|pcu
|S270-72-LN-2-AU|800-00586-r01|pcu
|S270-72-LN-2-AU|800-00586-r02|pcu
|S270-72-LN-2-AU|800-00586-r03|pcu
|S270-72-LN-2-AU|800-00586-r04|pcu
|S270-72-LN-2-AU|800-00586-r05|pcu
|S270-72-LN-5-AU|800-00587-r01|pcu
|S270-72-LN-5-AU|800-00587-r02|pcu
|S270-72-LN-5-AU|800-00587-r03|pcu
|S270-72-LN-5-AU|800-00587-r04|pcu
|S270-72-LN-5-AU|800-00587-r05|pcu
|S270-72-LN-ACB|800-00560-r01|acb
|S270-72-LN-ACB|800-00560-r02|acb
|S270-72-LN-ACB|800-00560-r03|acb
|S270-72-LN-ACB15|800-00930-r01|acb
|S270-72-LN-ACB15|800-00930-r03|acb
|S290-72-LL|800-00501-r01|pcu
|S290-72-LL|800-00501-r02|pcu
|IQ6-60-LL|800-00666-r01|pcu
|IQ6Plus-72-LL,India|800-00665-r08|pcu
|IQ6Plus-72-LL|800-00667-r01|pcu
|IQ7X-96-INTL P3 TOLL|800-00635-r01|pcu
|IQ7X-96-INTL P3 TOLL|800-00635-r02|pcu
|IQ7X-96-INTL Enphase BH, TOLL|800-00638-r02|pcu
|IQ7X-96-LL ROW BH, TOLL, INT, FET |800-01512-r01|pcu
|IQ7X-96 EN4 BH TOLL|800-01712-r02|pcu
|IQ7-60-LL P4 D2PAK|800-01102-r02|pcu
|IQ7-60-LL ROW QF4 BH, D2PAK|800-01122-r02|pcu
|IQ7Plus-72-LL P4 D2PAK|800-01103-r02|pcu
|IQ7Plus-72-LL QF4 BH, D2PAK|800-01123-r02|pcu
|IQ7-60-LL|800-00628-r01|pcu
|IQ7-60-LL|800-00628-r02|pcu
|IQ7-60-LL with bulkhead |800-00630-r01|pcu
|IQ7-60-LL with bulkhead |800-00630-r02|pcu
| IQ7-60-LL QF4 Bulkhard, TOLL|800-00636-r02|pcu
| IQ7-60-LL MC4 Bulkhead |800-01124-r02|pcu
| IQ7-60-LL QF4 Bulkhard, FET TOLL|800-01510-r01|pcu
|IQ7-60 EN4 BH TOLL|800-01710-r02|pcu
|IQ7Plus-72-LL|800-00629-r01|pcu
|IQ7Plus-72-LL|800-00629-r02|pcu
|IQ7Plus-72-LL with bulkhead |800-00631-r01|pcu
|IQ7Plus-72-LL with bulkhead |800-00631-r02|pcu
| IQ7+-72-LL QF4 Bulkhead, TOLL|800-00637-r02|pcu
|IQ7Plus-LL ROW BH, TOLL, INT, FET |800-01511-r01|pcu
|IQ7Plus-72 EN4 BH TOLL|800-01711-r02|pcu
|IQ7Plus-72 MC4 BH TOLL|800-01736-r02|pcu
|IQ7Plus-72-2-INT Enphase BH, STO36N60M6 |800-01141-r01|pcu
|IQ7A-72-2-INT MC4 Bulkhead, TOLL|800-01127-r02|pcu
|IQ7A-72-2-INT Enphase BH, TOLL|800-01138-r02|pcu
|IQ7A-72, INT, EN4 BH, TOLL|800-01714-r02|pcu
| IQ7+-72-2-INT Enphase BH, STO36N60M6 |800-01141-r02|pcu
|NSR-B 1P P1|800-00595-r01|nsrb
|NSR-B 3P External|800-00596-r01|nsrb
|NSR-B 3P Internal|800-00597-r01|nsrb
|NSR-B 3P Internal|800-00597-r02|nsrb
|NSR-B 1P P2|800-00598-r01|nsrb
|NSR-B 1P P2|800-00598-r02|nsrb
|NSR-B 1P P2+|800-00598-r03|nsrb
|NSR-B 1P P4 |800-00598-r04|nsrb
|IQ7Plus-72-RMA TO-LL S-series|800-01211-r02|pcu
|IQ7Plus-72-RMA TO-LL M-series|800-01210-r02|pcu
|IQ7Plus-72-RMA|800-01200-r02|pcu
|NSR 3P |800-00606-r02|nsrb

Matthew1471, 2023-08-31:
>You were able to crawl this data from some public available source?
🕵️ https://support.enphase.com/s/question/0D53m00009LtbtXCAR/report-new-security-vulnerability-but-cyber-security-email-address-does-not-exist 😀 

> Also nice library you have made there :)
Thanks, I wanted to make something people can include in their projects so we could all have a consistent library.. there's a lot of rubbish code out there to talk to the IQ Gateway and it meshes the program logic in with the Enphase logic.. The library is on PyPi so it can just be included.
 
> Edit: Unfortunately the table does not seem to exactly match earlier reports from people. Seems like Enphase is making a bit of a mess, or the source is not that accurate.
Did I get a RegEx wrong? Which ones don't match and I can go back to the source file.. fairly sure my source is accurate, any mistakes are likely to come from the RegEx I wrote. I manually checked a few in this thread and they seemed to check out. I couldn't RegEx the IQ8s but they are in the file, if it was useful I can hand extract those.

Let me know if you need any other info.

vincentwolsink, 2023-09-14:
@Matthew1471 Probably it is just Enphase naming things differently in the source than other places.

But for example:
For `800-01127-r02`, your table: `IQ7A-72-2-INT MC4 Bulkhead, TOLL`, users reported: `IQ7A-72-M-INT` (2 vs M)
For `800-00631-r02`, your table: `IQ7Plus-72-LL with bulkhead`, users reported: `IQ7PLUS-72-2-INT` (LL vs 2-INT)

hapklaar, 2025-05-22:
IQ8HC-72-M-INT: 800-01391-r03