vincentwolsink/home_assistant_enphase_envoy_installer #77: [BUG]

mvroosmalen1970, 2023-08-03:
**Describe the bug**
My enphase is recognized but get this error:
Logger: homeassistant.config_entries
Source: config_entries.py:952
First occurred: 8:09:45 PM (1 occurrences)
Last logged: 8:09:45 PM

Config entry 'Envoy xxxxxxxx' for enphase_envoy integration not ready yet; Retrying in background
in debug:
2023-08-03 19:17:05.851 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.xx/production.json?details=1: <Response [200 OK]>: {"production":[{"type":"inverters","activeCount":19,"readingTime":1691083025,"wNow":189,"whLifetime":686492}, etc..
All OK
2023-08-03 19:17:05.852 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_production_json_results TOOK 0.7850 seconds
2023-08-03 19:17:05.854 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_production_json_results.storage[0]
2023-08-03 19:17:05.854 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_production_json_results.storage[0] did not return anything!
2023-08-03 19:17:05.854 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_ensemble_json_results[0].devices
2023-08-03 19:17:05.854 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_ensemble_json_results[0].devices did not return anything!
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_home_json_results.enpower.grid_status
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_home_json_results.enpower.grid_status did not return anything!
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_inventory_results.[?(@.type=='PCU')].devices[?(@.dev_type==1)]
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_inventory_results.[?(@.type=='PCU')].devices[?(@.dev_type==1)] did not return anything!
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_production_inverters.[?(@.devType==1)]
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_production_inverters.[?(@.devType==1)] did not return anything!
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_devstatus.pcu[?(@.devType==1)]
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_devstatus.pcu[?(@.devType==1)] did not return anything!
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_production_power.powerForcedOff
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_production_power.powerForcedOff did not return anything!
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_inventory_results.[?(@.type=='NSRB')].devices[?(@.dev_type==12)]
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_inventory_results.[?(@.type=='NSRB')].devices[?(@.dev_type==12)] did not return anything!
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_devstatus.pcu[?(@.devType==12)]
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_devstatus.pcu[?(@.devType==12)] did not return anything!
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_devstatus.nsrb
2023-08-03 19:17:05.855 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_devstatus.nsrb did not return anything!
2023-08-03 19:17:05.856 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_inventory_results.[?(@.type=='NSRB')].devices[?(@.dev_type==12)]
2023-08-03 19:17:05.856 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] the configured path endpoint_inventory_results.[?(@.type=='NSRB')].devices[?(@.dev_type==12)] did not return anything!
2023-08-03 19:17:05.856 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] Updating endpoints {'endpoint_production_json_results', 'endpoint_production_inverters', 'endpoint_devstatus', 'endpoint_info_results', 'endpoint_inventory_results', 'endpoint_production_power', 'endpoint_ensemble_json_results', 'endpoint_pdm_energy', 'endpoint_home_json_results'}
2023-08-03 19:17:05.856 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_production_json_results: https://{}/production.json?details=1

and lot more and finally:

2023-08-03 19:17:31.220 DEBUG (MainThread) [custom_components.enphase_envoy] Finished fetching envoy Envoy xxxxxxxxxxxx data in 30.006 seconds (success: False)

**Account type**
- [ ] Installer
- [ x ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ x ] Metered without CTs
- [ ] Standard

- FW version: D7.6.175.230622 (f79c8d)
- Amount of micro inverters connected: 19

**Home Assistant**
- Version: 2023.8.0

**Additional context**
Same problem with previously working enphase dev plugin. Stopped working this morning. Did remove everything, update HA and 
plugin. Hope that some suggestions help......

Relevant snippet of Home Assistant error log. 
see above

mnederlof, 2023-08-04:
> 2023-08-03 19:17:31.220 DEBUG (MainThread) [custom_components.enphase_envoy] Finished fetching envoy Envoy xxxxxxxxxxxx data in 30.006 seconds (success: False)

This is a indication that the connection to you envoy seems really slow, and you hit the timeout of 30 seconds.
https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/blob/main/custom_components/enphase_envoy/__init__.py#L81

Try increasing that 30 to 60, and please let us know if that fixes the issue :)

catsmanac, 2023-08-04:
I've seen 30 sec timeout on home.json [reported](https://github.com/briancmpbll/home_assistant_custom_envoy/issues/123) that was solved by restarting the envoy.

mvroosmalen1970, 2023-08-04:
> I've seen 30 sec timeout on home.json [reported](https://github.com/briancmpbll/home_assistant_custom_envoy/issues/123) that was solved by restarting the envoy.

The change to 60 solved it.After a reboot of HA the panels are recognised

Thanks!!!!

PR #78 (2023-08-04): Add option to configure the timeout value to get data from envoy.
Fixes #77

Increased the timeout value from 30 to 60 seconds too, so when adding a envoy configuration, it will make use of a 60 second timeout, just to be sure.
closed 2023-08-04 by vincentwolsink