vincentwolsink/home_assistant_enphase_envoy_installer #80: [BUG]

muldy, 2023-08-05:
Can't add "sensor.envoy_SERIAL_current_power_production" to the energy dashboard.
It's not showing up on the "Add Solar production" section.

Is there a way to add it?

catsmanac, 2023-08-05:
Energy dashboard needs energy (Wh or J) entities, not power (W) entities. Isn't there an energy entity showing in the list like Lifetime Energy Production?

muldy, 2023-08-05:
[image]
Should I choose "Today's energy production" ?

Thank you

catsmanac, 2023-08-05:
I have mine running on lifetime and that works fine.
[image]

muldy, 2023-08-05:
Thanks!
I get values now!

But something seems wrong: 
[image]

Main power is shelly clamp in the house's grid entry, it's showing a surplus, but the solar value on the graph does not reflect that (Solar should be more than 1.8 kwh? ).

muldy, 2023-08-05:
I think I see it now, the "Energy distribution" distribution card are aggregate values.
Can close if this is true.

catsmanac, 2023-08-05:
Energy dashboard takes a while before it shows all because it's running from history.

I can't close issues as bystander, you as originator can or @vincentwolsink.

muldy, 2023-08-05:
thanks again!

closed 2023-08-05 by muldy