vincentwolsink/home_assistant_enphase_envoy_installer #81: [BUG] Realtime data error in log

Jagermyster, 2023-08-07:
**Describe the bug**
Two major(?) errors in log though everything seems to be working?

**Account type**
- [x ] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.6.175
- Amount of micro inverters connected: 14

**Home Assistant**
- Version: 2023.8.1

**Additional context**
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.enphase_envoy.envoy_reader
Source: custom_components/enphase_envoy/envoy_reader.py:1084
Integration: Enphase Envoy (Installer) (documentation, issues)
First occurred: 09:25:52 (5 occurrences)
Last logged: 09:36:25
----------------------------------------------------
Stopped reading realtime data```
Relevant snippet of Home Assistant error log. 
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.enphase_envoy.envoy_reader
Source: custom_components/enphase_envoy/envoy_reader.py:1061
Integration: Enphase Envoy (Installer) (documentation, issues)
First occurred: 09:25:52 (5 occurrences)
Last logged: 09:36:25

Realtime data error:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 131, in _call_sslobject_method
    result = func(*args)
             ^^^^^^^^^^^
  File "/usr/local/lib/python3.11/ssl.py", line 921, in read
    v = self._sslobj.read(len)
        ^^^^^^^^^^^^^^^^^^^^^^
ssl.SSLWantReadError: The operation did not complete (read) (_ssl.c:2576)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/httpcore/_backends/anyio.py", line 34, in read
    return await self._stream.receive(max_bytes=max_bytes)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 196, in receive
    data = await self._call_sslobject_method(self._ssl_object.read, max_bytes)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 138, in _call_sslobject_method
    data = await self.transport_stream.receive()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 1203, in receive
    await self._protocol.read_event.wait()
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/httpcore/_exceptions.py", line 10, in map_exceptions
    yield
  File "/usr/local/lib/python3.11/site-packages/httpcore/_backends/anyio.py", line 32, in read
    with anyio.fail_after(timeout):
  File "/usr/local/lib/python3.11/site-packages/anyio/_core/_tasks.py", line 119, in __exit__
    raise TimeoutError
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 60, in map_httpcore_exceptions
    yield
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 239, in __aiter__
    async for part in self._httpcore_stream:
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/connection_pool.py", line 347, in __aiter__
    async for part in self._stream:
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 325, in __aiter__
    raise exc
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 317, in __aiter__
    async for chunk in self._connection._receive_response_body(**kwargs):
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 186, in _receive_response_body
    event = await self._receive_event(timeout=timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 200, in _receive_event
    data = await self._network_stream.read(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpcore/_backends/anyio.py", line 31, in read
    with map_exceptions(exc_map):
  File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/usr/local/lib/python3.11/site-packages/httpcore/_exceptions.py", line 14, in map_exceptions
    raise to_exc(exc) from exc
httpcore.ReadTimeout

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1061, in stream_reader
    async for chunk in response.aiter_text():
  File "/usr/local/lib/python3.11/site-packages/httpx/_models.py", line 948, in aiter_text
    async for byte_content in self.aiter_bytes():
  File "/usr/local/lib/python3.11/site-packages/httpx/_models.py", line 927, in aiter_bytes
    async for raw_bytes in self.aiter_raw():
  File "/usr/local/lib/python3.11/site-packages/httpx/_models.py", line 985, in aiter_raw
    async for raw_stream_bytes in self.stream:
  File "/usr/local/lib/python3.11/site-packages/httpx/_client.py", line 146, in __aiter__
    async for chunk in self._stream:
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 238, in __aiter__
    with map_httpcore_exceptions():
  File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 77, in map_httpcore_exceptions
    raise mapped_exc(message) from exc
httpx.ReadTimeout
```

vincentwolsink, 2023-08-07:
Is your Envoy connected via Wifi or cable? This could be a bad/slow connection to the Envoy.

Jagermyster, 2023-08-07:
It is connected via cable. I noticed the slow connection, just don't understand why it is slow. The cable goes directly from the Envoy into the router, no issues with my internet connection. Can it be a faulty Envoy?

vincentwolsink, 2023-08-07:
Usually a power cycle of the Envoy will fix the slow response.

Jagermyster, 2023-08-07:
I did a powercycle 2 days ago because the integration could not be installed. After the power cycle it worked perfect. So it looks like my Envoy gets slow after a few days. Did a power cycle just now and keep monitoring it. Thanks for the reply so far!

Jagermyster, 2023-08-07:
After the power cycle the integration fails to start now :(

vincentwolsink, 2023-08-07:
Try to disable the realtime data in the configuration. And see if that improves things.

vincentwolsink, 2023-08-17:
Duplicate of #83

closed 2023-08-31 by vincentwolsink