vincentwolsink/home_assistant_enphase_envoy_installer #83: Loadoff Failed (Uitladen Mislukt)

gritcalculator, 2023-08-13:
Hi,

No real bug to be honest, however I do get the message "uitladen mislukt", which I think is the raw translation of loadoff failed.
As Home Assistant does not display any additional information regarding this alert (yellow alert sign is displayed), I am curious why this is visible and how it impacts the working of this excellent integration.

Kind regards,
Guus

vincentwolsink, 2023-08-15:
Hi Guus, this is probably "Unload failed". This integration supports unloading. But to be completely honest, I am not sure when Home Assistant would try to unload an integration. Is the error always there? Even after restarting Home Assistant? Is there anything in the logs about this?

gritcalculator, 2023-08-15:
Hi Vincent,

Thank you for the fast reply, appreciated!

I agree, would not expect the need for unloading. The integration is running non stop and seems to function as designed.

Yes, even after a reboot of the RPi, the error returns. Maybe I should just ingnore it, however the exclamation mark triggers attention :)..

Normally I would check for a seperate button ‘log’, do you have a quick how-to where this located in this integration? 

Kind regards,
Guus

vincentwolsink, 2023-08-15:
The error/message, if there is any, should be in your System logbook.

[image]

gritcalculator, 2023-08-15:
Hi Vincent,

Thanks, I think this is the error related.
I hope, I’ve redacted all ‘personal’ data :)..

Deze fout is ontstaan door een aangepaste integratie.

Logger: homeassistant.config_entries
Source: custom_components/enphase_envoy/envoy_reader.py:579
Integration: Enphase Envoy (Installer) (documentation, issues) 
First occurred: 14 augustus 2023 om 20:14:32 (1 occurrences) 
Last logged: 14 augustus 2023 om 20:14:32

Error unloading entry Envoy *redacted for privacy* for enphase_envoy
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 131, in _call_sslobject_method
    result = func(*args)
             ^^^^^^^^^^^
  File "/usr/local/lib/python3.11/ssl.py", line 921, in read
    v = self._sslobj.read(len)
        ^^^^^^^^^^^^^^^^^^^^^^
ssl.SSLWantReadError: The operation did not complete (read) (_ssl.c:2576)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/httpcore/_backends/anyio.py", line 34, in read
    return await self._stream.receive(max_bytes=max_bytes)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 196, in receive
    data = await self._call_sslobject_method(self._ssl_object.read, max_bytes)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/streams/tls.py", line 138, in _call_sslobject_method
    data = await self.transport_stream.receive()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/anyio/_backends/_asyncio.py", line 1203, in receive
    await self._protocol.read_event.wait()
  File "/usr/local/lib/python3.11/asyncio/locks.py", line 213, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/httpcore/_exceptions.py", line 10, in map_exceptions
    yield
  File "/usr/local/lib/python3.11/site-packages/httpcore/_backends/anyio.py", line 32, in read
    with anyio.fail_after(timeout):
  File "/usr/local/lib/python3.11/site-packages/anyio/_core/_tasks.py", line 119, in __exit__
    raise TimeoutError
TimeoutError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 60, in map_httpcore_exceptions
    yield
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 239, in __aiter__
    async for part in self._httpcore_stream:
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/connection_pool.py", line 347, in __aiter__
    async for part in self._stream:
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 325, in __aiter__
    raise exc
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 317, in __aiter__
    async for chunk in self._connection._receive_response_body(**kwargs):
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 186, in _receive_response_body
    event = await self._receive_event(timeout=timeout)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpcore/_async/http11.py", line 200, in _receive_event
    data = await self._network_stream.read(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/httpcore/_backends/anyio.py", line 31, in read
    with map_exceptions(exc_map):
  File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/usr/local/lib/python3.11/site-packages/httpcore/_exceptions.py", line 14, in map_exceptions
    raise to_exc(exc) from exc
httpcore.ReadTimeout

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 544, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 293, in async_unload_entry
    await task
  File "/config/custom_components/enphase_envoy/__init__.py", line 251, in read_realtime_updates
    result = await envoy_reader.stream_reader(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 579, in stream_reader
    async for chunk in response.aiter_text():
  File "/usr/local/lib/python3.11/site-packages/httpx/_models.py", line 948, in aiter_text
    async for byte_content in self.aiter_bytes():
  File "/usr/local/lib/python3.11/site-packages/httpx/_models.py", line 927, in aiter_bytes
    async for raw_bytes in self.aiter_raw():
  File "/usr/local/lib/python3.11/site-packages/httpx/_models.py", line 985, in aiter_raw
    async for raw_stream_bytes in self.stream:
  File "/usr/local/lib/python3.11/site-packages/httpx/_client.py", line 146, in __aiter__
    async for chunk in self._stream:
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 238, in __aiter__
    with map_httpcore_exceptions():
  File "/usr/local/lib/python3.11/contextlib.py", line 155, in __exit__
    self.gen.throw(typ, value, traceback)
  File "/usr/local/lib/python3.11/site-packages/httpx/_transports/default.py", line 77, in map_httpcore_exceptions
    raise mapped_exc(message) from exc
httpx.ReadTimeout

vincentwolsink, 2023-08-15:
The error seems to be the same as in #81. Do you have real-time updates enabled (See configure button at the integration)?

gritcalculator, 2023-08-15:
It seems indeed to be related.
Yes, real-time updates is enabled.

I read that you advise to disable it, and see if that results in no more errors, will try that this week.

For the record:

Account type: [x ] Installer
[x ] Metered with CTs installed
FW version: D7.6.175
Amount of micro inverters connected: 10
Home Assistant Version: 2023.8.2

gritcalculator, 2023-08-16:
I’ve disabled real-time updates, and the error is gone. If you prefer, this issue may be closed. However I will watch issue #81 . However if you would like to check/test things, let me know!

vincentwolsink, 2023-08-17:
@mnederlof, any idea about this one? It seems to throw an exception during unloading the stream endpoint. Not easy to debug for me with my Envoy Standard.

mnederlof, 2023-08-17:
It has to do with logic I added previously for handling the realtime task, but I can debug in a few days for sure

vincentwolsink, 2023-08-29:
@gritcalculator can you please check if [0.3.4-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.3.4-beta1) fixes your issue?

closed 2023-08-31 by vincentwolsink

gritcalculator, 2023-08-31:
> @gritcalculator can you please check if [0.3.4-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.3.4-beta1) fixes your issue?

Hi Vincent,

Sorry for waiting, was abroad a few days.

I've installed the 0.3.4 non-beta relaase. It seems that error messages do not populate directly, which was the case previously.

If things change, I'll let you know. For now case closed (saw that this has already been done).

Thanks for helping me out! Also a thank you to the other programmer/author which stepped in to debug!