vincentwolsink/home_assistant_enphase_envoy_installer #84: [FEATURE] Token caching

jsimonetti, 2023-08-21:
**Describe the feature**
Use a cache for the authentication token. The tokens have different lifetimes depending on the account type.
It would be nice if restarting HA and/or the integration did not always result in retrieving a new token.

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [ X ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ X ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.6.175
- Amount of micro inverters connected: 15

**Home Assistant**
- Version: 2023.8.3

vincentwolsink, 2023-08-21:
Hi @jsimonetti, this is already implemented. Do you see other behaviour?

jsimonetti, 2023-08-21:
Hi @vincentwolsink,

I do NOT explicitely see behaviour a cache is not used.
I did not find a cache option or log entry in my logs, nor did I see it mentioned in release notes, so I made the assumption that is was not implemented. (Initialising the integration took quite some time, which led me to think the token was retrieved every time).

Thank you for confirming it is implemented. If I happen to see behaviour that does not indicate caching, I will open a bug report.

Thank you for the work!

closed 2023-08-21 by jsimonetti