vincentwolsink/home_assistant_enphase_envoy_installer #85: When configuring, I get always "Failed to connect"

FredM67, 2023-08-27:
**Describe the bug**
When configuring, I get always "Failed to connect"

**Account type**
- [X] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [X] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: R5.0.54
- Amount of micro inverters connected: 19

**Home Assistant**
- Version: 2023.8.4

**Additional context**
In the Enlighten WebUI, I can retrieve everything, including the temperature of each inverter.
```
Relevant snippet of Home Assistant error log. 
```

vincentwolsink, 2023-08-27:
Hi @FredM67. This integration is only for Envoys with firmware 7 (see readme). It will not work with firmware 5.

FredM67, 2023-08-27:
> Hi @FredM67. This integration is only for Envoys with firmware 7 (see readme). It will not work with firmware 5.

Is there anything for 5.x so I can get the temperature in my HomeAssistant ?!?

vincentwolsink, 2023-08-28:
I am not aware of any integration that exposes things like temperatures for firmware 5.x