vincentwolsink/home_assistant_enphase_envoy_installer #89: Seems to be possible to use DIY or Installer features with only a Home Owner account

pbrunier, 2023-09-07:
**Describe the feature**
Today I made a mistake. On my HASS I have the https://github.com/briancmpbll/home_assistant_custom_envoy repo active and correctly working. The mistake was that I enabled THIS repo (envoy installer) add the same time. I guess it has overwritten scripts etc from the briancmpbll. I was not aware of that at that moment. I configured the installer integration (including the statement I only have a home owner account) and the only weird thing that happened was that, that HASS complained something like "Device already configured". Hmmm ok.
Then I saw something wonderful. HASS was showing me much more than I normally can see as a home owner.
Voltages DC and AC, currents, IQ relays, etc, etc. And....a production switch.
I toggled the switch....and production dropped. I toggled it again and production came back to life. Please note that this switch only works directly on the entity and not on the dashboard.

The mix up between the two integrations created some weird condition, but it turns out that APIs don't deny certain features for home owner accounts. Is it possible to include some of these? Especially turning of production would be nice in the Netherlands (paying for delivering power to the net)

I now have returned everything back to the custom_envoy repo, because it is unsupported by this integration.

I did a reproduction test and it is reproducable:
1. Install and configure custom_envoy
2. Install and configure THIS repo (with the home owner checkbox ENABLED)
3. Enjoy all entities (although a bit broken in the dashboard, but OK in the entity view)

When I first remove the custom_envoy config and remove the integration before installing THIS repo, the behaviour isn't there. No voltages DC and AC, currents, IQ relays, etc, etc.

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [X ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [ X] Standard

- FW version: D7.0.88
- Amount of micro inverters connected: 16

**Home Assistant**
- Version: 2023.9.0

vincentwolsink, 2023-09-07:
Hi @pbrunier, your Envoy runs a firmware version that exposes more data with a home owner account than newer firmwares do. Normally your Envoy would already have auto updated to a newer firmware version.

While your Envoy firmware is still on the current version you can use this integration and just not check the home owner checkbox. You should be able to see all the extra data.

pbrunier, 2023-09-07:
Ah thank you. That explains it. Will contact my installer about the firmware, because I think that's currently more important than the features.
Do you know if you buy Enphase Envoy Manager access, you also have all the nice features?

pbrunier, 2023-09-07:
In the mean time I managed to get a DIY account from Enphase and they also updated the firmware. Thank you for your information on this issue.

closed 2023-09-07 by pbrunier