vincentwolsink/home_assistant_enphase_envoy_installer #95: Abilty to set a Grid Profile

brodykenrick, 2023-10-29:
**Describe the feature**
With the installer or DIY installer permissions it would possible to set a "grid profile" (after adding options for those grid profiles in the installer app).

This is useful for setting a variety of aspects of a grid profile. One setting of particular usefulness is the "power export limit" setting that limits the amount of power the enphase system will export back to the grid (by limiting generation). For enphase system users exposed to export chargers or negative pricing at some times then the abilty to switch between two grid profiles is useful (to reduce costs from unwanted power in the grid and to help grid stabilty).

Further details for this usecase present to many enphase systems in Australia see a Ruby tool with the functionality for dynamically setting an appropriate grid profile at https://github.com/ZimbiX/amber-enphase-zero-export-switcher-tool#readme .

The URL/endpoint is (similar to power production):
> "#{base_url}/installer/agf/set_profile.json"

The request takes the form:
> http.put(set_grid_profile_url, json: { selected_profile: name })

Where selected_profile is one of two profiles of setup on this particular gateway for enabling or limiting power export.
e.g.
> AS/NZS 4777.2: 2020 Australia A Region  5 kW Export Limit:1.2.2
> AS/NZS 4777.2: 2020 Australia A Region 0 kW Export Limit:1.2.2

As the profiles are only valid for the local setup then some configuration of the profiles would be required for each home assistant/enphase instance.

This feature might be good to have something like a on/off limit_profile option (like for the power production) and a way to configure an on_profile and an off_profile (i.e. setting the two grid profile names in the user insterface for the custom component).

An alternative might be to support a callable service for set_profile so that control is handed back to the user and they can customise and call a service provided by this custom_component from other scripts on their HA system (triggered on appropriate events).

**Account type**
- [X] Installer
- [X] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [X] Metered with CTs installed
- [X] Metered without CTs
- [X] Standard

- FW version: D7++

vincentwolsink, 2023-10-31:
Hi @brodykenrick. Thanks for the suggestion and information. I agree this will be a nice to have feature and will take a look at it as soon I have some spare time 🙂

vincentwolsink, 2023-11-02:
I implemented services to get/set profiles and an extra sensor in 0.3.5-beta1. Please let me know if this works for you.

brodykenrick, 2023-11-02:
Thanks! I'll test this weekend.

brodykenrick, 2023-11-03:
@vincentwolsink It works amazingly well (all confirmed in just a few minutes). Thankyou!

[image]

The "get grid profile" service returned all my configured profiles and copying the desired profile text in to the "set grid profile" service also worked smoothly (there is a slight pause as it was waiting to be confirmed accepted - as you would expect).

I think there is a slight caching period in the "get grid profile" service call (or in the enphase gateway?) as the old selected profile was returned for about 30 seconds after the change was made.

Thanks again. Let me know if you'd like any documentation updates done or anything else.

brodykenrick, 2023-11-03:
@vincentwolsink Do you have a buymeacoffee account or similar? As I'd love to buy you a coffee/beer/plant to say thanks for this feature that is going to be very helpful to me and other home assistant/enphase users in Australia (and no doubt in many other places also).

vincentwolsink, 2023-11-03:
Great to hear! I added an extra refresh call to make the sensor/get service call update quicker.

And thank you for the coffee/beer offer! 😃 

[image]

closed 2023-11-03 by vincentwolsink