vincentwolsink/home_assistant_enphase_envoy_installer #100: Config flow could not be loaded

githubkab, 2023-11-07:
**Describe the bug**
When adding integration following message appears: 
Config flow could not be loaded: {"message":"Invalid handler specified"}

**Home Assistant**
- Version: 2023.6.2

**Additional context**
```
![image](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/assets/76258060/f4eb1d02-cf57-4f07-8d11-c25eb640efcd)
```

vincentwolsink, 2023-11-07:
Do you maybe have a different Enphase plugin currently configured (HA built-in or other version from HACS)? Can you check the Home Assistant error logs for more context?

githubkab, 2023-11-08:
I did have other enphase plug-ins installed, but they are deleted. See below homeassistant log:

```
2023-11-08 09:16:35.716 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-08 09:16:35.717 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration enphase_envoy which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-08 09:21:36.785 ERROR (MainThread) [frontend.js.latest.202306080] http://xxx.local:8123/config/integrations/dashboard:0:0 ResizeObserver loop completed with undelivered notifications.
2023-11-08 09:21:39.855 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration enphase_envoy: cannot import name 'ServiceResponse' from 'homeassistant.core' (/usr/src/homeassistant/homeassistant/core.py)

```

vincentwolsink, 2023-11-08:
Ah. This integration provides a servicecall with data response. This was implemented in Home Assistant 2023.7. You are running an older version. Please upgrade if you want to use this integration.

I will update the requirements.

githubkab, 2023-11-08:
Works!

closed 2023-11-08 by githubkab