vincentwolsink/home_assistant_enphase_envoy_installer #102: [BUG] Incorrect Status Display for switch.production After Turning Off in Home Assistant

Neowam, 2024-01-13:
**Describe the bug**
I am using the latest version of [home_assistant_enphase_envoy_installer](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer). I've encountered an issue with the synchronization of switch status between my Enphase controller and the switch.production sensor in Home Assistant.

**Account type**
- [X] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [ X] Standard

- FW version: D7.6.175 (f79c8d)
- Amount of micro inverters connected: 25

**Home Assistant**
Core 2024.1.3
Supervisor 2023.12.0
Operating System 11.4
Frontend 20240104.0

**Additional context**
```
I created a switch from the switch.production sensor in Home Assistant.
When I turn off this switch, I observe the following behavior:

- The Enphase controller goes completely offline and becomes accessible again after about 5 minutes.
- Via https://192.168.178.12/installer/setup/home, I can confirm that the production switch is indeed turned off.
- In Home Assistant, the sensor sensor.envoy_122227045337_current_power_production correctly drops to 0.
- 

However, the switch.production remains on ON and does not get updated or refreshed.
After half an hour, the Enphase controller is still OFF, but the switch.production switch still remains on ON.

![image](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/assets/94406673/a2af01ca-f039-47c1-a790-b94c8234e91b)
![image](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/assets/94406673/3549d6c6-a30a-48a9-98b7-178b9bb97294)
![image](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/assets/94406673/4a6330e5-03e4-46e4-8e23-87d03edb64fd)
![image](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/assets/94406673/cd79399e-c9f5-4ff3-b082-a9f03779e0f6)

Expected Result:
When the switch.production switch in Home Assistant is turned off, its status should correctly reflect in both the Enphase controller and in Home Assistant.

Actual Result:
The status of the switch.production switch in Home Assistant remains on ON, despite the Enphase controller being actually turned off. This suggests an issue in reading the switch status on the Enphase controller compared to the switch.production sensor in Home Assistant.
```

vincentwolsink, 2024-01-13:
The Envoy becoming slow/unavailable for some time unfortunately happens when triggering production on/off.

What I can think of causing this bug is that we cache the state of the production switch for an hour. It actively reloads the state after switching, but it might be that the Envoy needs some time to update and HA caches the old state for an hour.

I will create a beta version with a reduced cache time. Try that one and let me know if it behaves better.

Neowam, 2024-01-13:
> The Envoy becoming slow/unavailable for some time unfortunately happens when triggering production on/off.
>
> What I can think of causing this bug is that we cache the state of the production switch [...]

@vincentwolsink many thanks! I will test this case for you with pleasure. Let me know when you have a beta version ready of the code.

vincentwolsink, 2024-01-13:
[0.3.6-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.3.6-beta1)

Neowam, 2024-01-13:
Will try coming few days...

Neowam, 2024-02-17:
May be closed, I think this works fine now.

closed 2024-02-17 by vincentwolsink