vincentwolsink/home_assistant_enphase_envoy_installer #104: switch.production entity does not toggle correct - it swiches then it returns to ariginal state[BUG]

potterkop, 2024-01-17:
**Describe the bug**
A clear and concise description of what the bug is.

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: x

**Home Assistant**
- Version: 2023.x

**Additional context**
```
Relevant snippet of Home Assistant error log. 
```

potterkop, 2024-01-17:
> Please check #102
> <vincentwolsink/home_assistant_enphase_envoy_installer/issues/104/1895769092
> @github.com>

Thanks for your response ,

did the lower caching time resolve the issue?

Regards

Frans

vincentwolsink, 2024-01-17:
Try the version mentioned in that issue and you tell me 🙂