vincentwolsink/home_assistant_enphase_envoy_installer #105: [BUG] Instellen mislukt

Hemmik, 2024-01-25:
**Describe the bug**
Ik had de normale enphase integratie al ingesteld en probeer nu deze te installeren. Krijg de melding dat instellen mislukt is. Bij de normale integratie was ik ingelogd met mijn home owner account, ik heb tevens ook een installer account. Is de enige manier om dit aan het werk te krijgen de oorspronkelijke configuratie eerst verwijderen? Is er een manier om dit te doen zonder al mijn eerdere data (geschiedenis) te verwijderen? 

**Account type**
- [x ] Installer
- [ ] DIY / DHZ
- [ x] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [ x] Standard

- FW version: D7.6.168 (f1975b)
- Amount of micro inverters connected: 3

**Home Assistant**
- Version: 2024.1.5

**Additional context**
```
Relevant snippet of Home Assistant error log. 
Logger: homeassistant.config_entries
Source: config_entries.py:406
First occurred: 21:20:03 (2 occurrences)
Last logged: 21:20:24

Error setting up entry Envoy 122329089085 for enphase_envoy
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 406, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 77, in async_setup_entry
    enlighten_serial_num=config[CONF_SERIAL],
                         ~~~~~~^^^^^^^^^^^^^
KeyError: 'serial'

vincentwolsink, 2024-01-25:
De originele Home Assistant integratie is onlangs herschreven, waardoor de configuratie hiervan helaas niet meer 1 op 1 compatible is. De entiteit/sensoren namen zouden nog wel overeen moeten komen, waardoor historie behouden blijft. Je kunt dus de huidige integratie eerst verwijderen en deze daarna configureren zonder dat je geschiedenis kwijt raakt (tenzij je met de hand entiteit namen hebt aangepast, dit moet je dan opnieuw doen).

Hemmik, 2024-01-26:
Top, het is gelukt! Had de oorspronkelijke namen behouden in de originele integratie maar moest ze alsnog aanpassen. Komen niet helemaal overeen. Zowel productie per inverter als totale productie was net anders. Dank voor je hulp!

closed 2024-01-26 by Hemmik