vincentwolsink/home_assistant_enphase_envoy_installer #106: [BUG] not return 0 at midnight

berserk14350, 2024-01-30:
Hi,
i don't understand why my production not restart to 0 at midnight. each day start with 1,5kwH

[image]

**Account type**
-Envoy-S Metered (with CT)
par Enphase
Micrologiciel : D7.6.168

Home Assistant Core
par Home Assistant
Version : 2024.1.5

thank's a lot

vincentwolsink, 2024-01-30:
Hi @berserk14350, do you see the same in your Enphase app?

berserk14350, 2024-01-30:
no just in HA......

vincentwolsink, 2024-01-30:
Is the number correct at the end of the day or also 1,5 kwh higher?

berserk14350, 2024-01-30:
always 1.5 kWh more

berserk14350, 2024-01-31:
Hi,

This IS my log, i think i have a problem with the var whtoday?

``
temperature': 0, 'dc_voltage': 0.0, 'dc_current': 0.0, 'ac_voltage': 0.0, 'ac_power': 0.0}]
2024-01-31 07:17:05.007 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_inventory_results.[?(@.type=='PCU')].devices[?(@.dev_type==1)]
2024-01-31 07:17:05.199 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] EnvoyData.get(inverters_info) -> {'482324127480': {'part_num': '800-01396-r03', 'installed': '1703246243', 'serial_num': '482324127480', 'device_status': ['envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706634162', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246243', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627390225, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324127628': {'part_num': '800-01396-r03', 'installed': '1703246243', 'serial_num': '482324127628', 'device_status': ['envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706633980', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246243', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627390481, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324127726': {'part_num': '800-01396-r03', 'installed': '1703246245', 'serial_num': '482324127726', 'device_status': ['envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706633919', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246245', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627390737, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324127284': {'part_num': '800-01396-r03', 'installed': '1703246245', 'serial_num': '482324127284', 'device_status': ['envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706634010', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246245', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627390993, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324127383': {'part_num': '800-01396-r03', 'installed': '1703246245', 'serial_num': '482324127383', 'device_status': ['envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706633949', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246245', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627391249, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324128130': {'part_num': '800-01396-r03', 'installed': '1703246245', 'serial_num': '482324128130', 'device_status': ['envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706634041', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246245', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627391505, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324128107': {'part_num': '800-01396-r03', 'installed': '1703246245', 'serial_num': '482324128107', 'device_status': ['envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '0', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246245', 'img_load_date': '0', 'img_pnum_running': 'undetermined', 'ptpn': 'undetermined', 'chaneid': 1627391761, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324128129': {'part_num': '800-01396-r03', 'installed': '1703246246', 'serial_num': '482324128129', 'device_status': ['envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706634073', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246246', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627392017, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324128126': {'part_num': '800-01396-r03', 'installed': '1703246247', 'serial_num': '482324128126', 'device_status': ['envoy.cond_flags.pcu_chan.dcvoltagetoolow', 'envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706634196', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246247', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627392273, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}, '482324127285': {'part_num': '800-01396-r03', 'installed': '1703246247', 'serial_num': '482324127285', 'device_status': ['envoy.cond_flags.pcu_chan.dcvoltagetoolow', 'envoy.cond_flags.obs_strs.failure'], 'last_rpt_date': '1706634165', 'admin_state': 2, 'dev_type': 1, 'created_date': '1703246247', 'img_load_date': '0', 'img_pnum_running': '521-00005-r05-v06.05.02', 'ptpn': '549-00072-r01-v06.05.05', 'chaneid': 1627392529, 'device_control': [{'gficlearset': False}], 'producing': False, 'communicating': False, 'provisioned': False, 'operating': False}}
2024-01-31 07:17:05.201 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_devstatus.pcu[?(@.devType==1)]
2024-01-31 07:17:05.211 DEBUG (

Thank s

vincentwolsink, 2024-01-31:
Hi @berserk14350, your log output is clipped and doesn't contain `whtoday`.

berserk14350, 2024-01-31:
So i'm a loser......

i'm a beginner in HA and my english is poor.....

this is a log :

`2024-01-31 07:17:01.647 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.40/production.json?details=1: <Response [200 OK]>: {"production":[{"type":"inverters","activeCount":10,"readingTime":1706681808,"wNow":0,"whLifetime":129936},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1706681831,"wNow":-3.171,"whLifetime":128604.343,"varhLeadLifetime":41758.145,"varhLagLifetime":103673.09,"vahLifetime":253384.132,"rmsCurrent":0.703,"rmsVoltage":237.632,"reactPwr":165.074,"apprntPwr":166.877,"pwrFactor":-0.03,"whToday":1421.343,"whLastSevenDays":35309.343,"vahToday":1207.132,"varhLeadToday":0.145,"varhLagToday":1196.09,"lines":[{"wNow":-3.171,"whLifetime":128604.343,"varhLeadLifetime":41758.145,"varhLagLifetime":103673.09,"vahLifetime":253384.132,"rmsCurrent":0.703,"rmsVoltage":237.632,"reactPwr":165.074,"apprntPwr":166.877,"pwrFactor":-0.03,"whToday":1421.343,"whLastSevenDays":35309.343,"vahToday":1207.132,"varhLeadToday":0.145,"varhLagToday":1196.09}]}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1706681831,"wNow":946.268,"whLifetime":1460902.752,"varhLeadLifetime":253425.372,"varhLagLifetime":-58232.017,"vahLifetime":1496893.137,"rmsCurrent":5.379,"rmsVoltage":237.723,"reactPwr":-356.136,"apprntPwr":1278.706,"pwrFactor":0.74,"whToday":8660.752,"whLastSevenDays":31.752,"vahToday":9895.137,"varhLeadToday":2989.372,"varhLagToday":0.0,"lines":[{"wNow":946.268,"whLifetime":1460902.752,"varhLeadLifetime":253425.372,"varhLagLifetime":-58232.017,"vahLifetime":1496893.137,"rmsCurrent":5.379,"rmsVoltage":237.723,"reactPwr":-356.136,"apprntPwr":1278.706,"pwrFactor":0.74,"whToday":8660.752,"whLastSevenDays":31.752,"vahToday":9895.137,"varhLeadToday":2989.372,"varhLagToday":0.0}]},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1706681831,"wNow":949.439,"whLifetime":1332576.779,"varhLeadLifetime":295183.517,"varhLagLifetime":45441.073,"vahLifetime":1496893.137,"rmsCurrent":4.676,"rmsVoltage":237.723,"reactPwr":-521.21,"apprntPwr":1111.572,"pwrFactor":0.85,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0,"lines":[{"wNow":949.439,"whLifetime":1332576.779,"varhLeadLifetime":295183.517,"varhLagLifetime":45441.073,"vahLifetime":1496893.137,"rmsCurrent":4.676,"rmsVoltage":237.723,"reactPwr":-521.21,"apprntPwr":1111.572,"pwrFactor":0.85,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}]}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}
`

berserk14350, 2024-01-31:
when i turn off the metered and turn on, my production return a 1426wh

`EnvoyData.get(daily_production_l1) -> 1426.56`

vincentwolsink, 2024-01-31:
Weird. The values in the debug log are coming directly from the Envoy. Could it be your CT clamp is mounted wrong and the app/enlighten somehow compensates for this? In the morning try to visit the Envoy web interface (via the button in HA) and check the value there.

berserk14350, 2024-01-31:
my clamp is mounted correctly, it's not a problem with a clamp, i think metered send not the good value .....

berserk14350, 2024-02-01:
hi, for you're information:

i tri to get the information directly on metered in the same time by three url:

https://envoy.local/api/v1/production
`
{
  "wattHoursToday": 1857,
  "wattHoursSevenDays": 37478,
  "wattHoursLifetime": 137125,
  "wattsNow": 603
}
`

https://192.168.1.40/ivp/pdm/energy

`{
    "production": {
        "pcu": {
            "wattHoursToday": 1907,
            "wattHoursSevenDays": 38390,
            "wattHoursLifetime": 140125,
            "wattsNow": 557
        },
        "rgm": {
            "wattHoursToday": 0,
            "wattHoursSevenDays": 0,
            "wattHoursLifetime": 0,
            "wattsNow": 0
        },
        "eim": {
            "wattHoursToday": 1857,
            "wattHoursSevenDays": 37478,
            "wattHoursLifetime": 137125,
            "wattsNow": 603
        }
    },
    "consumption": {
        "eim": {
            "wattHoursToday": 8557,
            "wattHoursSevenDays": 0,
            "wattHoursLifetime": 1489462,
            "wattsNow": 708
        }
    }
}`

https://envoy.local/production.json?details=1

`{"production":[{"type":"inverters","activeCount":10,"readingTime":1706796736,"wNow":596,"whLifetime":140198},{"type":"eim","activeCount":1,"measurementType":"production","readingTime":1706796771,"wNow":527.394,"whLifetime":138712.313,"varhLeadLifetime":45127.336,"varhLagLifetime":106475.779,"vahLifetime":267303.396,"rmsCurrent":2.374,"rmsVoltage":241.019,"reactPwr":-166.512,"apprntPwr":572.316,"pwrFactor":0.92,"whToday":3444.313,"whLastSevenDays":39065.313,"vahToday":3855.396,"varhLeadToday":597.336,"varhLagToday":1521.779,"lines":[{"wNow":527.394,"whLifetime":138712.313,"varhLeadLifetime":45127.336,"varhLagLifetime":106475.779,"vahLifetime":267303.396,"rmsCurrent":2.374,"rmsVoltage":241.019,"reactPwr":-166.512,"apprntPwr":572.316,"pwrFactor":0.92,"whToday":3444.313,"whLastSevenDays":39065.313,"vahToday":3855.396,"varhLeadToday":597.336,"varhLagToday":1521.779}]}],"consumption":[{"type":"eim","activeCount":1,"measurementType":"total-consumption","readingTime":1706796771,"wNow":663.426,"whLifetime":1489634.447,"varhLeadLifetime":258363.385,"varhLagLifetime":-56313.397,"vahLifetime":1526667.361,"rmsCurrent":5.284,"rmsVoltage":240.956,"reactPwr":319.184,"apprntPwr":1273.273,"pwrFactor":0.52,"whToday":8729.447,"whLastSevenDays":172.447,"vahToday":10896.361,"varhLeadToday":3794.0,"varhLagToday":0.0,"lines":[{"wNow":663.426,"whLifetime":1489634.447,"varhLeadLifetime":258363.385,"varhLagLifetime":-56313.397,"vahLifetime":1526667.361,"rmsCurrent":5.284,"rmsVoltage":240.956,"reactPwr":319.184,"apprntPwr":1273.273,"pwrFactor":0.52,"whToday":8729.447,"whLastSevenDays":172.447,"vahToday":10896.361,"varhLeadToday":3794.0,"varhLagToday":0.0}]},{"type":"eim","activeCount":1,"measurementType":"net-consumption","readingTime":1706796771,"wNow":136.032,"whLifetime":1351200.504,"varhLeadLifetime":303490.721,"varhLagLifetime":50162.382,"vahLifetime":1526667.361,"rmsCurrent":2.91,"rmsVoltage":240.956,"reactPwr":485.696,"apprntPwr":701.298,"pwrFactor":0.19,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0,"lines":[{"wNow":136.032,"whLifetime":1351200.504,"varhLeadLifetime":303490.721,"varhLagLifetime":50162.382,"vahLifetime":1526667.361,"rmsCurrent":2.91,"rmsVoltage":240.956,"reactPwr":485.696,"apprntPwr":701.298,"pwrFactor":0.19,"whToday":0,"whLastSevenDays":0,"vahToday":0,"varhLeadToday":0,"varhLagToday":0}]}],"storage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}`

The Json is out....... whtoday

wait for update metered, it's a better idea?

thank's a lot

vincentwolsink, 2024-02-02:
You could open a ticket at Enphase support. Maybe they can correct/find the cause of this incorrect output.

berserk14350, 2024-02-28:
hi this is the answer:

[https://support.enphase.com/s/question/0D5Ps00000Axen2KAB/metered-api-wrong-value](url)

and i'm not understand......

Calyhre, 2024-03-05:
I have the same issue. My installation is a bit more than a month old, and the reset value of `whToday` is slowly growing over time. Every day the value is drifting a bit more.

I'm not sure what they tried to explain on the community support, but they suggest to use another endpoint.

vincentwolsink, 2024-03-05:
Unfortunately I do not see any answer to this issue in the thread. They are talking about the difference between realtime data and data generated every 5 min. The issues here is that apparently the Envoy fails to calculate the `whToday` value correctly.

I will take a look at potentially using a different endpoint for this value. Although every endpoint on the Envoy has its quirks, and also not all of them are available for home owners. Stay tuned.

vincentwolsink, 2024-03-05:
@berserk14350 @Calyhre Are you both using an Installer/DIY login? Or homeowner?

Calyhre, 2024-03-05:
> @berserk14350 @Calyhre Are you both using an Installer/DIY login? Or homeowner?

I'm using a homeowner login

berserk14350, 2024-03-05:
I'm not sur for the good answer:

The installation is make by mylself

vincentwolsink, 2024-03-05:
Could both of you please paste the output of https://envoy.local/ivp/meters and https://envoy.local/ivp/meters/readings (after logging in)?

Calyhre, 2024-03-05:
Sure, there you go:

<details>
<summary>Results for /ivp/meters</summary>

```json
[
    {
        "eid": 704643328,
        "state": "enabled",
        "measurementType": "production",
        "phaseMode": "three",
        "phaseCount": 1,
        "meteringStatus": "normal",
        "statusFlags": []
    },
    {
        "eid": 704643584,
        "state": "enabled",
        "measurementType": "net-consumption",
        "phaseMode": "three",
        "phaseCount": 1,
        "meteringStatus": "normal",
        "statusFlags": []
    }
]
```

</details>

<details>
<summary>Results for /ivp/meters/readings</summary>

```json
[
    {
        "eid": 704643328,
        "timestamp": 1709665769,
        "actEnergyDlvd": 369884.193,
        "actEnergyRcvd": 920.623,
        "apparentEnergy": 498457.380,
        "reactEnergyLagg": 74294.427,
        "reactEnergyLead": 134158.431,
        "instantaneousDemand": -1.170,
        "activePower": -1.170,
        "apparentPower": 130.808,
        "reactivePower": 129.167,
        "pwrFactor": 0.000,
        "voltage": 232.556,
        "current": 0.562,
        "freq": 50.000,
        "channels": [
            {
                "eid": 1778385169,
                "timestamp": 1709665769,
                "actEnergyDlvd": 369884.193,
                "actEnergyRcvd": 920.623,
                "apparentEnergy": 498457.380,
                "reactEnergyLagg": 74294.427,
                "reactEnergyLead": 134158.431,
                "instantaneousDemand": -1.170,
                "activePower": -1.170,
                "apparentPower": 130.808,
                "reactivePower": 129.167,
                "pwrFactor": 0.000,
                "voltage": 232.556,
                "current": 0.562,
                "freq": 50.000
            },
            {
                "eid": 1778385170,
                "timestamp": 1709665769,
                "actEnergyDlvd": 0.001,
                "actEnergyRcvd": 1407.920,
                "apparentEnergy": 1431.877,
                "reactEnergyLagg": 72.576,
                "reactEnergyLead": 0.115,
                "instantaneousDemand": -0.322,
                "activePower": -0.322,
                "apparentPower": 1.422,
                "reactivePower": -0.000,
                "pwrFactor": 0.000,
                "voltage": 11.592,
                "current": 0.123,
                "freq": 50.000
            },
            {
                "eid": 1778385171,
                "timestamp": 1709665769,
                "actEnergyDlvd": 0.001,
                "actEnergyRcvd": 2576.369,
                "apparentEnergy": 1386.644,
                "reactEnergyLagg": 72.765,
                "reactEnergyLead": 0.113,
                "instantaneousDemand": -0.661,
                "activePower": -0.661,
                "apparentPower": 1.365,
                "reactivePower": 0.000,
                "pwrFactor": 0.000,
                "voltage": 7.463,
                "current": 0.183,
                "freq": 50.000
            }
        ]
    },
    {
        "eid": 704643584,
        "timestamp": 1709665769,
        "actEnergyDlvd": 559746.179,
        "actEnergyRcvd": 163231.565,
        "apparentEnergy": 1003246.642,
        "reactEnergyLagg": 27453.687,
        "reactEnergyLead": 484116.521,
        "instantaneousDemand": 1752.715,
        "activePower": 1752.715,
        "apparentPower": 1978.539,
        "reactivePower": -814.910,
        "pwrFactor": 0.831,
        "voltage": 232.523,
        "current": 8.533,
        "freq": 50.000,
        "channels": [
            {
                "eid": 1778385425,
                "timestamp": 1709665769,
                "actEnergyDlvd": 559746.179,
                "actEnergyRcvd": 163231.565,
                "apparentEnergy": 1003246.642,
                "reactEnergyLagg": 27453.687,
                "reactEnergyLead": 484116.521,
                "instantaneousDemand": 1752.715,
                "activePower": 1752.715,
                "apparentPower": 1978.539,
                "reactivePower": -814.910,
                "pwrFactor": 0.831,
                "voltage": 232.523,
                "current": 8.533,
                "freq": 50.000
            },
            {
                "eid": 1778385426,
                "timestamp": 1709665769,
                "actEnergyDlvd": 0.000,
                "actEnergyRcvd": 1498.401,
                "apparentEnergy": 310.984,
                "reactEnergyLagg": 76.497,
                "reactEnergyLead": 0.007,
                "instantaneousDemand": -0.071,
                "activePower": -0.071,
                "apparentPower": 0.307,
                "reactivePower": 0.000,
                "pwrFactor": -1.000,
                "voltage": 1.723,
                "current": 0.178,
                "freq": 50.000
            },
            {
                "eid": 1778385427,
                "timestamp": 1709665769,
                "actEnergyDlvd": 0.001,
                "actEnergyRcvd": 2662.785,
                "apparentEnergy": 22.191,
                "reactEnergyLagg": 76.988,
                "reactEnergyLead": 0.189,
                "instantaneousDemand": -0.000,
                "activePower": -0.000,
                "apparentPower": -0.000,
                "reactivePower": -0.000,
                "pwrFactor": -1.000,
                "voltage": 6.004,
                "current": -0.000,
                "freq": 50.000
            }
        ]
    }
]
```

</details>

berserk14350, 2024-03-05:
results for /ivp/meters

`[
    {
        "eid": 704643328,
        "state": "enabled",
        "measurementType": "production",
        "phaseMode": "three",
        "phaseCount": 1,
        "meteringStatus": "normal",
        "statusFlags": []
    },
    {
        "eid": 704643584,
        "state": "enabled",
        "measurementType": "net-consumption",
        "phaseMode": "three",
        "phaseCount": 1,
        "meteringStatus": "normal",
        "statusFlags": []
    }
]`
results for /ivp/meters/readings:

`[
    {
        "eid": 704643328,
        "timestamp": 1709666827,
        "actEnergyDlvd": 282107.709,
        "actEnergyRcvd": 2577.267,
        "apparentEnergy": 514945.397,
        "reactEnergyLagg": 186799.406,
        "reactEnergyLead": 97472.523,
        "instantaneousDemand": -2.420,
        "activePower": -2.420,
        "apparentPower": 169.460,
        "reactivePower": 167.198,
        "pwrFactor": -0.043,
        "voltage": 238.963,
        "current": 0.709,
        "freq": 50.125,
        "channels": [
            {
                "eid": 1778385169,
                "timestamp": 1709666827,
                "actEnergyDlvd": 282107.709,
                "actEnergyRcvd": 2577.267,
                "apparentEnergy": 514945.397,
                "reactEnergyLagg": 186799.406,
                "reactEnergyLead": 97472.523,
                "instantaneousDemand": -2.420,
                "activePower": -2.420,
                "apparentPower": 169.460,
                "reactivePower": 167.198,
                "pwrFactor": -0.043,
                "voltage": 238.963,
                "current": 0.709,
                "freq": 50.125
            },
            {
                "eid": 1778385170,
                "timestamp": 1709666827,
                "actEnergyDlvd": 0.003,
                "actEnergyRcvd": 3401.956,
                "apparentEnergy": 1726.329,
                "reactEnergyLagg": 134.243,
                "reactEnergyLead": 0.338,
                "instantaneousDemand": -0.376,
                "activePower": -0.376,
                "apparentPower": 0.980,
                "reactivePower": -0.000,
                "pwrFactor": 0.000,
                "voltage": 6.333,
                "current": 0.153,
                "freq": 50.125
            },
            {
                "eid": 1778385171,
                "timestamp": 1709666827,
                "actEnergyDlvd": 0.000,
                "actEnergyRcvd": 5291.436,
                "apparentEnergy": 528.848,
                "reactEnergyLagg": 132.793,
                "reactEnergyLead": 0.000,
                "instantaneousDemand": -0.166,
                "activePower": -0.166,
                "apparentPower": 0.303,
                "reactivePower": 0.000,
                "pwrFactor": -1.000,
                "voltage": 2.171,
                "current": 0.141,
                "freq": 50.125
            }
        ]
    },
    {
        "eid": 704643584,
        "timestamp": 1709666827,
        "actEnergyDlvd": 2097414.837,
        "actEnergyRcvd": 17071.287,
        "apparentEnergy": 2443326.472,
        "reactEnergyLagg": 109229.159,
        "reactEnergyLead": 536617.936,
        "instantaneousDemand": 1439.666,
        "activePower": 1439.666,
        "apparentPower": 1469.572,
        "reactivePower": -227.860,
        "pwrFactor": 0.975,
        "voltage": 239.043,
        "current": 6.152,
        "freq": 50.062,
        "channels": [
            {
                "eid": 1778385425,
                "timestamp": 1709666827,
                "actEnergyDlvd": 2097414.837,
                "actEnergyRcvd": 17071.287,
                "apparentEnergy": 2443326.472,
                "reactEnergyLagg": 109229.159,
                "reactEnergyLead": 536617.936,
                "instantaneousDemand": 1439.666,
                "activePower": 1439.666,
                "apparentPower": 1469.572,
                "reactivePower": -227.860,
                "pwrFactor": 0.975,
                "voltage": 239.043,
                "current": 6.152,
                "freq": 50.062
            },
            {
                "eid": 1778385426,
                "timestamp": 1709666827,
                "actEnergyDlvd": 0.003,
                "actEnergyRcvd": 1535.303,
                "apparentEnergy": 1702.401,
                "reactEnergyLagg": 130.918,
                "reactEnergyLead": 0.248,
                "instantaneousDemand": -0.157,
                "activePower": -0.157,
                "apparentPower": 0.962,
                "reactivePower": 0.000,
                "pwrFactor": -1.000,
                "voltage": 4.892,
                "current": 0.197,
                "freq": 50.062
            },
            {
                "eid": 1778385427,
                "timestamp": 1709666827,
                "actEnergyDlvd": 0.003,
                "actEnergyRcvd": 3247.001,
                "apparentEnergy": 712.756,
                "reactEnergyLagg": 130.977,
                "reactEnergyLead": 0.310,
                "instantaneousDemand": -0.124,
                "activePower": -0.124,
                "apparentPower": 0.404,
                "reactivePower": -0.000,
                "pwrFactor": 0.000,
                "voltage": 5.452,
                "current": 0.074,
                "freq": 50.062
            }
        ]
    }
]`

vincentwolsink, 2024-03-08:
I can switch endpoint for the daily energy usage. But I haven't found an alternative endpoint yet for the daily energy usage per phase. If the two potentially return different values that will be very confusing.

catsmanac, 2024-03-13:
This issue of not returning to zero is reported in other Envoy integrations as well.  For energy dashboard best use lifetime values. To get a daily value you can opt to configure a HA utility meter sourced from lifetime value to work around this issue.

vincentwolsink, 2024-03-13:
I'm thinking about adding a configuration option to use the alternative daily endpoint. As a result you will lose the daily phase readings, but the daily total reading will be correct.

vincentwolsink, 2024-03-13:
A solution with the utility meter @catsmanac mentioned I previously described here: https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/issues/40#issuecomment-1633879122
This is also a perfectly working solution. And indeed, for the Energy dashboard use lifetime values.

closed 2024-11-27 by vincentwolsink