vincentwolsink/home_assistant_enphase_envoy_installer #110: Cannot use Enlighten manager account to switch off production

rhannink, 2024-03-12:
**Describe the bug**
A clear and concise description of what the bug is.

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [ *] Enlighten manager account
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ *] Metered without CTs
- [ ] Standard

- FW version: D7.6.175
- Amount of micro inverters connected: 19

**Home Assistant**
- Version: 2024.3

**Description of bug**

Hello, I only have a Home owner and an Enlighten manager account. It is not possible to transfer my installer account to my self.
In the Enlighten manager website, I can enable/disable the power production.
Why is it not possible to enable/disable this with this plugin using the Manager account?
If I use my manager account as an installer account, the plugin does not start.
If I use my home owner account and set the home owner option everything works fine but I cannot control the power production.

Kind regards,

Remco Hannink

vincentwolsink, 2024-03-12:
Hi @rhannink, I am only aware of three account types: Installer (company), DIY and Homeowner. If you as home owner upgraded to be able to use Enlighten Manager I would expect your account to be of the DIY type. But you have a separate account for Enlighten Manager?

Did you try using your home owner account without ticking the home owner option? Otherwise please share the relevant error log from your home assistant system logs when the plugin does not start.

suave123, 2024-03-12:
Hi Vincent,

I have a similar issue, I am home owner but also made an installer account which I connected to my home owner account as "onderhoudsmonteur". I can login and see lots of integrations and devices but I cant see the production turn on / off switch

vincentwolsink, 2024-03-12:
@suave123 Can you maybe share your diagnostics file?

suave123, 2024-03-13:
Hi Vincent, I am kind of a novice at HA, where can I find that?

vincentwolsink, 2024-03-13:
[image]

Press the three vertical dots next to your Envoy and then Download Diagnostics.

suave123, 2024-03-13:
[config_entry-enphase_envoy-24a0ea0fa1848573a7257d92905be264.json](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/14585083/config_entry-enphase_envoy-24a0ea0fa1848573a7257d92905be264.json)

vincentwolsink, 2024-03-13:
@suave123 The user you are logged in with does not have Installer/DIY rights to the Envoy. So you can only access home owner data/features.

suave123, 2024-03-13:
ok clear, I am connected with the same user as in my installer app, where I can see the production on/off switch....so thats strange, where can I see / alter login info for the integration?

vincentwolsink, 2024-03-13:
Which app are you talking about here? And can you make a screenshot of the production switch?

suave123, 2024-03-13:
wait I solved it!

I have reloaded the application and now it shows 204 entities instead of 50 ish....also production is visible now

suave123, 2024-03-13:
[config_entry-enphase_envoy-24a0ea0fa1848573a7257d92905be264 (1).json](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/14585439/config_entry-enphase_envoy-24a0ea0fa1848573a7257d92905be264.1.json)

suave123, 2024-03-13:
maybe you can see the difference now....this might also be happening for other people, pressing RELOAD fixed it...

vincentwolsink, 2024-03-14:
@suave123 Everything seems ok now!

@rhannink Still experiencing the issue? Please provide the error logs. :)