vincentwolsink/home_assistant_enphase_envoy_installer #112: Production switch

Ayuus, 2024-03-13:
**Describe the bug**

Production switch seems to work fine in integration. However, after switching production on again (more than 1 hour after switching off, during daylight), production does not resume, even after several hours. Switch is in state ON, but no power anymore.
However, next day, production starts normally again.

**Account type**
- [ ] Installer

**Envoy**
- [ ] Metered without CTs

- FW version: D7.6.175
- Amount of micro inverters connected: 17

**Home Assistant**
- Version: 2024.3

**Additional context**
```
none

Relevant snippet of Home Assistant error log. 
```
none

erjonkers, 2024-03-13:
Same issue! This might be a Enphase issue though... 

https://support.enphase.com/s/question/0D53m00009JkKntCAF/when-i-disable-and-later-enable-power-production-through-the-envoy-installer-web-interface-the-microinverters-only-start-producing-power-the-next-morning-instead-of-right-away-why-is-that-and-is-there-any-way-to-make-them-immediately-produce-power

vincentwolsink, 2024-03-13:
Hi, this is an known Enphase issue unfortunately. Nothing I can do about that in this integration.

erjonkers, 2024-03-13:
Thanks a lot Vincent, would you by any chance know what Enphase has shared about the issue and resolution timeline...?

Can't find it anywhere..

erjonkers, 2024-03-14:
Thanks Vincent, I've (re)raised the issue in the thread with Enphase I've shared above, hopefully we can get some intel on issue + fix timelines

erjonkers, 2024-03-15:
Hi Vincent, I saw you also chimed in on the thread with Enphase. Thanks for that, I'm sure the extra nudge has helped the case forward.

You wrote that you are able to restart production by triggering a 'manual AC power cycle'. Silly question but how would you do that? By switching of the breakers I guess? And which ones, would all of the solar panel breakers would do the trick?

vincentwolsink, 2024-03-15:
Switching the breaker indeed. My Envoy and inverters are on the same one, but I assume it is about the inverters.

kctje, 2024-04-03:
maybe dissable this function till enphase fixed this problem?

vincentwolsink, 2024-04-04:
Hi @kctje, this function is not broken for everyone. I can add some documentation or something to the readme.

kctje, 2024-04-04:
> <vincentwolsink/home_assistant_enphase_envoy_installer/issues/112/2036394610
> @github.com>

Maybe thats a good option

Kudos50, 2024-05-15:
Please note that resetting a breaker also does not always fix this issue. I have two installations and eventually (hours) flipped the breakers. Only a handful of micro's came back online on both installations. Next day, all is well.
So I guess all we can do is wait on Enphase whereas SolarEdge nowadays even has the "negative pricing disable feature" fully integrated in their mobile app for users.

ranjowd, 2024-08-29:
This week I got version D8.2.4264 on my Envoy-s metered without Current Transformers. Since then it seems that the production switch also enables production back on the same day. I just tested this on the end of the day and saw production dropping to 0 in around 10 minutes. After turning the switch back on, it also took another 10 minutes to get back on. 

Tomorrow I will try this at the highest production point during the day. So far I did not want to try that out..

Kudos50, 2024-09-16:
Fixed here as well but it did require a firmware update to the Micro inverters (IQ7+). In fact, the new firmware on the envoy caused the production stop - start (next day start issue) to fail completely. Nothing happened. I was unable to update the inverters myself but a simple support ticket to Enphase did the trick. Update from 4.27.04 to 4.28.07.

Stop start for me is not 10mins as mentioned here. But more like 30mins stop en 50mins start. But that’s OK.

closed 2024-09-17 by vincentwolsink