vincentwolsink/home_assistant_enphase_envoy_installer #113: [BUG] stopped working

suave123, 2024-03-21:
**Describe the bug**
A clear and concise description of what the bug is.

**Account type**
- [x ] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: x

**Home Assistant**
- Version: 2023.x

**Additional context**
```
Relevant snippet of Home Assistant error log. 
```

[home-assistant_enphase_envoy_2024-03-21T19-01-09.697Z.log](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/14705847/home-assistant_enphase_envoy_2024-03-21T19-01-09.697Z.log)

Hi Vincent, as of today it stopped working, I think earlier today Enphase had issues themselves and they have been fixed but the plugin cant load. (error communication with API peer closed connection without sending complete message body)

I have enclosed the log from this afternoon.

vincentwolsink, 2024-03-21:
Hi @suave123, it seems from the logs your Envoy is very slow to respond. Try power cycling it.

suave123, 2024-03-21:
That fixed it.Thanks

closed 2024-03-22 by vincentwolsink