vincentwolsink/home_assistant_enphase_envoy_installer #114: Production per Inverter

Dominic-070, 2024-03-23:
Request for sensors per inverter with production today and production lifetime.

vincentwolsink, 2024-03-23:
Hi @Dominic-070, the Envoy does not expose these metrics afaik. Only in total or per phase (if you have CTs installed).
If you want to have these metrics you can create [riemann sum](https://www.home-assistant.io/integrations/integration/#energy) sensors for this.

vincentwolsink, 2024-03-23:
Thinking about it, that won’t be accurate at all. Since the inverters only update the metrics every few minutes.

Dominic-070, 2024-03-23:
I don't mean to calculate the values based on current production. These values are maintained in the inverter and communicated to the envoy. you can see them in the enlighten app.
[image]

mnederlof, 2024-03-23:
Eventhough the data on enlighten is fed from your envoy system and the data should be there, we do not know of any endpoint on the local envoy that exposes this data. 

It would be possible to fetch this data from the enlighten cloud api, but that is beyond the perview of this plugin, as this focuses on the data provided by the envoy itself.

If you discover the endpoint for this data on your local envoy, then please provide it to us, and we can try to integrate this data.

closed 2024-08-06 by vincentwolsink