vincentwolsink/home_assistant_enphase_envoy_installer #116: [Question] Enphase IQ Batteries not showing up

keesvanbemmel, 2024-04-09:
Hi!

Has anyone been able to have this integration pick up Enphase home batteries? And then also added them to the Energy dashboard?

Currently the integration doesn't show the newly installed batteries, even after removing, readding and reauthenticating the integration.

Thanks for any pointers!

vincentwolsink, 2024-04-10:
Hi @keesvanbemmel, I don’t have batteries myself. Can you share your diagnostics file? I will take a look.

keesvanbemmel, 2024-04-10:
I would love to! Where can I find those? :)

vincentwolsink, 2024-04-10:
[image]

Click the three dots next to your Envoy on the right and click Download Diagnostics.

keesvanbemmel, 2024-04-10:
Is there a way I can DM it to you?

vincentwolsink, 2024-04-10:
You can send it to this temporary e-mail: ziener06joduur@icloud.com

keesvanbemmel, 2024-04-10:
done!

vincentwolsink, 2024-04-10:
Can you maybe send me the output of `https://YOUR_ENVOY_IP/ivp/ensemble/inventory` (after logging in to the Envoy).
If you don't know the IP of your Envoy you can go to the Envoy device in Home Assistant and press the 'Visit' button.

keesvanbemmel, 2024-04-12:
Sure! Same email address?

vincentwolsink, 2024-04-14:
I am working on a new implementation for the batteries. Will release a new beta soon.

keesvanbemmel, 2024-04-14:
Would be great if we could also set the import export tariff from the integration
[image]

vincentwolsink, 2024-04-14:
First step to better battery support: [0.3.7-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.3.7-beta1)

keesvanbemmel, 2024-04-14:
Batteries show up and also brings back all sensors (see #117 )

keesvanbemmel, 2024-04-14:
@vincentwolsink if you look at the raw data, is there a metric that shows how much energy is flowing to/from the batteries? Seems like I can only get the current capacity, but the energy dashboards requires energy flows

vincentwolsink, 2024-04-14:
I did not find it yet. I will take a closer look at the data later. Good to hear things are working for you now at least.

keesvanbemmel, 2024-04-14:
Thanks for such a quick turnaround!

vincentwolsink, 2024-04-15:
@keesvanbemmel Could you provide me your output for `https://YOUR_ENVOY_IP/production.json?details=1`

keesvanbemmel, 2024-04-15:
Sure thing, here you go. It doesn't really show anything battery related though. Only the two production phases (I have only 2, not 3) and the 3 consumption phases.
```
{
  "production": [
    {
      "type": "inverters",
      "activeCount": 28,
      "readingTime": 1713200109,
      "wNow": 2731,
      "whLifetime": 950214
    },
    {
      "type": "eim",
      "activeCount": 1,
      "measurementType": "production",
      "readingTime": 1713200138,
      "wNow": 502.888,
      "whLifetime": 191377.398,
      "varhLeadLifetime": 156.765,
      "varhLagLifetime": 109145.762,
      "vahLifetime": 255687.981,
      "rmsCurrent": 2.909,
      "rmsVoltage": 467.682,
      "reactPwr": 436.628,
      "apprntPwr": 678.833,
      "pwrFactor": 0.74,
      "whToday": 17354,
      "whLastSevenDays": 122780,
      "vahToday": 21023,
      "varhLeadToday": 0.765,
      "varhLagToday": 7542,
      "lines": [
        {
          "wNow": 254.284,
          "whLifetime": 95440.931,
          "varhLeadLifetime": 156.485,
          "varhLagLifetime": 55733.957,
          "vahLifetime": 128439.569,
          "rmsCurrent": 1.481,
          "rmsVoltage": 234.098,
          "reactPwr": 217.758,
          "apprntPwr": 345.267,
          "pwrFactor": 0.73,
          "whToday": 8820.931,
          "whLastSevenDays": 61281.931,
          "vahToday": 10627.569,
          "varhLeadToday": 0.485,
          "varhLagToday": 3805.957
        },
        {
          "wNow": 248.604,
          "whLifetime": 95936.467,
          "varhLeadLifetime": 0.28,
          "varhLagLifetime": 53411.805,
          "vahLifetime": 127248.412,
          "rmsCurrent": 1.428,
          "rmsVoltage": 233.584,
          "reactPwr": 218.871,
          "apprntPwr": 333.566,
          "pwrFactor": 0.75,
          "whToday": 8942.467,
          "whLastSevenDays": 61907.467,
          "vahToday": 10816.412,
          "varhLeadToday": 0.28,
          "varhLagToday": 3814.805
        }
      ]
    }
  ],
  "consumption": [
[... 136 lines cut ...]
          "rmsVoltage": 234.269,
          "reactPwr": 13.352,
          "apprntPwr": 36.618,
          "pwrFactor": 0.37,
          "whToday": 0,
          "whLastSevenDays": 0,
          "vahToday": 0,
          "varhLeadToday": 0,
          "varhLagToday": 0
        }
      ]
    }
  ],
  "storage": [
    {
      "type": "acb",
      "activeCount": 0,
      "readingTime": 0,
      "wNow": 0,
      "whNow": 0,
      "state": "idle"
    }
  ]
}
```

vincentwolsink, 2024-04-15:
Could you please also provide the output for the following urls (sorry for asking but I don't have batteries myself).

`https://YOUR_ENVOY_IP/ivp/ensemble/power`
`https://YOUR_ENVOY_IP/ivp/ensemble/secctrl`
`https://YOUR_ENVOY_IP/ivp/ensemble/status`
`https://YOUR_ENVOY_IP/ivp/ensemble/relay`

keesvanbemmel, 2024-04-15:
More than happy to help!

power:
```
{
    "devices:": [
        {
            "serial_num": "122323085065",
            "real_power_mw": 179000,
            "apparent_power_mva": 179000,
            "soc": 89
        },
        {
            "serial_num": "122323085067",
            "real_power_mw": 190000,
            "apparent_power_mva": 190000,
            "soc": 89
        },
        {
            "serial_num": "122323085069",
            "real_power_mw": 183000,
            "apparent_power_mva": 183000,
            "soc": 89
        }
    ]
}
```

secctrl:
```
{
    "shutdown": false,
    "freq_bias_hz": 0.19460000097751618,
    "voltage_bias_v": 2.365999937057495,
    "freq_bias_hz_q8": 313,
    "voltage_bias_v_q5": 75,
    "freq_bias_hz_phaseb": 0.0,
    "voltage_bias_v_phaseb": 0.0,
    "freq_bias_hz_q8_phaseb": 0,
    "voltage_bias_v_q5_phaseb": 0,
    "freq_bias_hz_phasec": 0.0,
    "voltage_bias_v_phasec": 0.0,
    "freq_bias_hz_q8_phasec": 0,
    "voltage_bias_v_q5_phasec": 0,
    "configured_backup_soc": 0,
    "adjusted_backup_soc": 0,
    "agg_soc": 89,
    "Max_energy": 10500,
    "ENC_agg_soc": 89,
    "ENC_agg_soh": 100,
    "ENC_agg_backup_energy": 0,
    "ENC_agg_avail_energy": 9345,
    "Enc_commissioned_capacity": 10500,
    "Enc_max_available_capacity": 10500,
    "ACB_agg_soc": 0,
    "ACB_agg_energy": 0
}
```

relay:
```
{
    "mains_admin_state": "closed",
    "mains_oper_state": "closed",
    "der1_state": 0,
    "der2_state": 0,
    "Enchg_grid_mode": "grid-tied",
    "Solar_grid_mode": "unknown"
}
```
[... 417 lines cut ...]
        "ENC_agg_soc": 89,
        "ENC_agg_soh": 100,
        "ENC_agg_backup_energy": 0,
        "ENC_agg_avail_energy": 9345,
        "Enc_commissioned_capacity": 10500,
        "Enc_max_available_capacity": 10500,
        "ACB_agg_soc": 0,
        "ACB_agg_energy": 0
    },
    "relay": {
        "mains_admin_state": "closed",
        "mains_oper_state": "closed",
        "der1_state": 0,
        "der2_state": 0,
        "Enchg_grid_mode": "grid-tied",
        "Solar_grid_mode": "unknown"
    },
    "profile": {
        "message": "Obsolete API, please use ivp/arf/profile"
    },
    "fakeit": {
        "fake_inventory_mode": false
    }
}
```

vincentwolsink, 2024-04-16:
In 0.3.7-beta2 I extended the battery support with more data. There is an endpoint for current power in W. Unfortunately still no endpoint for Wh.

keesvanbemmel, 2024-04-17:
Looks good! Do you know if the envoy reports the total battery charge state? The batteries themselves have the SOC, but the envoy doesn't show this in your integration atm for all batteries.

I'm looking for the 94% that is at the top of the screenshot. The other percentages are reported by the integration.
[image]

vincentwolsink, 2024-04-17:
It should actually. Did you install beta2? Can you make a screenshot of the available sensors on the   Envoy?

keesvanbemmel, 2024-04-18:
Sorry, didn't check there. It is available indeed!

vincentwolsink, 2024-04-28:
@keesvanbemmel Do you have dedicated CTs installed for your battery? In that case it is actually possible to get the kwh readings.

keesvanbemmel, 2024-04-28:
I have now. Added a Shelly pro em 50 to this setup that reads the kwh

vincentwolsink, 2024-04-28:
I was aiming at Envoy CTs. But this also works ofc. 🙂

keesvanbemmel, 2024-04-28:
Is there a way to hook up more than 6 cts to the envoy?

keesvanbemmel, 2024-04-29:
Well this setup with your additions work like a charm. You could close this issue if you want!

closed 2024-04-29 by vincentwolsink