vincentwolsink/home_assistant_enphase_envoy_installer #117: [BUG] Envoy S not reporting current power production

RoadXY, 2024-04-11:
**Describe the bug**
The envoy device does not report current production, lifetime production like the Envoy is reporting at my parents house where we have the same version of HomeAssistant and the same version Envoy.

I've tried restarting HomeAssistant, restarting the 

**Account type**
- [ ] Installer
- [x] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [x] Standard

- FW version: 7.6.175
- Amount of micro inverters connected: 15

**Home Assistant**
- Version: 2024.4.0

**Additional context**
Envoy has connected inverters:
[image]
The inverters are correctly reporting their data:
[image]

HomeAssistant debug info:
[config_entry-enphase_envoy-8d88b82995dcf247652b38183caed90e.json](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/14947953/config_entry-enphase_envoy-8d88b82995dcf247652b38183caed90e.json)

**Relevant snippet of Home Assistant error log.** 
See attached log file:
[home-assistant_enphase_envoy_2024-04-11T14-47-12.986Z.log](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/14947814/home-assistant_enphase_envoy_2024-04-11T14-47-12.986Z.log)

vincentwolsink, 2024-04-11:
Hi @RoadXY, your Envoy does not return any data on the needed endpoints. Can you try power cycling it?

vincentwolsink, 2024-04-12:
The sensors for current production and lifetime production, are they on 0 or unavailable or non existent?

RoadXY, 2024-04-12:
The sensors for current production are non existent.
My sensors:
[image]

At my parents house:
[image]

RoadXY, 2024-04-12:
I'm also using Node-Red for a while now and I can still get production data from this call: https://192.168.0.50/production.json
This never stopped working and has been working for the last 30 days now.

[image]

Slmarras, 2024-04-12:
Hi,

I have got the same problem for 2 days now. Is there anything I can provide (logging) to fix this problem? Please let me know which logging I can provide and where I can find it in home assistant. 
Kind regards,
Don2c

vincentwolsink, 2024-04-13:
@RoadXY Looking closer at the provided logs, indeed all the needed data is successfully retrieved from the Envoy. Looking at the debug info it is also stored in memory and available.

Since the sensors are not there at all, it seems the data somehow is not available during `async_setup_entry`. But I don't get why it is not happening for the inverter data. Any clue @mnederlof?

keesvanbemmel, 2024-04-14:
Not to pile on, but +1 on this exact same issue 😊

keesvanbemmel, 2024-04-14:
Oh, perhaps good to know: my inverters also don't show any sensors.

vincentwolsink, 2024-04-14:
@keesvanbemmel So basically you don’t have sensors on anything?

keesvanbemmel, 2024-04-14:
No sensors at all is right!

keesvanbemmel, 2024-04-14:
And this is my debug log captured around that time:
```
2024-04-14 19:19:01.719 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Token expires at: 2024-04-15 00:46:04
2024-04-14 19:19:01.720 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP POST Attempt: https://192.168.1.18/auth/check_jwt
2024-04-14 19:19:01.721 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP POST Data: None
2024-04-14 19:19:01.907 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP POST https://192.168.1.18/auth/check_jwt: <Response [200 OK]>: <!DOCTYPE html><h2>Valid token.</h2>

2024-04-14 19:19:01.907 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP POST Cookie: <Cookies[]>
2024-04-14 19:19:01.908 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] Updating endpoints {'endpoint_production_report', 'endpoint_devstatus', 'endpoint_ensemble_json_results', 'endpoint_inventory_results', 'endpoint_home_json_results', 'endpoint_production_json_results', 'endpoint_info_results', 'endpoint_production_power', 'endpoint_installer_agf'}
2024-04-14 19:19:01.909 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_production_report: https://{}/ivp/meters/reports/production
2024-04-14 19:19:01.909 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP GET Attempt #1: https://192.168.1.18/ivp/meters/reports/production: Header:{'Authorization': 'Bearer REDACTED'} Cookies:<Cookies[]>
2024-04-14 19:19:02.140 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.18/ivp/meters/reports/production: <Response [200 OK]>: {
    "createdAt": 1713115141,
    "reportType": "production",
    "cumulative": {
        "currW": 204.792,
        "actPower": 204.792,
        "apprntPwr": 477.099,
        "reactPwr": 417.409,
        "whDlvdCum": 173520.331,
        "whRcvdCum": 59.817,
        "varhLagCum": 99885.732,
        "varhLeadCum": 156.765,
        "vahCum": 232604.019,
        "rmsVoltage": 464.245,
        "rmsCurrent": 2.061,
        "pwrFactor": 0.41,
        "freqHz": 50.00
    },
    "lines": [
        {
            "currW": 102.238,
            "actPower": 102.238,
            "apprntPwr": 235.492,
            "reactPwr": 207.316,
            "whDlvdCum": 86573.223,
            "whRcvdCum": 59.817,
            "varhLagCum": 51110.547,
            "varhLeadCum": 156.485,
            "vahCum": 117008.187,
            "rmsVoltage": 231.733,
            "rmsCurrent": 1.023,
            "pwrFactor": 0.40,
            "freqHz": 50.00
        },
        {
            "currW": 102.555,
            "actPower": 102.555,
            "apprntPwr": 241.607,
            "reactPwr": 210.093,
            "whDlvdCum": 86947.108,
            "whRcvdCum": 0.000,
            "varhLagCum": 48775.185,
            "varhLeadCum": 0.280,
            "vahCum": 115595.833,
            "rmsVoltage": 232.512,
            "rmsCurrent": 1.038,
            "pwrFactor": 0.41,
            "freqHz": 50.00
        }
    ]
}
2024-04-14 19:19:02.141 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] - FETCHING ENDPOINT endpoint_production_report TOOK 0.2315 seconds
2024-04-14 19:19:02.141 INFO (MainThread) [custom_components.enphase_envoy.envoy_reader] UPDATING ENDPOINT endpoint_devstatus: https://{}/ivp/peb/devstatus
2024-04-14 19:19:02.141 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] HTTP GET Attempt #1: https://192.168.1.18/ivp/peb/devstatus: Header:{'Authorization': 'Bearer REDACTED'} Cookies:<Cookies[]>
2024-04-14 19:19:02.411 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.1.18/ivp/peb/devstatus: <Response [200 OK]>: { "counters": {"pcu": {"expected": 28,"discovered": 28,"ctrlsTotal": 28,"ctrlsGone": 0,"ctrlsCommunicating": 28,"chansTotal": 28,"chansRecent": 28,"chansProducing": 28}, "acb": {"expected": 0,"discovered": 0,"ctrlsTotal": 0,"ctrlsGone": 0,"ctrlsCommunicating": 0,"chansTotal": 0,"chansRecent": 0,"chansProducing": 0}, "nsrb": {"expected": 3,"discovered": 3,"ctrlsTotal": 3,"ctrlsGone": 0,"ctrlsCommunicating": 3,"chansTotal": 3,"chansRecent": 3,"chansProducing": 0}, "pld": {"expected": 31,"discovered": 31,"ctrlsTotal": 31,"ctrlsGone": 0,"ctrlsCommunicating": 31,"chansTotal": 31,"chansRecent": 31,"chansProducing": 28}, "esub": {"expected": 0,"discovered": 0,"ctrlsTotal": 0,"ctrlsGone": 0,"ctrlsCommunicating": 0,"chansTotal": 0,"chansRecent": 0,"chansProducing": 0}}, "pcu": {"fields": ["serialNumber", "devType", "communicating", "recent", "producing", "reportDate", "temperature", "dcVoltageINmV", "dcCurrentINmA", "acVoltageINmV", "acPowerINmW"], "values": [["121633021456", 1, true, true, true, 1713114968, 17, 31165, 290, 232767, 8199], ["121633020473", 1, true, true, true, 1713114999, 19, 30495, 270, 232467, 8032], ["121633021497", 1, true, true, true, 1713115029, 18, 30765, 291, 234051, 8497], ["121633021467", 1, true, true, true, 1713115030, 19, 30981, 289, 234679, 8462], ["121633020472", 1, true, true, true, 1713114969, 18, 31066, 276, 232286, 7768], ["121633020464", 1, true, true, true, 1713115059, 22, 30912, 285, 233974, 8285], ["121633020469", 1, true, true, true, 1713114847, 19, 31118, 323, 234461, 9455], ["121633021462", 1, true, true, true, 1713114877, 17, 31767, 290, 235915, 8729], ["121633020656", 1, true, true, true, 1713115031, 17, 30753, 270, 232484, 8023], ["12163302 [... cut ...]
2024-04-14 19:19:07.805 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] EnvoyData.get(voltage_l3) -> None
2024-04-14 19:19:07.805 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Resolving jsonpath endpoint_production_report.cumulative.rmsVoltage
2024-04-14 19:19:07.805 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] EnvoyData.get(voltage) -> 464.245
2024-04-14 19:19:07.805 DEBUG (MainThread) [custom_components.enphase_envoy] Finished fetching envoy Envoy 122009093221 data in 6.086 seconds (success: True)

```

vincentwolsink, 2024-04-15:
Does anyone see any relevant errors in the Home Assistant error logs? And could some of you please also provide me with their diagnostic file for the integration. @HABO7405 @Slmarras 

I assume this issue is fixed for you with the latest beta @keesvanbemmel.

HABO7405, 2024-04-15:
I wish I new how to deliver you the diagnostics file... Could you point to some explanation how to retrieve that?

Searching the error log gives this:

<CUT>
2024-04-12 13:51:20.585 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Envoy 122241087305 for enphase_envoy
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 551, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/__init__.py", line 78, in async_setup_entry
enlighten_serial_num=config[CONF_SERIAL],
~~~~~~^^^^^^^^^^^^^
KeyError: 'serial'
2024-04-12 14:29:09.822 ERROR (MainThread) [switchbot.devices.device] DA-D7-42-B5-D8-2F (DA:D7:42:B5:D8:2F): communication failed; Stopping trying; RSSI: -102
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 363, in establish_connection
await client.connect(
File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 275, in connect
wrapped_backend = self._async_get_best_available_backend_and_device(manager)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/habluetooth/wrappers.py", line 384, in _async_get_best_available_backend_and_device
raise BleakError(
bleak.exc.BleakError: No backend with an available connection slot that can reach address DA:D7:42:B5:D8:2F was found
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 185, in _send_command
return await self._send_command_locked(key, command)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 419, in _send_command_locked
await self._ensure_connected()
File "/usr/local/lib/python3.12/site-packages/switchbot/devices/device.py", line 279, in _ensure_connected
client: BleakClientWithServiceCache = await establish_connection(
^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 472, in establish_connection
_raise_if_needed(name, device.address, exc)
File "/usr/local/lib/python3.12/site-packages/bleak_retry_connector/__init__.py", line 332, in _raise_if_needed
raise BleakOutOfConnectionSlotsError(
bleak_retry_connector.BleakOutOfConnectionSlotsError: DA-D7-42-B5-D8-2F (DA:D7:42:B5:D8:2F) - DA:D7:42:B5:D8:2F: Failed to connect after 9 attempt(s): No backend with an available connection slot that can reach address DA:D7:42:B5:D8:2F was found: The proxy/adapter is out of connection slots or the device is no longer reachable; Add additional proxies (https://esphome.github.io/bluetooth-proxies/) near this device
<CUT>

...

<CUT>
2024-04-15 08:17:39.501 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Envoy 122241087305 for enphase_envoy
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/config_entries.py", line 551, in async_setup
result = await component.async_setup_entry(hass, self)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/enphase_envoy/__init__.py", line 78, in async_setup_entry
enlighten_serial_num=config[CONF_SERIAL],
~~~~~~^^^^^^^^^^^^^
KeyError: 'serial'
<CUT>

vincentwolsink, 2024-04-15:
@HABO7405 Looking at those errors, it seems for you the integration is not working at all? If so, can you maybe create a separate issue.

Slmarras, 2024-04-15:
I re-applied the (Enphase Installer) credentials and now the **current power production** is working again.

RoadXY, 2024-04-16:
Yes!
Deleting and re-adding also works!
[image]

closed 2024-04-18 by vincentwolsink