vincentwolsink/home_assistant_enphase_envoy_installer #126: [BUG] (or Envoy "feature") Lifetime production value stalls instead of continously increasing

KomtGoed, 2024-04-29:
**Describe the bug**
Great software, very much appreciated!

I was wondering whether the following is really a bug or just the way how the Envoy gateway provides the data.

When looking at the graph of the lifetime production, the value stays the same for several minutes, even tough the sun has been shining during that time and production was ongoing:
[image]

I would expect (and need for certain calculations) the lifetime production to be in sync with the data provided by the smart meter (which correctly registers energy being provided back to the grid in this case).

I have the following settings on my Envoy S:
[image]

Is there any logging etc. data I should provide?

Thanks a lot in advance!

**Account type**
- [X] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [X] Standard

- FW version: D7.6.358
- Amount of micro inverters connected: 10

**Home Assistant**
- Version: 2024.4.3

```

vincentwolsink, 2024-04-30:
Hi @KomtGoed. Since you have an Envoy Standard production data needs to be provided from the inverters. This data is updated every 5 minutes at best. If you need/want production data to be updated around every 10 seconds, the only option is to get an Envoy Metered with CTs and enable realtime updates.

Setting the polling interval to anything below 60 seconds does not make sense (since data is updated only every 5 minutes) and could potentially overload the Envoy.

catsmanac, 2024-04-30:
My Envoy standard 7.6.175 is updating lifetime production every 1 minute while it updates the inverter values indeed every 5 minutes. I was expecting the 5 min updates, but when I checked it's actually keeping up with 1 min updates. I'm not running this integration anymore as I have no installer account, and my fw is older by a bit, but just sharing what I see in my data, don't want to complicate matters for you. 

In picture below, top one is lifetime energy production from /api/v1/production and getting new values every 1 minute when polled. Bottom one is an inverter only refreshing every 5 scans as you describe. That /api/v1/production update rate varies when trying, on average is seems to keep up with ~once per 30 sec. I agree on your overload concern though and just run it at 1/min, but in my case it updates lifetime every 1 minute.

[image]

vincentwolsink, 2024-04-30:
Mine is too, actually 😄 @KomtGoed I recommended you to set polling interval back to 60 seconds to see if that improves things. If you want to go lower see my previous comment.

KomtGoed, 2024-04-30:
Gentlemen, thanks a lot for the quick help and helpful explanations. If I remember properly, the upper two values were set to 60s when I'd noticed the issue. Changed it back, we'll keep an eye on it.

Workaround for my use case (const totalMeterReading = envoyProduction - (outputHoog + outputLaag) + (inputHoog + inputLaag);) is to check whether the newly calculated value is higher than the one previously written to InfluxDB.

closed 2024-04-30 by KomtGoed