vincentwolsink/home_assistant_enphase_envoy_installer #128: [BUG] Entities stop responding after Core update to 2024.5.2

suave123, 2024-05-08:
**Describe the bug**
After I update HA Core from 2024.3.0 to 2024.5.2 the entities stop responding and are no longer showing results.

**Account type**
- [x] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: 22

**Home Assistant**
- Version: 2024.3.0

**Additional context**
```
Relevant snippet of Home Assistant error log. 
```

vincentwolsink, 2024-05-10:
Hi. Please check if there is any relevant errors in the HA logs. Otherwise you can try to remove and reconfigure.

suave123, 2024-05-10:
I loaded a backup of 2024.3.0 and then it works again...i will check for errors

MatthewWaanders, 2024-05-12:
I'm running into the same, I'll see if I can grab the logs for you from HASS

vincentwolsink, 2024-05-13:
Hi @MatthewWaanders. And this started happening directly after upgrading HA? Can you maybe send the diagnostics file for the integration?

suave123, 2024-05-13:
for me it started directly after upgrading HA. I eliminated all other updates as to this specific update. (iow I updated all other components I had without issues until HA Core was updated.)

MatthewWaanders, 2024-05-13:
Hi @vincentwolsink. I cannot confirm that for me it started after 2024.05.02 since I've only installed the integration recently and I was already on 2024.05.02. But the behaviour described in the bug is exactly what I'm also seeing: devices going unavailable and their entities reflecting this (and thus not displaying state information). 

For me it has been quite erratic, sometimes it is available for larger amounts of time with only small blips of unavailability in between, but yesterday it was unavailable for 3+ hours.

I've attached the requested file to this comment

[config_entry-enphase_envoy-def964ccfe93b27017df98e0b226a79c.json](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/files/15294625/config_entry-enphase_envoy-def964ccfe93b27017df98e0b226a79c.json)

suave123, 2024-05-13:
yes, correct, when I look at individual entities it seems they are online for a short time but went offline again for a longer period. With my current version of HA I have no issues.

suave123, 2024-05-15:
yes I run ipv4, is that the problem?

MatthewWaanders, 2024-05-15:
@vincentwolsink would a diagnostic file from a moment when the integration was not available help still?

I do have IPv4 and IPv6 enabled for my networks (Ubiquiti UDM SE) with no way of disabling it. The HA box as well as the Enphase Envoy have static IPv4 addresses in the network though. (And they're on the same VLAN) Also HA is set to IPv4 automatic and IPv6 disabled.

There's also some new messages in the logs regarding fetching the meter stream which results in HTTP 502 status codes being returned from /stream/meter upon reconnection attempts.

vincentwolsink, 2024-05-15:
There has been an issue in the past where the integration would alternate between the ipv4 and ipv6 address due to zeroconf on a dualstack network. Maybe you can check which IP it has configured when the integration is not working (easiest method is to click the visit button at the envoy device page in HA). And at the same time you can check if you can actually open the Envoy page while HA cannot connect to it.

suave123, 2024-05-15:
clicking visit on the device page, it says "blocked"

vincentwolsink, 2024-05-15:
@suave123, which url does it go to? And can you make a screenshot?

suave123, 2024-05-15:
browser says: about:blank#blocked

MatthewWaanders, 2024-05-22:
On my side this has magically disappeared and has been running stable for days now without me making any changes to the network or HA so that's welcome.

suave123, 2024-05-25:
all inverters are still visible and reporting, but all the sensors have stopped reporting (like L1 L2 L3 etc.)

closed 2024-10-10 by vincentwolsink