vincentwolsink/home_assistant_enphase_envoy_installer #129: [FEATURE] possibility to charge batteries from grid

keesvanbemmel, 2024-05-12:
Hi!

I would really like the possibility to charge my Enphase batteries from the grid if my dynamic contract hourly price is very cheap using an automation.

Do you know if it is possible to switch to a charging state from home assistant? Either by a switch or by setting the purchase and return prices of electricity in the envoy?

Cheers!

vincentwolsink, 2024-05-13:
Hi, is there any way to trigger this from the Enphase app? I don't have batteries so I don't know.

keesvanbemmel, 2024-05-13:
Only thing I can set is the import and export tariff I guess.

Based on this the envoy determines if it should charge of discharge
[image]

vincentwolsink, 2024-05-13:
I found some endpoint were we can enable/disable charging from the grid. Don't know how well it works, but will try to implement it in a beta release soon.

vincentwolsink, 2024-05-17:
Beta with charge switch and some other settings: [0.3.9-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.3.9-beta1)

gritstub, 2024-05-24:
Pretty new to Home Assistant, but trying my best to debug this.  I do have batteries and running 0.3.9.

I've been playing with Batteries Charge From Grid / switch.batteries_charge_from_grid feature, but it doesn't seem to affect the setting in the Enphase app:
Menu > Settings > Battery > Charge battery from the grid

gritstub, 2024-05-24:
Not sure if this is how it works through HA/Local, but through the enlighten.enphaseenergy.com web app, I see this request when Disabling Charge battery from the grid

```
curl 'https://enlighten.enphaseenergy.com/service/batteryConfig/api/v1/batterySettings/XXXXXX?userId=XXXXXX' \
  -X 'PUT' \
  -H 'accept: application/json, text/plain, */*' \
  -H 'accept-language: en-US,en;q=0.9' \
  -H 'content-type: application/json' \
  -H 'e-auth-token: XXXXXX' \
  -H 'origin: https://battery-profile-ui.enphaseenergy.com' \
  -H 'priority: u=1, i' \
  -H 'referer: https://battery-profile-ui.enphaseenergy.com/' \
  -H 'username: XXXXXX' \
  --data-raw '{"chargeFromGrid":false}' ;
```

and then this request when Enabling Charge battery from the grid
```
curl 'https://enlighten.enphaseenergy.com/service/batteryConfig/api/v1/batterySettings/acceptDisclaimer/XXXXXX' \
  -H 'accept: application/json, text/plain, */*' \
  -H 'accept-language: en-US,en;q=0.9' \
  -H 'content-type: application/json' \
  -H 'e-auth-token: XXXXXX' \
  -H 'origin: https://battery-profile-ui.enphaseenergy.com' \
  -H 'priority: u=1, i' \
  -H 'referer: https://battery-profile-ui.enphaseenergy.com/' \
  -H 'username: XXXXXX' \
  --data-raw '{"disclaimer-type":"itc"}'
```

vincentwolsink, 2024-06-01:
Hi @gritstub, the requests you discovered are via the Enphase cloud. This integration talks directly to the Envoy. There probably is a delay between changing the setting locally in the Envoy and updating the cloud.

keesvanbemmel, 2024-06-06:
The beta seems to work but I'm not 100% sure yet, seems delayed (which is not strange) OR something else is going on. 

Side question: I noticed that there is an "production" switch on the envoy device in this integration. Switching it doesn't seem to do anything, I was under the impression it would cease solar production?

vincentwolsink, 2024-06-10:
@keesvanbemmel Some delay is expected. But it could be there is something else needed for the Envoy to trigger battery charging. But since I don't have a battery I don't know.

The production switch is indeed supposed to stop production. Also here some delay is expected (a few minutes). Although this functionality is half broken for a lot of users in the Netherlands (see this repo README).

keesvanbemmel, 2024-07-30:
So, enphase is currently pushing firmware 8 to my envoy. Once that is done it should also include functionality to charge from grid etc (since firmware 8 supports trade on market). I'll let you know if the integration keeps working AND what it does in terms of trade config.

keesvanbemmel, 2024-08-24:
@vincentwolsink just received firmware V8 on my envoy. 

It seems I can now set the profile of my system to trade energy by itself:
[image]

This profile is not yet available in the integration (didn't expect it to be). 

Is there any output I can provide you from my system to determine what the integration can add in V8 of firmware?

catsmanac, 2024-08-28:
We found in the core integration that with this 8.2.48xx firmware and the AI optimization profile, things seems to [have changed](https://github.com/home-assistant/core/issues/123216) quite a bit. For homeowner accounts at least, the admin/lib/tariff page settings are no longer working, nor does the new profile show in the raw data of the endpoint. Control via the app works, but local control not. Might still work for installer accounts but just that you are aware.

closed 2024-11-27 by vincentwolsink