vincentwolsink/home_assistant_enphase_envoy_installer #132: Implementation of zero export feature

Michielgrs, 2024-06-07:
**Describe the feature**
A clear and concise description of what you would like.

**Account type**
- [x ] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- x[ ] Metered without CTs
- [ ] Standard

- FW version: D7.176
- Amount of micro inverters connected: 14

**Home Assistant**
- Version: 2023.x

Envoy has a rather new function called zero export function. Is that something that can be implemented. I think it can be used instead of the production function, which doesn't always work well. In case of negative prices, zero export appears to be react faster.

vincentwolsink, 2024-06-10:
Hi @Michielgrs. For some time already you could manually load a zero export grid profile, this is possible via the `set_grid_profile` function call in this integration. I read on some forums that from firmware 8 Enphase also has a zero export function tied to dynamic energy prices. As soon as this firmware hits my Envoy I will take a look.

Important detail, you noted down that you have an Envoy metered without CTs installed. For anything zero/limiting export etc you need to have CTs installed for the Envoy to measure your realtime energy production/usage. Without CTs this functionality will not work.

rogervdh, 2024-06-13:
Hi @vincentwolsink. So the Envoy Standard users won't be able to use the zero export functionality in the future? Thats too bad, because I know my real-time import/export in Home Assistant via the P1 port (basically the same information the CTs provide directly to the Envoy Metered) and it would be great if I could adjust the production of my 18 Enphase micro's dynamically via Home Assistant.

vincentwolsink, 2024-06-14:
@rogervdh Correct. This is how Enphase implemented things. All logic is internal to the Envoy and it doesn’t accept any datasource from the outside.

I also don’t see anything like you wish for happening soon, since Enphase provides zero support to any local api or opensource project.

mcjackdus, 2024-06-27:
I can see the option to limit the production in the enlightenment manager
Would be great if this is implemented in the intergration.

vdellevoet, 2024-09-06:
@vincentwolsink the zero grid export profile is not available anymore. Any chance to set "Power Production Limiting by Gateway" via this home assistant integration?

chicknwinner, 2024-11-14:
https://enlighten.enphaseenergy.com/site_pel_settings/5466336/edit?settings_view=true

I can't find any published docs but maybe using that url things can be reverse engineered? having the functions on that page available in HA would be awesome

rogervdh, 2024-11-14:
> https://enlighten.enphaseenergy.com/site_pel_settings/xxxxxx/edit?settings_view=true
> 
> I can't find any published docs but maybe using that url things can be reverse engineered? having the functions on that page available in HA would be awesome

Interesting! I'm using the IQ Gateway Standard and apparently it is possible to limit the production(?). I don't know exactly how it should work (Relay 1, 2, 3 and 4?), but it looks promising. 

[image]

chicknwinner, 2024-11-14:
an example of using the relays here, i'm trying to not have to add extra hardware though

https://enphase.com/en-au/download/setting-pv-power-production-limit-tesla-powerwall-2-tech-brief

catsmanac, 2024-11-14:
Here's an Enlighten [response ](https://support.enphase.com/s/question/0D5Ps00000DegScKAJ/info-drms-demand-response-modes) to a question on the topic. It states only DRM0, `full switch off' is supported. (Maybe more, who knows)

And a community item on the topic describing how to fully switch off: https://community.home-assistant.io/t/stop-solar-panels-from-producing-energy-when-electricity-price-is-negative-tibber-enhpase-envoy/750441/1 using an external relay.

vincentwolsink, 2024-11-15:
For full production switch off there is a switch available in this integration, so you don't have to fiddle with hardware. For production limiting, sadly, it seems Enphase did not make a local API endpoint available. Which is the reason why I cannot implement it in the integration. 

@rogervdh Production limitation can be done with the Envoy standard. For export limitation you need the CT clamps, thus also the Envoy Metered.

jsimonetti, 2024-11-30:
There is also the DRM0 port on your envoy. This works regardless of CT's.

I have resorted to using a shelly wave 1 connected to that port (remember to keep the resistor in series). When the shelly switch is on (relay is closed) production happens like normal, when the shelly switch is open, production stops almost immediately (there's a one or two second delay until production actually stops). Keep in mind to use a dry contact relay for this! The envoy has its own voltage.

This ended up being much more reliable then using the local API.

alexeiw123, 2024-12-16:
> Interesting! I'm using the IQ Gateway Standard and apparently it is possible to limit the production(?). I don't know exactly how it should work (Relay 1, 2, 3 and 4?), but it looks promising. [...]

[Enphase Micro Inverter Curtailment-detailed-NTA.pdf](https://github.com/user-attachments/files/18143374/Enphase.Micro.Inverter.Curtailment-detailed-NTA.pdf)

This document was written up by a member of the amber Australia FB group.

It details how to use this configuration page to set production or export limits, in either binary or stepped states.

At the moment I'm using this integration to limit all production when my batter is full and amber prices are negative.
In the absence of being able to set 'zero export', I've ordered a dry contact so I can set zero export mode via this method, note that I have the required CTs.

rogervdh, 2025-01-25:
I'm was playing with the production toggle on the Envoy in the Envoy Installer HA integration. I'm using the Envoy Standard (FW D8.2.4264) and it takes about 10 minutes for shutting down the production completely. Starting up the production takes again approximately 10 minutes (and I'm reading the real-time production from a MID certified kWh meter). Is such long period normal...?

javydekoning, 2025-05-14:
> I'm was playing with the production toggle on the Envoy in the Envoy Installer HA integration. I'm using the Envoy Standard (FW D8.2.4264) and it takes about 10 minutes for shutting down the [...]

I'm on the same firmware D8.2.4264. Takes about 15 minutes for me to complete shutdown with 25 IQ8+

Seems to take much longer.