vincentwolsink/home_assistant_enphase_envoy_installer #133: [BUG] "Error communicating with API"

rrozema, 2024-06-12:
**Describe the bug**
After installing HA Core 2024.6.2 all entities created by the Enphase Envoy (Installer) are shown as 'Unavailable'. The integration status cycles between 'Initializing' and 'Failed setup, will retry: Error communicating with API:'. Before the update everything worked fine.

Restarted the envoy, restarted HA, restarted router, reloaded the integration nothing helps.

Also -and I think this means that this is something caused by Enphase, not the integration- if I  open in a browser (both Chrome and Edge) http://envoy.local, the page takes a very long time (several minutes of just a wait cursor) to open, then 
```
504 Gateway Time-out
openresty/1.17.8.1
```
 is shown. If I use the envoy's ip address, http://192.168.128.103, the exact same thing happens. Yet, my envoy successfully reports to Enlighten, the data in there does get updated plus shows a success status, and even the live status screen -which is supposed to connect to the envoy directly- works correctly.

**Account type**
- [ X ] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ X ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.6.175
- Amount of micro inverters connected: 16

**Home Assistant**
- Version: 2024.6.2

**Additional context**
[home-assistant_enphase_envoy_2024-06-12T12-30-25.741Z.log](https://github.com/user-attachments/files/15804747/home-assistant_enphase_envoy_2024-06-12T12-30-25.741Z.log)

vincentwolsink, 2024-06-13:
Hi @rrozema, the process pushing data to Enlighten is a different one from serving the API on the Envoy. It is possible that the one broke and the other still working. Can you try to power cycle your Envoy?

rrozema, 2024-06-13:
> <vincentwolsink/home_assistant_enphase_envoy_installer/issues/133/2164518428
> @github.com>

I already did that, and I power cycled the router and ha too (several
times). I also removed the integration and reinstalled it. I also tried
removing the integration, removing the hacs project and tried installing
the official Enphase envoy integration. Still same result. I verified in my
router the envoy still has the same ip: it has a reserved op address in the
dhcp and it is on that ip.
So I am sure now that it is not something in the installer integration, or
at least if it is, it is in the official Enphase envoy integration too.
To me it looks like my IP is getting blocked (maybe intentionally or not?)
when trying to refresh the token required for getting local api access.

rrozema, 2024-06-13:
FYI: just to check I opened https://entrez.enphaseenergy.com/ from my work network and got myself a token from there for my envoy. Tonight I will try to do the same from my home network.

jsimonetti, 2024-06-14:
I have an identical problem. Also started yesterday. My gut says enphase did a booboo...

rrozema, 2024-06-15:
entrez.enephaseenergy.com works from home too, but gettting a fresh token doesn't fix the problem.

I just posted in the Enphase forum a question for help too:
[how-to-fix-504-gateway-timeout-openresty11781-when-calling-httpenvoylocal-or-httplocal-ip](https://support.enphase.com/s/question/0D5Ps00000JvpEUKAZ/how-to-fix-504-gateway-timeout-openresty11781-when-calling-httpenvoylocal-or-httplocal-ip)

rrozema, 2024-06-15:
> I have an identical problem. Also started yesterday. My gut says enphase did a booboo...
> 
> Edit: i just did a `reboot gateway` command from enlighten diagnostics dashboard and that fixed the issue.

How do I get to that enlighten diagnostics dashboard?

rrozema, 2024-06-15:
> > I have an identical problem. Also started yesterday. My gut says enphase did a booboo...
> > Edit: i just did a `reboot gateway` command from enlighten diagnostics dashboard and that fixed the issue.
> 
> How do I get to that enlighten diagnostics dashboard?

Found it.

For me "reboot gateway" did not yet fix the problem. But the situation did change...
However, when I tried opening http://envoy.local/production.json, it showed me this screen
[image] (401 Authorization required)
When I clicked the link in that page, the page again loaded for a long time, but this time I did get a page like this:
[image] (You are not authorized to view that page)

rrozema, 2024-06-15:
Not an issue in the integration. It's the local gateway in the envoy itself messing up with validating security via Enphase's stupid online token authorization (did I mention I hate cloud solutions already?).

closed 2024-06-15 by rrozema