vincentwolsink/home_assistant_enphase_envoy_installer #135: [FEATURE] Grid Profile Details and Upload

Jcr787, 2024-07-18:
**Describe the feature**
A clear and concise description of what you would like.

**Account type**
- [x ] Installer
- [x ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D8.2.127
- Amount of micro inverters connected: 28

**Home Assistant**
- Version: 2024.7.2

I would like to see the grid profile details as a service and upload a custom grid profile. It would be nice to have an example or a form/spreadsheet to fill out to generate the file in the right format to upload a custom grid profile.

vincentwolsink, 2024-07-19:
Hi @Jcr787, an `upload_grid_profile` service is available. A profile generator is kinda out of scope for this integration. Also it would surprise me if the Envoy would accept a profile without any form of signing from Enphase, since this can potentially be pretty dangerous.

Jcr787, 2024-07-20:
Can you provide more information on how yo use? Which format? It is json? Txt file? The idea is to utilize this for off grid situations where grid profile need to be customized.

vincentwolsink, 2024-11-27:
The grid profile files are binary files with high entropy, so it seems they are encrypted.

closed 2024-11-27 by vincentwolsink