vincentwolsink/home_assistant_enphase_envoy_installer #136: [BUG]

jerome6994, 2024-07-26:
**Describe the bug**
Hello, For 3 or 4 days I have had problems with the integration without having modified the configuration of the integration or my solar panel installation.
I just updated Home Assistant to the latest versions, nothing more.

The integration can no longer connect to my installation, whereas before I had no problems at all.
When the problem first appeared, I re-downloaded, restarted HA and the problem was resolved.
When the problem appeared again, I started again and there was nothing at all! So I uninstalled everything, restarted HA, reinstalled, restarted HA and installed the configured integration, checked the cases that were fine normally and still nothing, no connection to my system in phase!
From my Enlighten app, no problems on mobile or internet, I have my production!

If I install the official integration, I have a connection to my system.
Apart from the fact that it works with this integration, I don't have the same entity name so I have to redo everything, template, dashboard, but above all I don't have the same information as with your integration which is really great and which I have been using for quite some time before the update to D7.xxx

I'm a bit helpless to solve the problem and I don't know where to find the information of the problem either.

Thanks for your help

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [x] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.0.88
- Amount of micro inverters connected: 14

**Home Assistant**
- Version: 2024.7.3

vincentwolsink, 2024-07-26:
@jerome6994 Can you please enable debug logging and send the debug logs?

jerome6994, 2024-07-28:
Good morning

I just restarted the installation and I got this message
I would still like to point out that I am the owner of the solar panel installation and I am not the installer, I do not have access to the installer application

`Détails du journal (AVERTISSEMENT)
Cette erreur provient d'une intégration personnalisée
Enregistreur: custom_components.enphase_envoy.envoy_reader
Source: custom_components/enphase_envoy/envoy_reader.py:975
intégration: Enphase Envoy (Installer) ([documentation](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/), [problèmes](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/issues))
S'est produit pour la première fois: 14:51:31 (3 occurrences)
Dernier enregistrement: 14:52:30

Received token is of type owner, disabling installer account usage`

vincentwolsink, 2024-07-29:
Hi @jerome6994, the message about the received token is just a notification, not an error. I do see the error in the screenshot, but I need to know where exactly it happens.

Can you send me the full error log please? If you want you can also send it to me in private at this temporary e-mail address: ziener06joduur@icloud.com

jerome6994, 2024-07-31:
Good morning

I sent you the file via email
With thanks 

Jerome

vincentwolsink, 2024-07-31:
I received your e-mail, thank you. Also spotted the problem. I will try to work on a fix this week.
If you want a quick fix, power cycling your Envoy will most probably fix it for now.

jerome6994, 2024-07-31:
Good evening

Great if you saw the problem and thanks for fixing it in advance.
Have a good evening

vincentwolsink, 2024-08-02:
Please check if the issue is fixed in [0.4.0-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.4.0-beta1).

jerome6994, 2024-08-02:
hello
great ! 
update installed and everything works again without doing anything
more very good job great

closed 2024-08-04 by vincentwolsink