vincentwolsink/home_assistant_enphase_envoy_installer #139: [BUG] The net-consumption not an entity anymore but still present in production.json from what I can see

sebsx, 2024-08-11:
Just a heads up, I upgraded from 0.3.9 to 0.4.1 and the net consumption entity is not created in HA anymore.

**Account type**
- [ ] Installer
- [ ] DIY / DHZ
- [ x] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.6

**Home Assistant**
- Version: 2024.8.1

vincentwolsink, 2024-08-11:
Hi @sebsx, which sensor are you exactly referring to? This integration never exposed the net consumption values, only the calculated consumption values (we could make it an option if you like).

federicotravaini, 2025-02-01:
I shall report the same issue: the default enphase integration priveds net power and net consumption but if I install your custom indegration I don't have them anymore

vincentwolsink, 2025-02-03:
Sensors should be available in [0.6.6](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.6.6)

federicotravaini, 2025-02-05:
Hi, would it be possible to have "Lifetime Net Energy Consumption" and not only the L1 L2 L2 entities? I know I can sum them but I'm thinking to every other new user who will install this integration and won't find an easy way to configure the energy dashboard.

vincentwolsink, 2025-02-05:
Hi, somehow I forgot to create those sensors… Will add them asap.

federicotravaini, 2025-02-05:
> Hi, somehow I forgot to create those sensors… Will add them asap.

if possible it would be awesome to have all sensor required for energy dashboard:

* lifetime net consumption (energy imported from grid)
* lifetime net production (energy exported off grid)
* lifetime power production (solar production - already available)
* lifetime battery production
* lifetime battery consumption

I create most of these sensors using templates and integral sum but if I were a new user I would love to have them available.

Thank you!

vincentwolsink, 2025-02-08:
@federicotravaini, since I don't have CT clamps myself. Could you maybe attach the output of https://envoy.local/ivp/meters and https://envoy.local/ivp/meters/readings. So I can take a look at implementing the requested sensors.

federicotravaini, 2025-02-08:
> [@federicotravaini](https://github.com/federicotravaini), since I don't have CT clamps myself. Could you maybe attach the output of https://envoy.local/ivp/meters and https://envoy.local/ivp/meters/readings. So I can take a look at implementing the requested sensors.

it says I have no authorization to view the page and it asks for a token login but I am no installer. Any workaround?

federicotravaini, 2025-02-08:
> [@federicotravaini](https://github.com/federicotravaini), since I don't have CT clamps myself. Could you maybe attach the output of https://envoy.local/ivp/meters and https://envoy.local/ivp/meters/readings. So I can take a look at implementing the requested sensors.

sorry my bad, didn't see I can login with enphase credentials...

[meters.log](https://github.com/user-attachments/files/18719083/meters.log)
[readings.log](https://github.com/user-attachments/files/18719082/readings.log)

vincentwolsink, 2025-02-24:
All requested sensors should be available in version 0.6.7.

closed 2025-02-24 by vincentwolsink

federicotravaini, 2025-02-24:
Thank you. I will post a feedback after a couple of days just to confirm everything is working fine.

I cannot see 

* lifetime battery production
* lifetime battery consumption

Where you able to add them?

vincentwolsink, 2025-02-24:
For those to appear you need to have storage CTs installed. I don’t see them in the readings you provided.

federicotravaini, 2025-02-24:
> For those to appear you need to have storage CTs installed. I don’t see them in the readings you provided.

mmm very weird because I have them. I will double check and let you know.

Thank you!

federicotravaini, 2025-04-10:
After several weeks of testing, I believe there is an issue with the "Lifetime Net Energy Production" value. As I understand it, Net Energy Production refers to the amount of energy sent back to the grid. In my case, this should be close to zero, since I'm not yet allowed to export energy to the grid (yay Italy, such a great country).
However, this value keeps increasing, as if I'm sending a lot of energy back (10/12 kWh a day).
Could there be a mistake in how this entity is calculated?