vincentwolsink/home_assistant_enphase_envoy_installer #143: [FEATURE] Option to integrate "envoy sensors"

InitialGKL, 2024-08-21:
**Describe the feature**
add the (optional) functionality to read all the "envoy sensors" just like the non-custom enphase envoy integration and (DEV) integration do.
For example the "sensor.envoy_############_today_s_energy_consumption_l2"  and "sensor.envoy_############_current_power_consumption"
I like your integration but personally can't miss these sensors for statistics.

**Account type**
- [ X ] Installer
- [ x ] DIY / DHZ
- [ x ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ x] Metered with CTs installed (3phase)
- [ ] Metered without CTs
- [ ] Standard

- FW version: D8.2.4264.240627
- Amount of micro inverters connected: 31

**Home Assistant**
- Version: 2024.8.2

vincentwolsink, 2024-08-22:
Hi @InitialGKL, the sensors you are referring to should be available (depending on your Envoy configuration). The integration currently supports the following sensors:

```
sensor.envoy_xxx_batteries_available_energy
sensor.envoy_xxx_batteries_capacity
sensor.envoy_xxx_batteries_charge
sensor.envoy_xxx_batteries_power
sensor.envoy_xxx_current_power_consumption
sensor.envoy_xxx_current_power_consumption_l1
sensor.envoy_xxx_current_power_production
sensor.envoy_xxx_current_power_production_l1
sensor.envoy_xxx_current_voltage
sensor.envoy_xxx_current_voltage_l1
sensor.envoy_xxx_grid_profile
sensor.envoy_xxx_lifetime_energy_consumption
sensor.envoy_xxx_lifetime_energy_consumption_l1
sensor.envoy_xxx_lifetime_energy_production
sensor.envoy_xxx_lifetime_energy_production_l1
sensor.envoy_xxx_today_s_energy_consumption
sensor.envoy_xxx_today_s_energy_consumption_l1
sensor.envoy_xxx_today_s_energy_production
sensor.envoy_xxx_today_s_energy_production_l1
sensor.envoy_xxx_today_s_energy_production_l2
sensor.envoy_xxx_today_s_energy_production_l3
binary_sensor.envoy_xxx_grid_status
```

If you don't see the sensors you expect, please supply me with the debug/diagnostic information from the integration.

InitialGKL, 2024-08-23:
Yes. After removing my previous enphase envoy custom repository again, this time with extra HA restarts, your integration works as expected. Just one thing: for a succesful setup of the envoy i had to configure an increased max time for data retrieval from the envoy, i chose 300s. 60 seconds was too short, it kept failing.
Thanks!