vincentwolsink/home_assistant_enphase_envoy_installer #147: Instructions to download / upload a grid profile file or alternative advice

fhopley, 2024-09-02:
Thank you for building this HA integration.

**Describe the bug**
On a previous Envoy firmware (7.6) I had the ability to select and set specific custom grid profiles that Enphase support had put onto my Envoy.  However, since an upgrade (to 8.2) these custom grid profiles are not persistent.  I can get Enphase support to enable one of them, but then the other is no longer available in the grid profiles via local access (or via enphase_envoy.get_grid_profiles).  

What is the best manner to use set_grid_profile in this case?  Is there a way to download the grid profile (after getting Enphase support to enable it for me) and then use upload_grid_profile when I next want to use that profile ?

**Account type**
- [ ] Installer
- [x] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D8.2.127
- Amount of micro inverters connected: 40

**Home Assistant**
- Version: 2024.8.3

**Additional context**
The first grid profile has no export limit, and is generally in use.  The second grid profile has a 0.2kW export limit and is used when feed in tariffs go negative.  The reason I need the 0.2 kW export is that with 0 kW export at times the battery would service home load and be drained, rather than the curtailment of the solar production being reduced.  

Relevant snippet of Home Assistant error log. 
```

vincentwolsink, 2024-09-04:
Hi @fhopley, unfortunately I didn’t have time to look into the grid profiles changes and firmware 8 yet. So currently I do not have an advise for you. I will also look into grid profiles itself at some point, see if I can extract one at some point. But I wonder how feasible it is to re-upload one every time.

upsuper, 2024-10-05:
> But I wonder how feasible it is to re-upload one every time.

It is very feasible to re-upload one every time.

Since my Envoy system starts with 8.x, the grid profile never persists for me (one gets deleted after system switching away from it). I'm using `upload_grid_profile` just before calling `set_grid_profile`, and it has been working reliably ever since I [added](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/pull/108) it to this integration.

But I agree that extracting the grid profile is a tricky one. I have extracted grid profile files, but I feel that it is not appropriate to share my approach publicly. If you need some help, we could chat privately.

HeyyyItsAl, 2024-10-17:
Would be good to see if there is a way to extract grid profiles. Working through with someone who has everything ready, installer access etc but now needing some guidance on setting grid profiles.

Cheers

upsuper, 2024-10-17:
(Feel free to send an email to me if anyone is interested in my approach.)

ppastur, 2025-03-13:
@upsuper I'd be interested. Do you want me to DM you on your profile email?

liamosaur, 2025-09-20:
For the use case where you want to stop exporting to the grid temporarily, the other way to do that other than using Grid Profiles is by using the Power Export Limiting (PEL) feature. This is only available on Envoy S Metered or IQ Gateway with consumption CTs installed.
This can apparently be done via the local API by sending the right POST data to the https://envoy_ip/ivp/ss/dpel endpoint. There is an example on this blog https://enever.nl/zonnepanelen-aan-uit-zetten-enphase/ but I can't find it in any Enphase docs. 
This PEL approach is probably easier than working with grid profiles. @vincentwolsink would you consider adding in a Control that allowed PEL to be turned on and off?

liamosaur, 2025-09-28:
@vincentwolsink I had a go at adding Services that enable/disable Dynamic Power Export Limiting (DPEL), and as of today, it seems to be working! I have never written any Home Assistant code before, so I had no idea what I was doing (please look at the final branch diff, not my trail of misery in the commit log), but it was a fun journey. I'll test it for a while before I submit a PR. At the moment I have the export cap hard-coded at 50 Watts, but it really should be a parameter of the Enable DPEL service. And it would be also good to add in a Binary Sensor to show the state of DPEL (it's just a GET to the same endpoint instead of a POST). You can see the changes here: https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/compare/main...liamosaur:home_assistant_enphase_envoy_installer:main

fhopley, 2025-09-28:
> [@vincentwolsink](https://github.com/vincentwolsink) I had a go at adding Services that enable/disable Dynamic Power Export Limiting (DPEL), and as of today, it seems to be working! I have never written any Home Assistant code before, so I had no idea what I was doing (please look at the final branch diff, not my trail of misery in the commit log), but it was a fun journey. I'll test it for a while before I submit a PR. At the moment I have the export cap hard-coded at 50 Watts, but it really should be a parameter of the Enable DPEL service. And it would be also good to add in a Binary Sensor to show the state of DPEL (it's just a GET to the same endpoint instead of a POST). You can see the changes here: [main...liamosaur:home_assistant_enphase_envoy_installer:main](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/compare/main...liamosaur:home_assistant_enphase_envoy_installer:main)

Wow @liamosaur - Congratulations!  I've stuffed around with the grid profile option in the past and actually got it to work but not reliably.  This sounds like a much better option - and solves a real problem.  I've been investigating a new ability from my retailer (Amber Electric) to control Enphase curtailment but would much rather keep that control myself for a number of reasons.  If this continues to work I'll be keen to use it (ideally via an update)!

liamosaur, 2025-09-30:
@fhopley thanks! I'm using Amber too. I've had this running for a couple of days, and it seems to do exactly what I was hoping. I have two simple automations that are triggered by the Amber FIT value - if it drops below zero, I call the "enable_dpel" service, and if the FIT goes above zero, I call the "disable_dpel" service. I've just pushed an updated release to my fork that exposes the DPEL setting as a binary sensor in the integration.

fhopley, 2025-10-18:
> [@fhopley](https://github.com/fhopley) this added functionality has now been merged into the main branch. You can enable/disable DPEL, exposed as Home Assistant Services. There is also a Sensor for whether DPEL is currently enabled or not. [@vincentwolsink](https://github.com/vincentwolsink) I think this issue can be closed :)

Thanks @liamosaur ... I've updated the integration and can see the enable_dpel and disable_dpel services, but can't see the binary_sensor.dpel_enabled (or any variant of that).  When I call the action/service enable_dpel in the actions tab of developer tools it shows that the perform action button has been pressed but no feedback on success or failure is given (and I couldn't find anything in the system logs related to this).  The diagnostics in the envoy app had only one line in it related to dpel which gave this : "dpel_enabled": null

The firmware on the envoy is D8.2.127 (SKU: ENV-S-WM-230)

I have noticed that I have only intermittent connectivity to the envoy, for example see a voltage trace attached - it could be that I will need to continuously call the service until I get it lucky and call it when there is connectivity.

Any suggestions?

[image]

liamosaur, 2025-10-20:
> The firmware on the envoy is D8.2.127 (SKY ENV-S-WM-230
> <vincentwolsink/home_assistant_enphase_envoy_installer/issues/147/3417643775
> @github.com>

Sounds like there's a couple of issues to work through. The intermittent
connectivity complicates things - is the envoy on your wifi? Could you move
the access point closer or add another AP?

For DPEL - are you using a DIY Installer level account? That is needed to
change these settings.