vincentwolsink/home_assistant_enphase_envoy_installer #148: [FEATURE] Configurable endpoints

jsimonetti, 2024-09-04:
**Describe the feature**
Currently my gateway seems so overloaded, that with the addition of more and more endpoints it starts to stop responding. This has been most evident with the latest addition of pcu communication endpoint (I had to revert to 0.4.0 to get it working again).

In the past I already struggled with the (lack of) reliability of the envoy itself, that I opted to use a modpub power meter to get statistics from my solar installation. Therefor, most statistics are redundant for me and I have even resorted to changing the interval to once every 15 minutes.

There is however one feature of this addon that I currently rely on, which is the production switch. When prices go negative for me, I disable production of my solar installation and enable it again when prices go possitive. This works very well (thank you for that addition!). However, as you may understant, an unresponsive/overloaded gateway gets 'unavailable' in hass, (even though the switch might still work).

I would like to request a configuration item where I can choose which endpoints to disable (obviously, the default should be enabled). This would allow me to disable all endpoints (with the exception of the production switch endpoint) and hopefully improve the reliability of my local api.

**Account type**
- [X] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [X] Metered without CTs
- [ ] Standard

- FW version: D8.2.4264
- Amount of micro inverters connected: 15

**Home Assistant**
- Version: 2024.8

vincentwolsink, 2024-09-04:
Hi @jsimonetti, the Envoy should be capable of handling the load just fine, 15 connected inverters is also no that much. Does it maybe have a bad wifi connection?

Still I can look into optionally disabling endpoints, but I need to see how well the current code can handle this.

jsimonetti, 2024-09-04:
Hi @vincentwolsink ,
Thank you for looking in to this.

The local api of my envoy is very unreliable, in such a way that it very often gives a `504 Gateway Time-out openresty/1.17.8.1` error after waiting several minutes. I think this is just because of horrible software design from enphase's side.

mnederlof, 2024-09-04:
have you also tried power cycling your envoy? Whenever i have issues like this (and it keeps being slow), a powercycle of the envoy fixes the issue for me.

jsimonetti, 2024-09-04:
Hi @mnederlof ,
Yes, unfortunately I have to powercycle the envoy at least once every month when it gets stuck.

Purely out of interest, and not related to this feature request, here's the logging from hass when that happens:
```
2024-09-04 10:12:18.881 ERROR (MainThread) [custom_components.enphase_envoy] Unexpected error fetching envoy Envoy xxxxxxxxx data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 354, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 271, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 91, in async_update_data
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1144, in get_data
    await self.update_endpoints()  # fetch all remaining endpoints
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1128, in update_endpoints
    self.data.set_endpoint_data(endpoint, getattr(self, endpoint))
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 269, in set_endpoint_data
    self.data[endpoint] = response.json()
                          ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_models.py", line 764, in json
    return jsonlib.loads(self.content, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
2024-09-04 10:14:01.628 ERROR (MainThread) [custom_components.enphase_envoy] Unexpected error fetching envoy Envoy xxxxxxxx data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 354, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 271, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 91, in async_update_data
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1135, in get_data
    await self.init_authentication()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 997, in init_authentication
    await self._get_enphase_token()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 919, in _get_enphase_token
    self._token = await self._fetch_envoy_token_json()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 912, in _fetch_envoy_token_json
    raise EnvoyError(
custom_components.enphase_envoy.envoy_reader.EnvoyError: Could not fetch access token from envoy; HTTP 502: <html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
<hr><center>openresty/1.17.8.1</center>
</body>
</html>
```

vincentwolsink, 2024-09-04:
Please try [0.4.4-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.4.4-beta1)

vincentwolsink, 2024-09-04:
I am also thinking about being smarter with endpoint updates. If we discover there are no batteries in the system, do not try to keep updating the battery endpoints for example.

jsimonetti, 2024-09-04:
0.4.4-beta1 works as advertised 
I see the entities belonging to disabled endpoint going unavailable in hass

subjectively, the integration also loads (quite a bit) faster, though i have not done any measurements