vincentwolsink/home_assistant_enphase_envoy_installer #150: [FEATURE] support V8 firmware

keesvanbemmel, 2024-09-09:
**Describe the feature**
My envoy just got the V8 version of its software pushed. It looks like not everything is supported anymore. For example setting battery modes doesn't work anymore.

Would love to see this software supported. What can I do to provide the correct D8 options / endpoints etc?

**Account type**
- [x] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D8.xxx
- Amount of micro inverters connected: 28

**Home Assistant**
- Version: 2024.8.3

vincentwolsink, 2024-09-12:
Hi @keesvanbemmel, V8 firmware is supported. But maybe for the batteries something broke. Can you please supply me with your diagnostics file and any relevant errors from the Home Assistant log?

keesvanbemmel, 2024-09-28:
Well there are not really errors I think. Before v8 firmware I could set the mode of my battery to "backup" which does nothing anymore. In the installer app there is a new mode "AI". Before this this mode was not there. But the backup mode was also never in the app.

Perhaps they've changed the possible modes of the battery? I'm seeing similar behaviour with the "Bateries reservce charge" setting. This does nothing anymore.

Perhaps you need more info from my setup (endpoint jsons eg) to see what's going on?

catsmanac, 2024-09-28:
Hi @keesvanbemmel and @vincentwolsink, in the HA core integration just using the Home Owner account we received [similar reports](https://github.com/home-assistant/core/issues/123216). All data is working but changing settings doesn't work anymore. We tried quite a bit, but no luck yet. Was kind of hoping it would not impact installer users, but with your report it seems an issue all over :-(.

keesvanbemmel, 2024-09-28:
Funnily enough, I came here just now to link the same issue :)

Yeah seems this is not an integration specific issue...

closed 2024-11-27 by vincentwolsink