vincentwolsink/home_assistant_enphase_envoy_installer #153: [BUG] Can't initialize gateway

InitialGKL, 2024-09-15:
**Describe the bug**
Working integration stopped working. Will not initialize gateway. 

**Account type**
- [x ] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [x ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D8.2.4264
- Amount of micro inverters connected: 31

**Home Assistant**
- Version: 2024.9.1

**Additional context**
Tried reinstalling integration, updating HA core. Installing old version.

Relevant snippet of Home Assistant error log.
`` Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.enphase_envoy
Bron: helpers/update_coordinator.py:354
integratie: Enphase Envoy (Installer) (documentatie, problemen)
Eerst voorgekomen: 12:04:16 (6 gebeurtenissen)
Laatst gelogd: 12:13:42

Unexpected error fetching envoy Envoy 122203094873 data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 354, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 271, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 92, in async_update_data
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1125, in get_data
    await self.update_endpoints()  # fetch all remaining endpoints
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1109, in update_endpoints
    self.data.set_endpoint_data(endpoint, getattr(self, endpoint))
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 255, in set_endpoint_data
    self.data[endpoint] = response.json()
                          ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/httpx/_models.py", line 764, in json
    return jsonlib.loads(self.content, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char `0)``

InitialGKL, 2024-09-16:
When I add the hub with my consumer login it does work, but with only 61 entities or so. However I would like the installer level info and options.

vincentwolsink, 2024-09-16:
Can you please enable debug logging, reload the integration (with your installer credentials applied) and send the log output?

InitialGKL, 2024-09-16:
[home-assistant_enphase_envoy_2024-09-15T10-34-57.399Z.log](https://github.com/user-attachments/files/17015396/home-assistant_enphase_envoy_2024-09-15T10-34-57.399Z.log)

vincentwolsink, 2024-09-16:
Somehow the `pcu_comm_check` endpoints returns a 301 on your envoy.

```
2024-09-15 12:16:05.265 DEBUG (MainThread) [custom_components.enphase_envoy.envoy_reader] Fetched from https://192.168.3.134/installer/pcu_comm_check: <Response [301 Moved Permanently]>: 
2
```

Could you try locally logging in to the Envoy as installer and visit https://192.168.3.134/installer/pcu_comm_check to see what output it gives or where it sends you?

InitialGKL, 2024-09-16:
tried again because I  wasn't sure the installer login was active. The Gateway local page is SUPER slow to react. This time i got an Emu picture with page not found.:

"Emu
Page Not Found

The page you tried to view does not exist Try the [home](https://192.168.3.134/) page"""

It looks like an old version of the Enphase installer login webpage with Home, Events, Inventory and Administration sections.

vincentwolsink, 2024-09-16:
Best you can do in that case is disable the endpoint in the integration. Click ‘configure’ and select the ‘pcu_comm_status’ endpoint in the disabled endpoints selector.

InitialGKL, 2024-09-16:
I tried power cycling the envoy first. Nothing changed.
Disabled the pcu_comm_status endpoint did the trick.

Big thanks. But what does this mean? I did have a AC disconnect event in the middle of the night september 14th and a subwoofer I've had for years has died maybe the same day. The subwoofer explodes its fuse (tried a spare fuse 1x) and trips the breaker.

vincentwolsink, 2024-09-17:
The specific endpoint is only used for reading signal levels to the micros. Why your Envoy is so slow I have no idea, might be worth an e-mail to Enphase support.

I don’t expect it having any relation to the power outage. Because if that affected the Envoy it would probably also be completely broken.

closed 2024-09-17 by vincentwolsink