vincentwolsink/home_assistant_enphase_envoy_installer #155: [BUG] lots of disconnections/reconnections

Ayuus, 2024-09-20:
Since a few weeks, I have extra solar panels (and a Q-relais), and some time before, a new firmware version of my Envoy. 
Since then, but not sure if it is sincee Envoy firmware update or extra solar panels, I see lots of disconnections, making all Enphase device unavailable, followed my almost immediate reconnections.
I see this because I get notifications for unavailable entities, but Enphase is not mentioned, because the are already reconnected.
In the Logbook, I can see unavailable/connected for every inverter, envoy and Q-relais.
This happens every 5-10 minutes.
Anyone else experiencing this? Any ideas regarding cause and/or solution?

**Account type**
- [x] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [ ] Metered without CTs
- [x] Standard

- FW version: D8.2.4264
- Amount of micro inverters connected: 29

**Home Assistant**
- Version: 2024.9.2

**Additional context**
New solar panels use slightly different inverters: 
IQ7A: Firmware: 520-00082-r01-v04.30.23 Hardware: 800-00630-r02
where old solare panels have: 
IQ7 inverters Firmware: 520-00082-r01-v04.27.04 Hardware: 800-01127-r02
Anyone else experiencing this? Any ideas regarding cause and/or solution?

Relevant snippet of Home Assistant error log. 
Nothing relevant in there

vincentwolsink, 2024-09-21:
It could be that fetching all data from the Envoy takes around a minute and hits the limit periodically. An option would be to increase timeouts and fetch data every 2 mins. Or you could disable a slow endpoint.

If you turn on debug logging, reload the integration and post the log output here I can see what would be the best option.

Rijswijker, 2024-09-21:
[image]

It looks like I have the same issue

```
2024-09-21 11:57:00.652 WARNING (MainThread) [custom_components.enphase_envoy.envoy_reader] Received token is of type owner, disabling installer account usage
2024-09-21 11:57:01.072 ERROR (MainThread) [custom_components.enphase_envoy] Unexpected error fetching envoy Envoy ************ data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 354, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 271, in _async_update_data
    return await self.update_method()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/__init__.py", line 92, in async_update_data
    await envoy_reader.get_data()
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1125, in get_data
    await self.update_endpoints()  # fetch all remaining endpoints
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 1109, in update_endpoints
    self.data.set_endpoint_data(endpoint, getattr(self, endpoint))
  File "/config/custom_components/enphase_envoy/envoy_reader.py", line 246, in set_endpoint_data
    if response.status_code > 400:
       ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'status_code'
```

Envoy firmware version: D8.2.4264 (882efd)
When i downgrade the plugin to 0.39 everything is working fine.

Ayuus, 2024-09-21:
Increased the timeout for fetching data from Envoy from 60s to 180s. In the log I see several lines like:

`2024-09-21 09:32:52.203 ERROR (MainThread) [custom_components.enphase_envoy] Timeout fetching envoy Envoy 122024035221 data
`
What is a normal timeout? Is the recommended value for this setting dependent on the number of inverters?

vincentwolsink, 2024-09-22:
@Ayuus 60 seconds as timeout should normally be fine. But lately Envoys seem to be slow on certain endpoints, especially on the D8 firmware.

You can also send the debug log to ziener06joduur@icloud.com (temporary e-mail) if you don’t want it to be public.

@Rijswijker That should not happen. I will take a look at the code how to fix this.

Ayuus, 2024-09-22:
Increasing timeout for getting data from Envoy to180 s did resolve this issue. So, not a bug in the integration.
Set debug log to see if anything else could have caused this.

closed 2024-09-22 by Ayuus

Ayuus, 2024-09-23:
**@vincentwolsink** could you see in the log I sent you if Envoy performance was indeed the cause of this issue?

vincentwolsink, 2024-09-23:
@Ayuus Mostly due to the pcu_comm_check endpoint, which gives the communication levels. This is very slow at times. I am considering disabling this endpoint by default.

Ayuus, 2024-09-24:
Disabled Communication Level sensors, reset timeout to 60. To be continued...