vincentwolsink/home_assistant_enphase_envoy_installer #156: [BUG]  after 9.2 unable to load

TGgrimreaper, 2024-09-22:
**Describe the bug**
A clear and concise description of what the bug is.

**Account type**
- [ x] Installer
- [ ] DIY / DHZ
- [x ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ x] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: 12

**Home Assistant**
- Version: 2024.9.2

**Additional context**
```
Envoy *************

Geen apparaten of entiteiten

Instellen mislukt, wordt opnieuw geprobeerd: 'NoneType' object has no attribute 'status_code

Relevant snippet of Home Assistant error log. 
```

Whatsek, 2024-09-22:
I have the exact same issue

epposan, 2024-09-22:
I manage a _Standard_ installation, but it shows the same error.
Reverting to version 0.4.5 seems to fix the issue ...

TGgrimreaper, 2024-09-22:
Cam confirm 0.45 solves isseu

closed 2024-09-23 by vincentwolsink