vincentwolsink/home_assistant_enphase_envoy_installer #160: [BUG] All sensor.envoy_xxx_current_power_consumption not work on 0.4.7 & 0.4.8

strati64, 2024-09-24:
**Describe the bug**
Probe consumption values ​​are no longer returned since update 0.4.7.
Update 0.4.8 Beta has the same problem.
A downgrade to 0.4.6 fixes the problem

**Account type**
- [ ] Installer
- [X ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [X ] Metered with CTs installed
- [ ] Metered without CTs
- [ ] Standard

- FW version: D7.xxx
- Amount of micro inverters connected: 10

**Home Assistant**
-  Version Core 2024.9.2
- Version Supervisor 2024.09.1
- Version Operating System 13.1
- User Interface 20240909.1

**Additional context**
[image]
Relevant snippet of Home Assistant error log. 
```

strati64, 2024-09-24:
Diagnostic files : [0.4.7 Export - config_entry-enphase_envoy-55e0e2e9a889f81140537e8a22efa33c.json](https://github.com/user-attachments/files/17110851/0.4.7.Export.-.config_entry-enphase_envoy-55e0e2e9a889f81140537e8a22efa33c.json) : 0.4.7 not work

[0.4.6 Export - config_entry-enphase_envoy-55e0e2e9a889f81140537e8a22efa33c.json](https://github.com/user-attachments/files/17110865/0.4.6.Export.-.config_entry-enphase_envoy-55e0e2e9a889f81140537e8a22efa33c.json) : 0.4.6 Work !

Jurgen1987, 2024-09-24:
same issue here

Stevenk1234, 2024-09-24:
Yup, same issue here as well

vincentwolsink, 2024-09-24:
If someone could please enable debug logging, reload the integration and send the log here or to temporary e-mail [ziener06joduur@icloud.com](mailto:ziener06joduur@icloud.com)

mvzut, 2024-09-25:
Same here, I mostly miss the Today's Energy Production entity since my energy dashboard uses that!

I'll see if I can post a full debug log

strati64, 2024-09-25:
> If someone could please enable debug logging, reload the integration and send the log here or to temporary e-mail [ziener06joduur@icloud.com](mailto:ziener06joduur@icloud.com)

It's sent !

jarixe, 2024-09-25:
I have the same issue, I have reverted back to 0.45.

vincentwolsink, 2024-09-26:
Please try [0.4.8-beta2](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.4.8-beta2)

jarixe, 2024-09-26:
[image]
[image]

It works for me with the beta version, thank you 😄 👍

mvzut, 2024-09-26:
Yes it works again with 0.4.8b2!

DMPproductions, 2024-09-26:
Yes!  Thank you!!!

Jurgen1987, 2024-09-26:
It Woriking, thank you :)

strati64, 2024-09-27:
I have update this morning (directly 0.4.8) and it's work ! 
Thank's! <3

skillshot28, 2024-10-01:
no, it still not working.

vincentwolsink, 2024-10-01:
Hi @skillshot28, since it is now working for other people your issue might be something local. Can you give some more details what exactly is not working for you?

skillshot28, 2024-10-02:
Hi Vincent,
this morning I still see an accumulated value of the todays envoy of 89kWh
I have solved it with a utility meter.

but after looking further, there is something completely off with HA as my my Energy Distribution also says I used from midnight until now 9:00am, 21kWh which is impossible.

there must be something wrong with HA