vincentwolsink/home_assistant_enphase_envoy_installer #163: [BUG]

Vincentb19, 2024-09-29:
**Describe the bug**
The Lifetime production is not the same as found in the enphase app or the enlighten manager. I don't have the CT's installed and they are also not configured. I checked the production.json and there it also is showing the wrong data? 
[image]
[image]

What i do know is that my enphase is updated from version 2.someting.someting a few months ago by the enphase team itself. I don't remember what the reading was before that because i called them to update the gateway to install this integration. (Could not manually install this update)

**Account type**
- [X] Installer
- [ ] DIY / DHZ
- [ ] Home Owner (This integration is not (fully) functional with a home owner account!)

**Envoy**
- [ ] Metered with CTs installed
- [X] Metered without CTs
- [ ] Standard

- FW version: D8.2.4264
- Amount of micro inverters connected: 14

**Home Assistant**
- Version: 2024.9.1

**Additional context**
```
Relevant snippet of Home Assistant error log. 
```

vincentwolsink, 2024-09-30:
This is unfortunately a know issue with Envoys. Local counters might reset in certain situations. Enphase keeps a separate database for readings in their cloud platform so these readings are correct. But since we only read locally from the Envoy, if the data is wrong there, nothing much we can do.

I might consider adding an option to correct the lifetime data via the integration configuration if you are interested.

Vincentb19, 2024-10-02:
That would be great! So when you configure the integration that you can give it a offset, or tell the integration what the starting values are? Something like that?

vincentwolsink, 2024-10-03:
Thanks @catsmanac. @Vincentb19 can you check if https://ENVOY_IP/api/v1/production reports the correct value?

Vincentb19, 2024-10-03:
[image]
This is what i see when using that endpoint with the installer login. 
I am not able to login into the gateway with the homeowner account so i don't know what that is giving me as a value.

catsmanac, 2024-10-03:
:-(  too bad.

catsmanac, 2024-10-03:
Checked again and and I was wrong in trying to use the /api/v1/production, should have said /production. Apologize for the confusion. @vincentwolsink is right in his remark that "Local counters might reset in certain situations". The values in /production inverters section are reportedly resetting to zero every now and then.

In HA, the energy dashboard can handle the resets, as well as utility meters I think. So you could use these to handle the resets. But getting the same numbers for lifetime as enphase website may need some template logic or what @vincentwolsink has in mind.

vincentwolsink, 2024-10-03:
@Vincentb19 Please try [0.5.0-beta1](https://github.com/vincentwolsink/home_assistant_enphase_envoy_installer/releases/tag/0.5.0-beta1), correct the lifetime production value in the configuration options and reload the integration.

P.S. Beware that this will cause a huge spike in your energy dashboard that you might want to correct after.

Vincentb19, 2024-10-05:
Installed the update yesterday, and it works great. The values increased when the sun was up today. So i am happy with this feature. Thanks for the fast update!